Kinovarobotics/ros_kortex #124: robot moves very slow when call the playCartesianTrajectory

huiwenzhang, 2020-10-13:
Hi, I controlled the robot to move in Cartesian space by calling the service 'playCartesianTrajectory`, as shown in example code:
https://github.com/Kinovarobotics/ros_kortex/blob/239636e49d78cf33e1755205eb0bb15d333a289f/kortex_examples/src/full_arm/example_full_arm_movement.cpp#L143-L182
I want to make the robot move faster, so I increased the poseSpeed, however, the robot still moves very slow. I checked the `readAction` service, same problems happens. Is there any way to make it faster?

alexvannobel, 2020-11-23:
Hi @huiwenzhang ,
Sorry for the long delay.

Are you on firmware version 2.2.0? 
If yes, did you check what are the speed limits for the robot? To see (and set) them, you can go in the WebApp. On the left, click Configuration and then Speed Limits.

Essentially, the robot will not go faster than `poseSpeed.translation` only if this value is lower than the speed limit. The robot never goes faster than the speed limit.

Let me know how this goes!
Best,
Alex

elevenjiang1, 2021-02-19:
> Hi @huiwenzhang ,
> Sorry for the long delay.
>
> Are you on firmware version 2.2.0?
> If yes, did you check what are the speed limits for the robot? To see (and set) them, you can go in the WebApp. On the [...]

My Robot is also move really really slow,Though change the poseSpeed.tranlation,it can not work!

elevenjiang1, 2021-02-21:
After test ros and kortex api of playCartesianTrajectory(kortex api may be action.reach_pose),I find that the speed are same.It may not cause by the robot planning method,or speed limit in robot itself but not in webapp.
Now I can pub cartesian speed to /my_gens/in/cartesian_velocity to make robot move fast,but it is hard to let robot arrive to a specify pose.(0.5m/s)

felixmaisonneuve, 2021-06-16:
Hi,

I will assume that @huiwenzhang's problem was fixed by increasing the speed limit from the web app, then increasing `poseSpeed.translation` and `poseSpeed.orientation`.

Since @elevenjiang1 you also created [this issue](https://github.com/Kinovarobotics/ros_kortex/issues/148), I will clause this one and help you on the other.

Regards,
Felix

closed 2021-06-16 by felixmaisonneuve