Kinovarobotics/ros_kortex #127: Gen3 Lite - All joints publishing 0.0

civerachb-cpr, 2020-10-21:
I'm trying to mount a Gen3 Lite onto a mobile robot platform. I've added the arm to the robot's URDF, and the driver appears to be launching without any errors.  But when I echo the joint states all joints (except the fingers) report 0.0 for their position, regardless of how the arm is physically oriented.

Output of launching the driver:

started roslaunch server http://cpr-dingo-nano:36175/

```SUMMARY
========

PARAMETERS
 * /arm/arm_driver/api_connection_inactivity_timeout_ms: 20000
 * /arm/arm_driver/api_rpc_timeout_ms: 2000
 * /arm/arm_driver/api_session_inactivity_timeout_ms: 35000
 * /arm/arm_driver/arm: gen3_lite
 * /arm/arm_driver/cyclic_data_publish_rate: 40
 * /arm/arm_driver/default_goal_time_tolerance: 0.5
 * /arm/arm_driver/default_goal_tolerance: 0.005
 * /arm/arm_driver/dof: 6
 * /arm/arm_driver/gripper: gen3_lite_2f
 * /arm/arm_driver/gripper_joint_limits_max: [-0.09]
 * /arm/arm_driver/gripper_joint_limits_min: [0.96]
 * /arm/arm_driver/gripper_joint_names: ['arm_right_finge...
 * /arm/arm_driver/ip_address: 192.168.1.10
 * /arm/arm_driver/joint_names: ['arm_joint_1', '...
 * /arm/arm_driver/maximum_accelerations: [1.0, 0.5, 0.4, 1...
 * /arm/arm_driver/maximum_velocities: [0.5, 0.5, 0.5, 0...
 * /arm/arm_driver/password: admin
 * /arm/arm_driver/prefix: arm_
 * /arm/arm_driver/robot_name: arm
 * /arm/arm_driver/sim: False
 * /arm/arm_driver/use_hard_limits: False
 * /arm/arm_driver/username: admin
 * /arm/hztest_test_base_feedback/hz: 40
 * /arm/hztest_test_base_feedback/hzerror: 5.0
 * /arm/hztest_test_base_feedback/test_duration: 30.0
 * /arm/hztest_test_base_feedback/topic: base_feedback
 * /arm/hztest_test_base_feedback/wait_time: 10.0
 * /arm/hztest_test_driver_joint_state/hz: 40
 * /arm/hztest_test_driver_joint_state/hzerror: 5.0
 * /arm/hztest_test_driver_joint_state/test_duration: 30.0
 * /arm/hztest_test_driver_joint_state/topic: base_feedback/joi...
 * /arm/hztest_test_driver_joint_state/wait_time: 10.0
 * /arm/hztest_test_joint_state_publisher_joint_states/hz: 40
 * /arm/hztest_test_joint_state_publisher_joint_states/hzerror: 5.0
 * /arm/hztest_test_joint_state_publisher_joint_states/test_duration: 30.0
 * /arm/hztest_test_joint_state_publisher_joint_states/topic: joint_states
 * /arm/hztest_test_joint_state_publisher_joint_states/wait_time: 10.0
 * /arm/joint_state_publisher/rate: 40
 * /arm/joint_state_publisher/source_list: ['base_feedback/j...
 * /arm/joint_state_publisher/use_gui: False
 * /arm/kortex_driver_tests/api_connection_inactivity_timeout_ms: 20000
 * /arm/kortex_driver_tests/api_rpc_timeout_ms: 2000
 * /arm/kortex_driver_tests/api_session_inactivity_timeout_ms: 35000
 * /arm/kortex_driver_tests/arm: gen3_lite
 * /arm/kortex_driver_tests/cyclic_data_publish_rate: 40
 * /arm/kortex_driver_tests/default_goal_time_tolerance: 0.5
 * /arm/kortex_driver_tests/default_goal_tolerance: 0.005
 * /arm/kortex_driver_tests/dof: 6
 * /arm/kortex_driver_tests/gripper: gen3_lite_2f
 * /arm/kortex_driver_tests/gripper_joint_limits_max: [-0.09]
 * /arm/kortex_driver_tests/gripper_joint_limits_min: [0.96]
 * /arm/kortex_driver_tests/gripper_joint_names: ['arm_right_finge...
 * /arm/kortex_driver_tests/ip_address: 192.168.1.10
 * /arm/kortex_driver_tests/joint_names: ['arm_joint_1', '...
 * /arm/kortex_driver_tests/maximum_accelerations: [1.0, 0.5, 0.4, 1...
 * /arm/kortex_driver_tests/maximum_velocities: [0.5, 0.5, 0.5, 0...
 * /arm/publish_test_kortex_driver/topics: [{'name': '/arm/b...
 * /rosdistro: melodic
 * /rosversion: 1.14.9

NODES
  /arm/
    arm_driver (kortex_driver/kortex_arm_driver)
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

ROS_MASTER_URI=http://localhost:11311

process[arm/arm_driver-1]: started with pid [19233]
process[arm/joint_state_publisher-2]: started with pid [19234]
process[arm/robot_state_publisher-3]: started with pid [19235]
[ WARN] [1603303485.136751012]: The root link arm_base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[ INFO] [1603303485.762735895]: Session created successfully for TCP services
[ INFO] [1603303485.772901267]: Session created successfully for UDP services
[ INFO] [1603303485.782607917]: -------------------------------------------------
[ INFO] [1603303485.782719433]: Scanning all devices in robot... 
[ INFO] [1603303485.807014394]: Base device was found on device identifier 0
[ INFO] [1603303485.808454990]: Actuator device of type SMALL_ACTUATOR was found on device identifier 7
[ INFO] [1603303485.808566350]: Actuator device of type SMALL_ACTUATOR was found on device identifier 5
[ INFO] [1603303485.808614269]: Actuator device of type MEDIUM_ACTUATOR was found on device identifier 1
[ INFO] [1603303485.808665938]: Actuator device of type BIG_ACTUATOR was found on device identifier 2
[ INFO] [1603303485.808708024]: Actuator device of type MEDIUM_ACTUATOR was found on device identifier 3
[ INFO] [1603303485.808758443]: Actuator device of type SMALL_ACTUATOR was found on device identifier 4
[ INFO] [1603303485.808799279]: -------------------------------------------------
[ INFO] [1603303485.879463733]: State changed from INITIALIZING to IDLE

[ INFO] [1603303485.900031573]: -------------------------------------------------
[ INFO] [1603303485.900223198]: Initializing Kortex Driver's services...
[ INFO] [1603303488.253699341]: Kortex Driver's services initialized correctly.
[ INFO] [1603303488.253829296]: -------------------------------------------------
[ INFO] [1603303488.257125407]: The Kortex driver has been initialized correctly!```

Output of echoing the joint state:

```rostopic echo /arm/joint_states  -n 5
header: 
  seq: 14199
  stamp: 
    secs: 1603303902
    nsecs: 118694067
  frame_id: ''
name: [arm_joint_1, arm_joint_2, arm_joint_3, arm_joint_4, arm_joint_5, arm_joint_6, arm_right_finger_bottom_joint,
  arm_right_finger_tip_joint, arm_left_finger_bottom_joint, arm_left_finger_tip_joint]
position: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.149, 0.0, 0.149]
velocity: []
effort: []
---
header: 
  seq: 14200
  stamp: 
    secs: 1603303902
    nsecs: 143697977
  frame_id: ''
name: [arm_joint_1, arm_joint_2, arm_joint_3, arm_joint_4, arm_joint_5, arm_joint_6, arm_right_finger_bottom_joint,
  arm_right_finger_tip_joint, arm_left_finger_bottom_joint, arm_left_finger_tip_joint]
position: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.149, 0.0, 0.149]
velocity: []
effort: []
---
header: 
  seq: 14201
  stamp: 
    secs: 1603303902
    nsecs: 168684005
  frame_id: ''
name: [arm_joint_1, arm_joint_2, arm_joint_3, arm_joint_4, arm_joint_5, arm_joint_6, arm_right_finger_bottom_joint,
  arm_right_finger_tip_joint, arm_left_finger_bottom_joint, arm_left_finger_tip_joint]
position: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.149, 0.0, 0.149]
velocity: []
effort: []
---
header: 
  seq: 14202
  stamp: 
    secs: 1603303902
    nsecs: 193634033
  frame_id: ''
name: [arm_joint_1, arm_joint_2, arm_joint_3, arm_joint_4, arm_joint_5, arm_joint_6, arm_right_finger_bottom_joint,
  arm_right_finger_tip_joint, arm_left_finger_bottom_joint, arm_left_finger_tip_joint]
position: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.149, 0.0, 0.149]
velocity: []
effort: []
---
header: 
  seq: 14203
  stamp: 
    secs: 1603303902
    nsecs: 218625068
  frame_id: ''
name: [arm_joint_1, arm_joint_2, arm_joint_3, arm_joint_4, arm_joint_5, arm_joint_6, arm_right_finger_bottom_joint,
  arm_right_finger_tip_joint, arm_left_finger_bottom_joint, arm_left_finger_tip_joint]
position: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.149, 0.0, 0.149]
velocity: []
effort: []
---```

In rviz I can see the robot's arm sticking straight up in the air, despite it being curled up in the stowed position on the physical robot.

I can't see any errors that would be preventing the arm from working, but maybe I'm missing something?  I've updated the arm's firmware to the latest version & rebuilt the driver a few times, but still no luck.

alexvannobel, 2020-10-21:
Hi @civerachb-cpr ,

This is a weird error, because the driver looks like it initializes just fine.

A couple questions to help debug this:
- Can you show what your launch file looks like?
- With what exact command line do you launch this file?
- Can `rostopic echo /arm/base_feedback` and `rostopic echo /arm/base_feedback/joint_state`? Those are the topics the arm driver output. The `base_feedback/joint_state` is the topic we use [in our launch file](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/launch/kortex_driver.launch#L103) as a source_list for the robot's joint states. Can you make sure you also launch your `joint_state_publisher` like so?

Best,
Alex

civerachb-cpr, 2020-10-21:
The launch file I'm using is largely the same as the default kortex_driver.launch file, just with some of the defaults replaced with optenv's:

I've exported the IP address to the default 192.168.1.10, and can ping the arm & access the web interface as-expected.

```<launch>
  <!-- Arm configuration -->
  <arg name="arm" default="$(optenv DINGO_ARM_MODEL gen3_lite)"/>
  <arg name="dof" default="$(optenv DINGO_ARM_DOF 6)"/>
  <arg name="vision" default="false"/>
  <arg name="use_hard_limits" default="false"/>
  <arg name="gripper" default="$(optenv DINGO_ARM_GRIPPER gen3_lite_2f)"/>

  <!-- Namespace -->
  <arg name="robot_name" default="$(optenv DINGO_ARM_PREFIX arm)"/>
  <arg name="prefix" default="$(optenv DINGO_ARM_PREFIX arm)_"/>

  <!-- Kortex API options -->
  <arg name="ip_address" default="$(optenv DINGO_ARM_HOST 192.168.131.40)"/>
  <arg name="username" default="admin"/>
  <arg name="password" default="admin"/>
  <arg name="cyclic_data_publish_rate" default="100"/> <!--Hz-->
  <arg name="api_rpc_timeout_ms" default="2000"/> <!--milliseconds-->
  <arg name="api_session_inactivity_timeout_ms" default="35000"/> <!--milliseconds-->
  <arg name="api_connection_inactivity_timeout_ms" default="20000"/> <!--milliseconds-->

  <!-- Automatically start other modules -->
  <arg name="start_rviz" default="false"/>
  <arg name="start_moveit" default="false"/>

  <!-- Action server params -->
  <arg name="default_goal_time_tolerance" default="0.5"/> <!--seconds-->
  <arg name="default_goal_tolerance" default="0.005"/> <!--radians-->

  <group ns="$(arg robot_name)">
      <group if="$(eval not arg('prefix'))">
          <!-- Load the description for the robot -->
          <!-- Without gripper -->
          <param name="robot_description" command="$(find xacro)/xacro --inorder $(find kortex_description)/robots/$(arg arm).xacro dof:=$(arg dof) vision:=$(arg vision) sim:=false"
              if="$(eval not arg('gripper'))"/>
          <!-- With gripper -->
          <param name="robot_description" command="$(find xacro)/xacro --inorder $(find kortex_description)/robots/$(arg arm)_$(arg gripper).xacro dof:=$(arg dof) vision:=$(arg vision) sim:=false"
              unless="$(eval not arg('gripper'))"/>
      </group>
      <group unless="$(eval not arg('prefix'))">
          <!-- Load the description for the robot -->
          <!-- Without gripper -->
          <param name="robot_description" command="$(find xacro)/xacro --inorder $(find kortex_description)/robots/$(arg arm).xacro dof:=$(arg dof) vision:=$(arg vision) sim:=false prefix:=$(arg prefix)"
              if="$(eval not arg('gripper'))"/>
          <!-- With gripper -->
          <param name="robot_description" command="$(find xacro)/xacro --inorder $(find kortex_description)/robots/$(arg arm)_$(arg gripper).xacro dof:=$(arg dof) vision:=$(arg vision) sim:=false prefix:=$(arg prefix)"
              unless="$(eval not arg('gripper'))"/>
      </group>

      <!-- Start the kortex_driver node -->
      <node name="$(arg robot_name)_driver" pkg="kortex_driver" type="kortex_arm_driver" output="screen"> <!--launch-prefix="gdb -ex run args"-->
          <param name="sim" value="false"/>
          <param name="ip_address" value="$(arg ip_address)"/>
          <param name="username" value="$(arg username)"/>
          <param name="password" value="$(arg password)"/>
          <param name="cyclic_data_publish_rate" value="$(arg cyclic_data_publish_rate)"/>
          <param name="api_rpc_timeout_ms" value="$(arg api_rpc_timeout_ms)"/>
          <param name="api_session_inactivity_timeout_ms" value="$(arg api_session_inactivity_timeout_ms)"/>
          <param name="api_connection_inactivity_timeout_ms" value="$(arg api_connection_inactivity_timeout_ms)"/>
          <param name="default_goal_time_tolerance" value="$(arg default_goal_time_tolerance)"/>
          <param name="default_goal_tolerance" value="$(arg default_goal_tolerance)"/>
          <param name="arm" value="$(arg arm)"/>
          <param name="gripper" value="$(arg gripper)"/>
          <param name="dof" value="$(arg dof)"/>
          <param name="use_hard_limits" value="$(arg use_hard_limits)"/>
          <param name="robot_name" value="$(arg robot_name)"/>
[... 116 lines cut ...]
  stamp: 
    secs: 1603313774
    nsecs: 304326730
  frame_id: ''
name: [arm_arm_joint_1, arm_arm_joint_2, arm_arm_joint_3, arm_arm_joint_4, arm_arm_joint_5,
  arm_arm_joint_6, arm_arm_right_finger_bottom_joint]
position: [-0.06578220945382984, 2.262942839358784, 2.637309252745621, -1.4626229843736223, -0.853158568908988, -1.7358111142583867, 0.9598949980846789]
velocity: [7.1410476755214e-05, 3.698577933632552e-05, 5.2836825881753035e-05, 0.00011102011344922246, -7.132900847284415e-05, 7.409650813126577e-05, -0.002166579943150282]
effort: [-0.0068731061182916164, 0.0, 0.10309658944606781, 0.08319978415966034, -0.06988781690597534, 0.003327991347759962, 0.0]
---
header: 
  seq: 207
  stamp: 
    secs: 1603313774
    nsecs: 329433329
  frame_id: ''
name: [arm_arm_joint_1, arm_arm_joint_2, arm_arm_joint_3, arm_arm_joint_4, arm_arm_joint_5,
  arm_arm_joint_6, arm_arm_right_finger_bottom_joint]
position: [-0.06578647051157382, 2.262942839358784, 2.6373140464355833, -1.4626261801669305, -0.8531436552068836, -1.7358111142583867, 0.9598937773764524]
velocity: [4.7553143699943855e-05, 2.1146688082534686e-05, 7.397155867265104e-05, 0.0, 0.0, 3.696487992609866e-05, -0.0010083504021167755]
effort: [-0.013746212236583233, 0.02290387824177742, 0.0893503800034523, 0.08652777224779129, -0.05990384519100189, 0.003327991347759962, 0.0]
---
```

It would appear that base_feedback/joint_state is reporting the correct angles, but the joint names have an extra "arm_" prefix.  Not sure where that's coming from, as the prefix set in the launch file is simply "arm_" (this is confirmed in the output of the parameters when starting the driver, in the initial post).

alexvannobel, 2020-10-22:
Hi @civerachb-cpr ,
Can you pull the latest of `kinetic-devel` and try again? I just merged a PR that fixed it on my side.
Don't hesitate to reopen if the problem persists.
Best,
Alex

closed 2020-10-22 by alexvannobel

civerachb-cpr, 2020-10-22:
That appears to have fixed the immediate problem of the arm's joints not being published correctly.  Echoing `/arm/joint_states` gives meaningful information now.

civerachb-cpr, 2020-10-22:
Looks like this bug is present in a second location.  When I try to plan motions with moveit the driver reports that goal positions have the extra prefix:

```
...
Oct 22 14:37:13 cpr-dingo-nano ros-start[24488]: arm_joint_1, arm_joint_2, arm_joint_3, arm_joint_4, arm_joint_5, arm_joint_6,
Oct 22 14:37:13 cpr-dingo-nano ros-start[24488]: [ERROR] [1603391833.855958303]: Joint Trajectory Goal is rejected.
Oct 22 14:37:54 cpr-dingo-nano ros-start[24488]: [ERROR] [1603391874.773974213]: There is a mismatch between the goal's joint names and the action server's joint names.
Oct 22 14:37:54 cpr-dingo-nano ros-start[24488]: [ INFO] [1603391874.773895254]: New goal received.
Oct 22 14:37:54 cpr-dingo-nano ros-start[24488]: [ INFO] [1603391874.774037599]: Action server joint names are :
Oct 22 14:37:54 cpr-dingo-nano ros-start[24488]: arm_arm_joint_1, arm_arm_joint_2, arm_arm_joint_3, arm_arm_joint_4, arm_arm_joint_5, arm_arm_joint_6,
Oct 22 14:37:54 cpr-dingo-nano ros-start[24488]: [ INFO] [1603391874.774088901]: Goal joint names are :
Oct 22 14:37:54 cpr-dingo-nano ros-start[24488]: arm_joint_1, arm_joint_2, arm_joint_3, arm_joint_4, arm_joint_5, arm_joint_6,
Oct 22 14:37:54 cpr-dingo-nano ros-start[24488]: [ERROR] [1603391874.774181507]: Joint Trajectory Goal is rejected.
```

Moveit just gives a warning that the execution failed:

```
[ INFO] [1603391874.774663334]: Controller arm/gen3_lite_joint_trajectory_controller successfully finished
[ WARN] [1603391874.774846617]: Controller handle arm/gen3_lite_joint_trajectory_controller reports status FAILED
```

Similarly rviz reports that the planning succeeded but the execution failed:

```
[ INFO] [1603391611.366458370]: Ready to take commands for planning group manipulator.
[ INFO] [1603391611.366518067]: Looking around: no
[ INFO] [1603391611.366541456]: Replanning: no
[ WARN] [1603391654.102937901]: Fail: ABORTED: Motion plan was found but it seems to be invalid (possibly due to postprocessing). Not executing.
[ INFO] [1603391798.872249196]: ABORTED: Solution found but controller failed during execution
[ INFO] [1603391834.199511676]: ABORTED: Solution found but controller failed during execution
[ INFO] [1603391874.806639211]: ABORTED: Solution found but controller failed during execution
```

civerachb-cpr, 2020-10-26:
The Bugfix/double prefix branch appears to fix the problem.  Thanks for the quick turn-around!

alexvannobel, 2020-10-26:
I just merged the branch.
Thanks again for the feedback.
Best,
Alex

PR #128 (2020-10-22): Remove prefix from driver as it's already in the yaml joint_limits file
Fixes #127