Kinovarobotics/ros_kortex #130: Joint torque control with Kinova gen3 lite.

ruoshiwen, 2020-10-27:
Hi, I am using a Kinova Gen3 lite. Is it possible to do joint torque control with this arm? And what feedback can I have access to from each joint? Thanks a lot.

alexvannobel, 2020-11-23:
Hi @ruoshiwen ,
Sorry for the delay.

Gen3 lite does not have torque sensors, so you cannot technically do joint torque control.

What you could do instead is try to control the arm with a current loop. You have access to each actuator's current in the feedback and can control the actuators in current in low-level mode, and that's what's probably the closest to achieving joint torque control on Gen3 lite. 
It is complicated, cumbersome and not perfect but nonetheless an alternative worth mentioning.

I'll close the issue since actual torque control is not possible on Gen3 lite.
Best,
Alex

closed 2020-11-23 by alexvannobel

alexvannobel, 2020-11-23:
I can also add that current control is our most advanced mode and also the least safe.
Care must be taken when using it.

ruoshiwen, 2020-11-23:
Thank you for the detailed instruction and heads up. It’ll be very helpful.

JaeYoungLee79, 2021-12-01:
I'm using gen3_lite. There is not a example of current control? And gen3_lite has little vibration with moving and also it remaining after the end of position.

felixmaisonneuve, 2021-12-01:
Hi @JaeYoungLee79,

Alex was referring to Low-Level mode. This is not related to ROS and you will find examples and documentation in our other repositery https://github.com/Kinovarobotics/kortex. If you say your arm vibrates when moving, please open a new issue and we can look into it