Kinovarobotics/ros_kortex #131: ABORTED: Solution found but controller failed during execution

heiyumiao, 2020-10-28:
When I use /ros_kortex/kortex_examples/src/move_it/example_move_it_trajectories.py to control the robot,it always show some errors.  
It says  Invalid Position, how to plan an trajectory without  out of limit?

--------------------------------------------------------------------------------- 
Error #31
Type : TRAJECTORY_ERROR_TYPE_JOINT_POSITION_LIMIT
Identifier : TRAJECTORY_ERROR_IDENTIFIER_POSITION
Actuator : 4
Erroneous value is -147.964 but minimum permitted is -147.823 and maximum permitted is 147.823
Additional message is : Invalid Position - Position of actuator(3) in Trajectory Point (2046) exceeds limits
-----------------------------------------------------------------------------------------------------

Here is the code:
def reach_go_to_stable(self, tolerance):
    arm_group = self.arm_group
    success = True

    # Get the current joint positions
    joint_positions = arm_group.get_current_joint_values()
    rospy.loginfo("Printing current joint positions before movement :")
    for p in joint_positions: rospy.loginfo(p)

    # Set the goal joint tolerance
    self.arm_group.set_goal_joint_tolerance(tolerance)

    joint_positions[0] = 0
    joint_positions[1] = -1.27
    joint_positions[2] = 4.15
    joint_positions[3] = -2.38
    joint_positions[4] = 0
    joint_positions[5] = -0.5
    joint_positions[6] = 1.57
  
    arm_group.set_joint_value_target(joint_positions)
    success &= arm_group.go(wait=True)
    new_joint_positions = arm_group.get_current_joint_values()
    return success

alexvannobel, 2020-11-23:
Hi @heiyumiao ,

This is a validation error returned by the robot whenever it validates a MoveIt trajectory before playing it. This particular error means a joint value for the fourth actuator exceeds the joint limit in your trajectory.
I don't see this exact position used [in our example](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_examples/src/move_it/example_move_it_trajectories.py#L116-L123).
Can you change the joint goal for the fourth actuator and see if the error is still there?.

Let me know how this goes,
Alex

heiyumiao, 2020-11-24:
We modified it from the following code.

```
  def reach_joint_angles(self, tolerance):
    arm_group = self.arm_group
    success = True

    # Get the current joint positions
    joint_positions = arm_group.get_current_joint_values()
    rospy.loginfo("Printing current joint positions before movement :")
    for p in joint_positions: rospy.loginfo(p)

    # Set the goal joint tolerance
    self.arm_group.set_goal_joint_tolerance(tolerance)

    # Set the joint target configuration
    if self.degrees_of_freedom == 7:
      joint_positions[0] = pi/2
      joint_positions[1] = 0
      joint_positions[2] = pi/4
      joint_positions[3] = -pi/4
      joint_positions[4] = 0
      joint_positions[5] = pi/2
      joint_positions[6] = 0.2
```

The error isn't always there when I'm not changing the joint goal. Sometimes it works.
Besides the same error happened when we changed the joint goal for the fourth actuator. 

We want to know how to avoid exceeding the limit. Thanks!

alexvannobel, 2020-11-24:
Hi @heiyumiao ,

There could be a couple reasons here why the joint limit is exceeded during the trajectory:
- One (or some) joint **goals** exceed the limit. The solution to this is to specify goals that don't exceed joint goals. You can see those hard limits [in the URDF](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_description/arms/gen3/7dof/urdf/gen3_macro.xacro#L99).
- The interpolator exceeds a joint limit **in between two points**. This is not supposed to happen and it never happened to me, but if you see that your ending point (or starting point) is not the problem, you could try narrowing a bit the joint limits in the MoveIt configuration files. [This](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_move_it_config/gen3_move_it_config/config/7dof/hard_joint_limits.yaml) is the file for the 7DOF Gen3 with no gripper but you would have to modify the MoveIt config you use. By narrowing the joint limits just a smidge on the problematic joint it should make the interpolator respect the arm's joint angles more (define the `min_position` and `max_position` fields).

Let me know how this goes,
Alex

Guillaume-Syc, 2021-03-02:
Hello @heiyumiao and @alexvannobel,

I have the same issue, have you solve it ?
I just import the kortex_ros project and launch the kortex_driver, the moveit_exemple. The robot don't move and I get the same errors:

Error #2
Type : TRAJECTORY_ERROR_TYPE_LARGE_SPEED
Identifier : TRAJECTORY_ERROR_IDENTIFIER_VELOCITY
Actuator : 3
Erroneous value is 25.0294 but minimum permitted is 0 and maximum permitted is 25.0039
Additional message is : Invalid Speed - Speed of actuator(2) in Trajectory Point (1483) exceeds limits
-----------------------------
-----------------------------
Error #3
Type : TRAJECTORY_ERROR_TYPE_INVALID_POSITION
Identifier : TRAJECTORY_ERROR_IDENTIFIER_POSITION
Actuator : 3
Erroneous value is 0.0250258 but minimum permitted is -0.0250039 and maximum permitted is 0.0250039
Additional message is : Invalid Position - Position variation between Trajectory Point (1483) and previous one unacheivable with current speed limits
-----------------------------

I tried to change the value of  the hard_joint_limits.yaml file but nothing change

HeegerGao, 2021-03-20:
I have got the same issue when using your moveit example. I found that in the simulation, the trajecotry can be executed corectly, but in the real arm, it always failed when I want to move for a relatively large distance, for example, 0.2m in Cartesian space. If I only need to move 0.02m, the moveit example can run normally.

heiyumiao, 2021-03-20:
The problem haven't been solved.  It still happens sometimes. @alexvannobel @Guillaume-Syc

felixmaisonneuve, 2021-06-14:
Hi guys,

With the release of the version 2.3, modifications in MoveIt configurations have been made that should fix this type of errors. More details about the new release [here](https://github.com/Kinovarobotics/ros_kortex/releases/tag/v2.3.0).

I will keep this issue open so you can try it. Please leave a comment if you still get an error with the new release. Otherwise, I will prbably close this issue in a couple of weeks 

Regards,
Felix

felixmaisonneuve, 2021-07-26:
I will close this issue. 

Please open a new issue or comment/re-open this one if you still get this error

Thank you,
Felix

closed 2021-07-26 by felixmaisonneuve