Kinovarobotics/ros_kortex #132: Gen3 arm no response any more when using moveit for a while

flymaxty, 2020-11-14:
Hi,
I' have met a strange issue with moveit and real Gen3 arm

When we using moveit to control Gen3 arm, It works well in first few minutes, after then, the kortex driver while throw an error like below, and kortex_arm_driver node died.

```bash
terminate called after throwing an instance of 'std::runtime_error'
   what(): timeout detected: BaseClinet::ClearFaults
```

And if this happens, the arm will not be able to connect again with both kortex_driver and WebApp.

The kortex will throw the following error:
 ```bash
terminate called after throwing an instance of 'std::runtime_error'
   what(): timeout detected: SessionClient::CreateSession
```

The jotstick, which connect to the arm USB port, can not control the arm either.

All LED looks normal, no red or blink, network PING works, and the button on the end effector also can active the admittance.

Looks like the low level controller(MCU??)  works well, but the mainboard or onboard process died and, there's no watchdog for this situation.

We can only reboot the arm by power button, and then all works well.

We also tried the kortext_examples node, `example_full_arm_movement.py` more than 10 minutes, and nothing happens.
But `example_move_it_trajectories` returns the same error while execute more than 8 minutes.(all node add a infinity loop)

Is that meas the issue cames from the low-level control? Because the `example_full_arm_movement.py` use the high-level control, and moveit use low-level control to control the actuators, is I understand correctly.

Is anyone has ever met the same issue or has any idea?

Thank you!

elevenjiang1, 2021-02-19:
Yeah,I also meet the problem for using FollowJointTrajectory to control robot!!!

alexvannobel, 2021-06-04:
Hi @flymaxty and @elevenjiang1 ,

We just released firmware v2.3.0, which contains the bugfix for the Pre-Conmputed Trajectories memory leak which caused this problem with MoveIt.

I'll close this issue since the related bug is now fixed.

You can always open a new issue if you encounter a problem.
Best,
Alex

closed 2021-06-04 by alexvannobel