Kinovarobotics/ros_kortex #133: How to use robot control topic in gazebo?

liyucheng251, 2020-11-16:
I tried to use the following command in gazebo:

`rostopic pub /my_gen3/in/cartesian_velocity kortex_driver/TwistCommand "reference_frame: 0
twist: {linear_x: 0.0, linear_y: 0.0, linear_z: 0.05, angular_x: 0.0, angular_y: 0.0,
angular_z: 0.0}
duration: 0"`

But it didn't work, Is speed control not supported in gazebo?

Thanks

alexvannobel, 2020-11-23:
Hi @liyucheng251 ,

The twist commands have actually been added as a POC to the simulation tools, but they have to be uncommented first in the code (and the package rebuilt) to be tried out. The reason is because the simulated twist commands were not stable enough at the moment they were developed so we chose not to include them in ros_kortex upon normal cloning of the repo. We added the POC implementation in the code though, so anyone who wants to enhance it and stabilie it can start somewhere.

You can have a look at [this](https://github.com/Kinovarobotics/ros_kortex/tree/kinetic-devel/kortex_examples#the-ways-to-use-a-kortex-arm-with-ros) (there is a paragraph on SendTwistCommand in simulation). The code you have to uncomment is [this](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/src/non-generated/driver/kortex_arm_driver.cpp#L677-L678) and the POC implementation is [here](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp#L1325-L1535). I also described the problem a bit more thoroughly [in my PR](https://github.com/Kinovarobotics/ros_kortex/pull/109#issue-448953223) when merging this.

I'll close this issue since this is technically not supported but an imperfect implementation is given for people to try out stuff.
Hope this helps,
Best,
Alex

closed 2020-11-23 by alexvannobel

liyucheng251, 2020-11-24:
> Hi @liyucheng251 ,
>
> The twist commands have actually been added as a POC to the simulation tools, but they have to be uncommented first in the code (and the package rebuilt) to be tried out. The [...]
Thanks!