Kinovarobotics/ros_kortex #134: Gen3 is not moving

Hadimohandes, 2020-11-20:
Hi 

we have bought a Gen3 Kinova arm from your company at SFU.
so far, I am able to use the gazebo simulator and send commands to the robot in simulation. 
I am now trying to connect to the real hardware. after I run the kortex driver node, I can see all the rostopics are generated. 
I can also go to the web application and change the angular values and meanwhile see the change in the following: 
rostopic echo /my_gen3/joint_states
but when I send commands like joint trajectory commands or run the default examples (moveit_example.launch) you have put in the examples' directory,  I get an error and the robot does not move.
I have attached the error file from the driver node terminal. 
Has anyone faced the same problem? 

[error.txt](https://github.com/Kinovarobotics/ros_kortex/files/5575353/error.txt)

alexvannobel, 2020-11-23:
Hi @Hadimohandes ,

I have a couple questions to try and find the problem:
1. With what command do you launch the example(s)? Did you only try the MoveIt example?
2. Did you modify the examples? If yes, can you provide the code?
3. What is the output of the `kortex_driver` node after you started it?
4. Do you see the robot in RViz and is it in the same position as the real robot?

Cheers,
Alex

Hadimohandes, 2020-11-24:
Hi Alexander,

Thanks for your reply.

I assume Santiago Moya has sent you the chain of emails and also the outputs.

Here are the answers to your questions:

1: yes, moveit example

2: I have not modified the example

3: I will attach again here

4: yes, I see the robot in Rviz the same the real robot's configurations

Thanks for your help

Best,

Hadi

roslaunch kortex_examples moveit_example.launch 
... logging to /home/hadi/.ros/log/2369a16c-2dc8-11eb-a199-509a4c09a04b/roslaunch-ensc-ramp-20-14399.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.2.200:39973/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.16

NODES
  /my_gen3/
    moveit_example (kortex_examples/example_move_it_trajectories.py)

ROS_MASTER_URI=http://192.168.2.200:11311

process[my_gen3/moveit_example-1]: started with pid [14416]
Failed to import pyassimp, see https://github.com/ros-planning/moveit/issues/86 for more info
[ INFO] [1606162461.465265598]: Loading robot model 'gen3'...
[ INFO] [1606162461.465300392]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1606162461.572966520]: Loading robot model 'gen3'...
[ INFO] [1606162461.572989867]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1606162461.637464960]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[ INFO] [1606162462.732799384]: Ready to take commands for planning group arm.
[INFO] [1606162462.734057]: Initializing node in namespace /my_gen3/
[INFO] [1606162462.735076]: Reaching Named Target Vertical...
[INFO] [1606162462.735309]: Going to named target vertical
[ INFO] [1606162464.677015806]: ABORTED: Solution found but controller failed during execution
[INFO] [1606162464.679848]: Reaching Joint Angles...
[INFO] [1606162464.857227]: Printing current joint positions before movement :
[INFO] [1606162464.857738]: 0.0387808727311
[INFO] [1606162464.858100]: 2.42918188198
[INFO] [1606162464.858588]: 3.07684508222
[INFO] [1606162464.858960]: 2.63320958256
[INFO] [1606162464.859464]: 0.126631187627
[INFO] [1606162464.859900]: 0.10376750026
[INFO] [1606162464.860302]: 1.24778990714
[ INFO] [1606162466.610325572]: ABORTED: Solution found but controller failed during execution
[INFO] [1606162466.632162]: Printing current joint positions after movement :
[INFO] [1606162466.632804]: 0.0387808727311
[INFO] [1606162466.633105]: 2.42918188198
[INFO] [1606162466.633396]: 3.07684508222
[INFO] [1606162466.633683]: 2.63320958256
[... 49 lines cut ...]
[ INFO] [1605308545.802732372]: Execution request received
[ INFO] [1605308545.829950327]: New goal received.
[ INFO] [1605308546.851376771]: Joint Trajectory Goal is accepted.
[ INFO] [1605308547.325012861]: State changed from IDLE to PRE_PROCESSING_PENDING

[ERROR] [1605308561.308763588]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 15.480000 seconds). Stopping trajectory.
[ INFO] [1605308561.308855235]: Cancelling execution for gen3_joint_trajectory_controller
[ INFO] [1605308561.308955986]: Completed trajectory execution with status TIMED_OUT ...
[ INFO] [1605308561.310226257]: Execution completed: TIMED_OUT
[ INFO] [1605308561.524812652]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1605308561.525007749]: Planning attempt 1 of at most 1
[ INFO] [1605308561.525580235]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1605308561.525814361]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1605308561.537822126]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1605308561.537871845]: Solution found in 0.012122 seconds
[ INFO] [1605308561.549758867]: SimpleSetup: Path simplification took 0.011840 seconds and changed from 4 to 2 states
[ INFO] [1605308561.559915362]: Interpolated time exceeds original trajectory (quitting), original: 7.07791 final interpolated time: 7.078
[ INFO] [1605308561.559969733]: Uniform sampling, resample duraction: 0.001 input traj. size: 62 output traj. size: 7079
[ INFO] [1605308561.654729167]: New goal received.
[ WARN] [1605308561.654852168]: There is already an active goal. It is being cancelled.
[ INFO] [1605308561.654880931]: Calling Stop on the robot.
[ INFO] [1605308562.702675136]: Joint Trajectory Goal is accepted.
[ INFO] [1605308563.119128034]: State changed from PRE_PROCESSING_PENDING to PRE_PROCESSING_PENDING

[ERROR] [1605308577.809896327]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 16.156000 seconds). Stopping trajectory.

alexvannobel, 2020-11-25:
Hi @Hadimohandes ,

This is a most bizarre error. It seems like the arm either starts validation of the trajectory or never receives the call, but the action server goes in the corresponding state.

What is the starting position of the arm when sending it the MoveIt trajectory? Can you go in the WebApp, bring it to Home position and then try this again? I don't see why it would be related to starting position but it's worth trying.

Also, can you show the output of the `kortex_driver` node (before even running the example) when it starts? It's worth checking for errors in this log too.

Best,
Alex

Hadimohandes, 2020-12-01:
Hi Alex,

I did as you suggested (moved the robot to the home position) and it's working fine now.

but isn't there any way to do this without homing the robot?

we haven't faced this issue in simulation.

I will add the output of kortex driver in my next msg.

Thanks

Best,

Hadi

alexvannobel, 2020-12-07:
Hi @Hadimohandes ,

I think the arm actually complains that the starting position it is in is out of bounds for a PreComputed trajectory (which MoveIt uses).

A workaround would be to run a custom node which checks for this specific actuator's position upon starting the `kortex_driver`, and send a joint command to make it "in bounds" (e.g. if actuator's command is 150.01 and needs to be under 150.00, send a 149.9 command to make sure the actuator is ready to start a MoveIt trajectory upon boot). I had in mind something like [this](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_examples/src/full_arm/example_full_arm_movement.cpp#L184-L214).

I'll keep this issue open because I might add this kind of check on startup for next release, but this won't be added in the short term.

Best,

Alex

JaeYoungLee79, 2021-12-01:
I empathy for first question. 
We spend much time to search and to solve for error with running examples.
Just we can control robot. where is Control Example?
1. Desire X,Y,X : waypoint example is run
2. Desire Joint : ????
3. or Advanced is Current Control : ????

That's all. 
Unfortunately, kortex example are mixed. Code does not have enough explanation.
In Korea, new job position emerging for kinova control application.

felixmaisonneuve, 2021-12-01:
I forgot to close this issue, but it was fixed with #184 and #185. Now, if the arm is out of bounds when launching the driver, each joint will move to the closest point "in bounds". 

@JaeYoungLee79, I am not sure your comment is related to the original issue. 
We have plenty of examples that show how to move the arm in different scenarios. Please open a new issue and I will help you there, but there is no point continuing on this thread.

closed 2021-12-01 by felixmaisonneuve