Kinovarobotics/ros_kortex #135: kortex_driver/Base_JointSpeeds invalid msg_arg type

jaredlawson, 2020-11-25:
I'm trying to publish joint speeds on the topic /my_gen3/in/joint_velocities which are type kortex_driver/Base_JointSpeeds.

Below is my rostopic pub command and the error message. Please let me know what I could be doing wrong.

jaredlawson@polar-bear:~/kinova_ws$ rostopic pub /my_gen3/in/joint_velocity kortex_driver/Base_JointSpeeds "{joint_speeds : [1.0,0.0,0.0,0.0,0.0,0.0,0.0], duration: 2}"
Traceback (most recent call last):
  File "/opt/ros/melodic/bin/rostopic", line 35, in <module>
    rostopic.rostopicmain()
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostopic/__init__.py", line 2140, in rostopicmain
    _rostopic_cmd_pub(argv)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostopic/__init__.py", line 1812, in _rostopic_cmd_pub
    argv_publish(pub, msg_class, pub_args, rate, options.once, options.verbose, substitute_keywords=options.substitute_keywords)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostopic/__init__.py", line 1836, in argv_publish
    publish_message(pub, msg_class, pub_args, rate, once, verbose=verbose, substitute_keywords=substitute_keywords)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostopic/__init__.py", line 1684, in publish_message
    _fillMessageArgs(msg, pub_args)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rostopic/__init__.py", line 1722, in _fillMessageArgs
    genpy.message.fill_message_args(msg, pub_args, keys=keys)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genpy/message.py", line 606, in fill_message_args
    _fill_message_args(msg, msg_args[0], keys, '')
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genpy/message.py", line 551, in _fill_message_args
    _fill_val(msg, f, v, keys, prefix)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genpy/message.py", line 521, in _fill_val
    _fill_message_args(inner_msg, el, keys, prefix)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genpy/message.py", line 565, in _fill_message_args
    raise ValueError('invalid msg_args type: %s' % str(msg_args))
ValueError: invalid msg_args type: 1.0

alexvannobel, 2020-11-25:
Hi @jaredlawson ,

By looking at the [msg file for Base_JointSpeeds](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/msg/generated/base/Base_JointSpeeds.msg#L2), I see that `joint_speeds` is not an array of floats but rather an array of `JointSpeed` objects, so your syntax (an array of floats) won't work, as indicated by the `ValueError: invalid msg_args type`.

From the `rostopic pub` command line, I believe the way to send custom objects would be something like that (and I read it [here](https://answers.ros.org/question/269695/how-to-rostopic-pub-msg-with-array-members/) too):
```
rostopic pub /my_gen3/in/joint_velocity kortex_driver/Base_JointSpeeds "{joint_speeds : [{joint_identifier: 0, value: 1.0},{joint_identifier: 1, value: 0.0},{joint_identifier: 2, value: 0.0},{joint_identifier: 3, value: 0.0},{joint_identifier: 4, value: 0.0},{joint_identifier: 5, value: 0.0},{joint_identifier: 6, value: 0.0}], duration: 2}"
```

I am pretty sure `joint_identifier` is 0-based, but try starting at 1 if this does not work.

Tell me if this works,
Best,
Alex

felixmaisonneuve, 2021-06-11:
Hi @jaredlawson , Do you have any update on this? Is your issue resolved? I'll wait a few days than I'll close the issue if I have no answer.

closed 2021-07-26 by felixmaisonneuve