Kinovarobotics/ros_kortex #136: Is 'example_cartesian_poses_with_notifications.py' work correctly?

haraisao, 2020-12-10:
Hi, we have bought a Kinova Gen3 recently.
Then we trid to execute kinova_example/example_cartesian_poses_with_notifications.py, 
but the robot stopped in a strange posture.

The posture of the 'pause1' is theta_x=-57.6, theta_y=91.1, theta_z=2.3, but a posture of the result  seemes to be theta_x=0, theta_y=91.1, theta_z=59.9.

Is the result is correct?
I tried changing the posture value in various ways, but we fail ed to move a roll angle of postures.

Please let me know the more detail information to use '/base/execute_action' service.

Thank you.

alexvannobel, 2020-12-11:
Hi @haraisao ,

Last time I checked, yes this example worked correctly.

It is supposed to:
1. bring the arm to Home position
2. make a Cartesian movement to reach a given pose
3. reach a second pose slightly lower
4. reach a third pose slightly lower than the second

I have a couple of questions to better understand what you ran : 
- Are you up-to-date with the latest commit of the `kinetic-devel` branch?
- What is the firmware version of your arm? You can see it in the Upgrade page in the WebApp.
- What is the exact command line you used to launch the `kortex_driver`?
- What is the exact command line you used to launch `example_cartesian_poses_with_notifications.py`?
- Is your Gen3 7DOF or 6DOF?
- Did you modify the example before it failed or did you run it as is? If you modified it, what pose did you try?

Best regards,
Alex

haraisao, 2020-12-14:
Thank you for your prompt reply.

 - Are you up-to-date with the latest commit of the kinetic-devel branchtry
 --  yes. I tried the latest version of the kinetic-devel branch on the github.com

 - What is the firmware version of your arm? You can see it in the Upgrade page in the WebApp.
 -- The firmware version of our robot are followings:
     Web App version: 2.8.0-r33, Web Client API version: 2.2.0-r27, Firmware Bundle: 2.2.0-r51, Firmware versions: Armbase: 2.2.0-r58, Actuator 0-6 and Interconnect:2.2.0-15, Vision:2.2.0-r22

  - Did you modify the example before it failed or did you run it as is? If you modified it, what pose did you try?
  --  First I tried the original version, but the orietation of the end-effector seems to be incorrect (the roll angle seems to be zero)
     Second I tried to change pose 2 and pose 3 as followings:
      pose2: my_constrained_pose.target_pose.theta_x = 0
             my_constrained_pose.target_pose.theta_z = 59.9

      pose3: my_constrained_pose.target_pose.theta_x = -59.9
             my_constrained_pose.target_pose.theta_z = 0

      The result is no movement occured in pose 2 and pose 3.
     

  - When it fails, what is displayed in both the driver and the example consoles?
  --  No error message desplyed.

If this example is correct, please show how to move roll angle of end-effector? 

Best regards.

alexvannobel, 2020-12-18:
Hi @haraisao ,

Thanks for the precise answers. I tested your exact command lines and there was in fact a typo in the example : the subscription to action notifications was made after the "go to home" command, which made it wait forever.

I just merged a PR that fixed it for me. Don't hesitate to reopen the issue if you still experience problems.

Best regards,

Alex

PR #138 (2020-12-18): Change notification subscription to be done before home
Fixes #136
closed 2020-12-18 by alexvannobel