Kinovarobotics/ros_kortex #137: admitance mode in ROS for Gen3 arm

Hadimohandes, 2020-12-16:
Hi 

I am Hadi from RAMP lab at sfu. 
we have bought a Gen3 Kinova arm from your company at SFU.
I have some questions: 
1- Is there any way to put the robot in admittance mode in ROS and do trajectory planning (pos control)? 
2- if the answer to above is yes, How does the arm differentiate between the external force and it's own force?
3- How can we put the robot in force or torque control mode in ROS? 
4- what is the gear ratio in admittance mode? 
Thanks

alexvannobel, 2021-01-13:
Hi @Hadimohandes ,

1. You can use the [dedicated service to put the arm in admittance](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/srv/generated/base/SetAdmittance.srv). For trajectory planning, you can use MoveIt to plan and execute the trajectories.
2. The arm has a gravity model and knows what "it's own force" is at any moment.
3. You can only do torque control in low-level mode, which we don't support in ROS because it cannot really respect a 1KHz frequency. You can have a look at [the torque control C++ example](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/108-Gen3_torque_control/01-torque_control_cyclic.cpp) to see how we do it only with the API. There is also a [Python example](https://github.com/Kinovarobotics/kortex/blob/master/api_python/examples/108-Gen3_torque_control/01-torque_control_cyclic.py).
4. The actuators' gear ratio is a mechanical property that does not change depending on the mode it is in. For Gen3 it is 100:1.

Hope this helps,
Best regards,
Alex

Hadimohandes, 2021-01-28:
Hi Alex, 

Thanks for your answer. 
It helped a lot. 

I am still wondering: 
what is in the low level control of these admittance modes? ( I want to know how this mode is done )
Our final goal is to do a object handover task and we want to use this feature of the robot (compliance mode) to comply with the receiver. 
Thanks so much 

Best, 
Hadi

alexvannobel, 2021-02-04:
Hi @Hadimohandes ,

I don't quite understand your question. Do you mean you want to know how such a mode is implemented?

The admittance mode in our control library essentially takes the torque measures of the actuators as input, calculates the efforts applied on the arm and transfers that into position commands.

I don't think I can give much more details about the mode.

Hope that helps,
Best,
Alex

Hadimohandes, 2021-02-04:
Hi Alexander, 

Thanks. it makes more sense now. 
I am wondering, since we want to put our own gripper (SDH) on the robot,  if there is any way to add the weight of the gripper to the gravity model so that in the admittance, it would be considered a part of the robot, not an external force. 
That's why I needed to know more about this admittance mode. 
I appreciate your help and time
Thanks 

Best, 
Hadi

alexvannobel, 2021-02-04:
Hi @Hadimohandes ,

You can do it in the WebApp in Configurations -> Robot -> Arm -> Tool

You can input there the tool mass center, mass and transform. This tab also contains a configuration page for the Gravity Vector if you are to mount your robot  with a non-default gravity vector.

Hope this helps,
Alex

Hadimohandes, 2021-02-04:
Hi Alex,

Thanks. I will try it.
So if I change it in the web app, will it change when I run the robot through ROS?

alexvannobel, 2021-02-04:
Yes it will and it will also be saved in the robot's database so it will persist reboots.

felixmaisonneuve, 2021-06-11:
Hi @Hadimohandes, I assume your issue has been resolved, so I will close it.

If you still have a problem, you can always reopen it.

closed 2021-06-11 by felixmaisonneuve