Kinovarobotics/ros_kortex #139: A problem with conanfile.py

wyh96, 2020-12-28:
Hello, I encountered an error with catkin make.
[image]
I checked conanfile.py. It hadn't been touched. 
The ros package worked well last week. Could you please help me to solve the problem?

polthergeist, 2021-01-05:
Hi @wyh96 , do you still have that problem? On my side it is working well.

alexvannobel, 2021-01-05:
I also don't have the problem.
I looks like a connection problem to me : either our Artifactory server was momentarily down or your connection was faulty.

wyh96, 2021-01-06:
Hello, I used backup files and it works.

polthergeist, 2021-01-06:
Ok, sounds good. I will close the issue but if you have any other problem don't hesitate to open a new one.

polthergeist, 2021-01-06:
Probably a connection problem.

closed 2021-01-06 by polthergeist