Kinovarobotics/ros_kortex #143: End-effector serial resource

alextoind, 2021-01-20:
Hello there, I'd like to know if it is possible on Kinova Gen 3 robots to expose the RS485 serial communication on the end-effector pins, to the user like a normal `/dev/ttyX` serial resource.

I've not found anything close to this in the documentation, but I might have miss something.

Thank your for your assistance

RaduCorcodel, 2021-02-12:
I don't think that functionality is exposed. My team was interested in that too to have access to the force control that Robotiq grippers all have, but Kinova did not include in the ros driver, even after we bugged them about it a few times. In the manual on page 65 you can get the pinout of the gripper connector. Once you're connected, the robot issues the 'activate' command at startup, and then sends packets over Modbus RTU, altough it's probably not what you want. We ended up buying an external controller for the gripper because Kinova never responded to our request to include force control into the API. Which can't be that difficult since the "force" parameter needs to be populated anyway when sending the command to the gripper. So instead of sending it a constant 100 probably, why not add it as a parameter just like position and velocity?!

felixmaisonneuve, 2021-06-17:
Hi @alextoind,

The Kortex API allows you to make a bridge (UART, GPIO or I2C) with the expansion connector pinout in the interconnect (the wrist) to access the serial communication to the end-effector. If that is something you are interested in, I would recommend looking at the examples ([UART](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/103-Gen3_uart_bridge/01-uart_bridge.cpp), [GPIO](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/104-Gen3_gpio_bridge/01-gpio_bridge.cpp), [I2C](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/105-Gen3_i2c_bridge/01-i2c_bridge.cpp)) provided with our Kortex API.
All of those could be done using our ros_kortex API with the services [here](https://github.com/Kinovarobotics/ros_kortex/tree/kinetic-devel/kortex_driver/srv/generated/interconnect_config) and [EnableBridge](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/srv/generated/base/EnableBridge.srv)/[DisableBridge](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_driver/srv/generated/base/DisableBridge.srv) for the UART communication. Everything should be almost identical from the examples. 

As @RaduCorcodel mentionned, Robotiq grippers are connected via the spring-loaded connector and our API does not allow direct access to this connector, therefore you cannot use force control of the gripper.

Hope this helps,
Felix

tkelestemur, 2021-07-16:
@RaduCorcodel It's not related to this specific issue, but after realizing that the gripper force setting can't be accessed via the ROS API, we realized that it can be accessed through the low-level cortex API. This line shows how to change the max force for the position controller, however, I'm still not sure if the low-level API allows direct force control.

felixmaisonneuve, 2021-07-16:
Hi @tkelestemur,

It is impossible to access the gripper force setting, even with the low-level API. There is a field available, but it doesn't do anything, this is a bug and it should be removed. In other words, it is not possible in any case to control the gripper force.

I am not sure what line you are talking about :  
> This line shows how to change ...

Regards,
Felix

tkelestemur, 2021-07-16:
Hey @felixmaisonneuve, sorry I forgot to put the line I was mentioning. I was talking about this [line](https://github.com/Kinovarobotics/kortex/blob/master/api_python/examples/107-Gripper_low_level_command/01-gripper_low_level_command.py#L149). When we change the `self.motorcmd.force` value, the max force limit for the position controller changes.

felixmaisonneuve, 2021-07-16:
Yep, exactly. This line does nothing, whatever value you put.
As per the [documentation](https://github.com/Kinovarobotics/kortex/blob/master/api_python/doc/markdown/messages/GripperCyclic/MotorCommand.md#), this is deprecated and unused. There is no way to control the gripper force.

Sorry about that

closed 2021-07-26 by felixmaisonneuve