Kinovarobotics/ros_kortex #144: Action Abort Notification

devesh1508, 2021-02-03:
We were using the '/base/play_joint_trajectory' service trying to send joint commands to Kinova Gen3. In an experiment, I had to press the emergency stop button to abort the movement. After that, I restarted the robot and all the nodes. But up on running the same code, I still get the following message [INFO] [1612381368.639386]: Sending the joint angle position...
[INFO] [1612381368.669128]: Received ACTION_ABORT notification
 and the robot aborts the movement.
Surprisingly, this code works from another account without any problem. Please suggest.

polthergeist, 2021-02-03:
Hi @devesh1508 ,
By any chances, are you connected to the robot's web application while you are executing your service?

felixmaisonneuve, 2021-06-11:
Hi @devesh1508, do you have any feedback? Do you still have an issue?

closed 2021-07-26 by felixmaisonneuve