Kinovarobotics/ros_kortex #146: Adding a gripper other than 2F85 or 2F140

RaduCorcodel, 2021-02-12:
We are currently using two grippers with the Kinova arm, a 2F85 and a Hand-E. Is there a way to include the Hand-E option in the Web app? The communication is identical, just the limits differ somewhat. I already modified our robot_descriptions to load a Hand-E gripper when the robot thinks a 2F85 is connected but we would like to have the descriptions separate because we are using the 2F85 as well. We can't do that unless in the web app we specify that something other than the 2F85 and 2F140 is connected. Do you have any recommendations on how to do that? The robot_description part we can handle ourselves no problem.

felixmaisonneuve, 2021-06-10:
Hi @RaduCorcodel,
If I understand corerctly, you want to add another configuration of gripper using the Web App.
It is possible as mentioned in [this comment](https://github.com/Kinovarobotics/ros_kortex/issues/137#issuecomment-773515277).

>You can do it in the WebApp in Configurations -> Robot -> Arm -> Tool
>You can input there the tool mass center, mass and transform. This tab also contains a configuration page for the Gravity Vector if you are to mount your robot with a non-default gravity vector.

Also, this will be saved in the database so it persist reboots.

Hope this helps,
Felix

felixmaisonneuve, 2021-08-20:
Since this thread seems to be inactive, I will close it. 

If you still encounter this issue, comment / re-open it or create a new issue.

Regards,
Felix

closed 2021-08-20 by felixmaisonneuve