Kinovarobotics/ros_kortex #147: gen3 ignores cartesian trajectory

adamkrawczyk, 2021-02-15:
Hi, 

I'm facing a strange behavior 

In one node I try to send sometimes angular pose or cartesian pose, the robot always follows angular commands but linear sometimes are ignored. 

Linear are all the times ignored when gripper got problem closing to desired pose (ex: send to close to 0.95 but it only made 0.91) then next cartesian move is ignored. 

In other situations, it happens randomly? 

I'm using a modified example.

Best,
Adam

felixmaisonneuve, 2021-06-16:
Hi @adamkrawczyk,

Can you give an example of a command that is ignored and a example of a gripper that doesn't close to desired position?
A new release has been merge recently, can you try this new 2.3 version. 
If the problem still occurs, I would appreciate to see what modifications you made, the commands you use to launch your example and maybe the output of those commands. 

In what case does the gripper stop without reaching the desired position?

Best,
Felix

adamkrawczyk, 2021-06-17:
Hi,

Thank you for your response, but unfortunately, I no longer have this robot - got only for demo.

Best,
Adam

closed 2021-06-17 by felixmaisonneuve