Kinovarobotics/ros_kortex #148: CartesianSpeed is not correspond!!

elevenjiang1, 2021-02-19:
When using kortex_examples/example_full_arm_movement.py,I find that the example_send_cartesian_pose function can not execute a correspond CartesianSpeed

My code just let the robot move repeatedly(move forward 0.3m and move backward 0.3m) and pose_speed.translation change from 0.1 to 1

0.1->3.284s
0.3->1.810s
0.5->1.774s
0.7->1.767s
1.0->1.767s

Meanwhile,I also use xbox to control robot,and set webApp Linear Speed of 50cm/s,the robot move really really fast.
This brother me so so so much!!

```
feedback = rospy.wait_for_message("/" + self.robot_name + "/base_feedback", BaseCyclic_Feedback)

req = PlayCartesianTrajectoryRequest()
req.input.target_pose.x = feedback.base.commanded_tool_pose_x+MOTION_X_ADD
req.input.target_pose.y = feedback.base.commanded_tool_pose_y
req.input.target_pose.z = feedback.base.commanded_tool_pose_z 
req.input.target_pose.theta_x = feedback.base.commanded_tool_pose_theta_x
req.input.target_pose.theta_y = feedback.base.commanded_tool_pose_theta_y
req.input.target_pose.theta_z = feedback.base.commanded_tool_pose_theta_z

pose_speed = CartesianSpeed()
pose_speed.translation = 0.1
pose_speed.orientation = 90

# The constraint is a one_of in Protobuf. The one_of concept does not exist in ROS
# To specify a one_of, create it and put it in the appropriate list of the oneof_type member of the ROS object : 
req.input.constraint.oneof_type.speed.append(pose_speed)

# Call the service
rospy.loginfo("Sending the robot to the cartesian pose...")
try:
    motion_begin_time=time.time()
    self.play_cartesian_trajectory(req)
except rospy.ServiceException:
    rospy.logerr("Failed to call PlayCartesianTrajectory")
    return False
else:
    motion_result=self.wait_for_action_end_or_abort()
    print("Robot motion time take:{}".format(time.time()-motion_begin_time))
    return motion_result
```

elevenjiang1, 2021-02-21:
After test ros and kortex api of playCartesianTrajectory(kortex api may be action.reach_pose),I find that the speed are same.It may not cause by the robot planning method,or speed limit in robot itself but not in webapp.
Now I can pub cartesian speed to /my_gens/in/cartesian_velocity to make robot move fast,but it is hard to let robot arrive to a specify pose.
However,it is really not good for user!

I think it is basic function for a user can let robot move to a specify pose with a desire speed at least.
Of course,the FollowJointTrajectory of ROS also has problem!,it will died for a while #132 
Hoping it can be solved soon...

felixmaisonneuve, 2021-06-16:
Hi @elevenjiang1,

I am not sure what is your problem here. Your pose_speed seems to be working correctly. 
Times are not linear because of the acceleration and deceleration of your robot so it can stop at the correct point. 
The acceleration is also why it is difficult to use cartesian_velocity to get to a particular point. 
Can you be more precise on what the issue is?

Also, the issue you mentionned with FollowJointTrajectory was a memory leak that has been fixed in the recent update.

Best,
Felix

elevenjiang1, 2021-06-17:
Thank you for reply.
I have already asked customer service for help.They said that in order to ensure robot move accuracy,robot can not move very fast in "play_cartesian_trajectory" but can move fast in "SendTwistCommand".
Now I have to use "SendTwistCommand" function to control robot to target pose and monitor robot end pose,when end pose reach target pose,then to tell it to stop.
To some extend,I solve this problem~

closed 2021-06-17 by felixmaisonneuve