Kinovarobotics/ros_kortex #149: Action Enc Notification for robotiq_2f_85

devesh1508, 2021-03-04:
We are trying to use the wait_for_action_end_or_abort() function (see here https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_examples/src/full_arm/example_cartesian_poses_with_notifications.py) with /base/send_gripper_command service. However, it seems that this never returns anything, and the python code is blocked indefinitely. However, if we dont use this and the gripper moves without getting to the actual position, then we see very bizarre behavior (like the robot will skip the next command in the python script). Does this command - wait_for_action_end_or_abort() work with the gripper?

pchang17, 2021-05-29:
I had the same issue. My solution for now is to use `rospy.sleep()` after the gripper command. Using `rospy.sleep()` instead of `time.sleep()` as in the example code prevents the robot from skipping the next command.

felixmaisonneuve, 2021-06-16:
Hi @devesh1508,

Like you mentionned, `wait_for_action_end_abort()` won't work for gripper actions because grippers actions don't send notifications (apparently) as I found out [here](https://github.com/Kinovarobotics/ros_kortex/blob/eef6758e901d408df5656bb16edc5bd00534d6f9/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp#L735). I will do some more research about that, sorry for the inconvenience.

In the mean time, I would do as @pchang17 said and use `rospy.sleep()`. I am not sure why `time.sleep()` would skip the next action. 

Can you provide an example where an action would be skipped? 

Best,
Felix

devesh1508, 2021-06-17:
I see. Okay. We will provide a new example in the next few days and update
you. We have encountered some problems with the action_abort_notification()
while sending twist commands to the robot.
On another note, the new firmware is supposed to implement the trajectory
control in a smooth fashion. But we implemented the trajectory_control
while sending a trajectory in 6-D for the end-effector. The robot stops at
every waypoint on the specified trajectory. I can open a new issue for this
problem but can you confirm this behavior for trajectory_control ?

felixmaisonneuve, 2021-06-17:
Great, I will wait for your info about the twist commands.

For the trajectory control, you can add multiple waypoints to make your robot follow a trajectory like in the example [here](https://github.com/Kinovarobotics/ros_kortex/blob/eef6758e901d408df5656bb16edc5bd00534d6f9/kortex_examples/src/full_arm/example_full_arm_movement.cpp#L268). The robot will then move to each waypoint, stop, then move to the next point, like you saw. 

To have a smooth trajectory, you need to modify the `blending_radius` attribute in your waypoint. This is a radius around the point that defines a "zone" where the robot can go instead having to go exactly to the point. In other words, instead of moving to the exact point, stopping, and moving to your next waypoint, your robot will only need to move "somewhere" in that sphere defined by the radius, so the movement in smooth. 

You can define the `blending_radius` or set `trajectory.use_optimal_blending = true;` (an other example where `use_optimal_blending` is used [here](https://github.com/Kinovarobotics/ros_kortex/blob/eef6758e901d408df5656bb16edc5bd00534d6f9/kortex_examples/src/full_arm/example_waypoint_action_client.cpp#L152)) to set an optimal blending_radius for each point in your tajectory.

I will link [this issue](https://github.com/Kinovarobotics/ros_kortex/issues/156) that is also related to blending_radius. 

Best,
Felix

felixmaisonneuve, 2021-07-26:
Hi @devesh1508,

Did you have time to try using `time.sleep()` instead of `wait_for_action_end_abort()` for gripper actions. 
Also, have you tried changing blending radius? Does the arm still stop at every waypoint?

Looking forward to your response,
Felix

aalmrad, 2024-10-16:
Closing because of inactivity. Please reply to this thread to reopen.

Best,
Abed

closed 2024-10-16 by aalmrad