Kinovarobotics/ros_kortex #152: How to read velocity of endpoint of Gen3 in the Cartesian space

JDchen-Jaden, 2021-04-07:
After I run `roslaunch kortex_driver kortex_driver.launch` I did not find any `rosmsg` and `rostopic` that can be used to subscribe the velocity of the end effector in the Cartesian space. Does anyone know how to publish the velocity in the Cartesian space? Thank!

felixmaisonneuve, 2021-06-16:
Hi @JDchen-Jaden,

kortex_driver.launch can take mulptiple parameters define [here](https://github.com/Kinovarobotics/ros_kortex/tree/kinetic-devel/kortex_driver#usage). 
Are you sure your robot uses all the default options?
Does your robot have an IP address of 192.168.1.10?

Best,
Felix

felixmaisonneuve, 2021-08-20:
Since this thread seems to be inactive, I will close it. 

If you still encounter this issue, comment / re-open it or create a new issue.

Regards,
Felix

closed 2021-08-20 by felixmaisonneuve