Kinovarobotics/ros_kortex #153: Cannot grasp an object in a Gazebo simulation

goncalo-carrola-silva, 2021-04-16:
Hello there, in my work I need to grasp a certain box using KINOVA Gen3 Lite in a Gazebo simulation. If I understand it correctly, this gripper has no proximity sensors on the fingers, and the only function available is to set a certain position and velocity of the fingers. Knowing that the box can have several sizes, how do I know how much value I have to give to the position of the fingers in order to grab the box and be able to lift it? If I give the maximum value, the box "slips" due to excess force. Or is there another way that is escaping me? 

I am new at this, so I am sorry if this issue is too obvious...

felixmaisonneuve, 2021-06-14:
Hi @goncalo-carrola-silva,

I think what you are looking for is a reading of the force applied by the end-effector.
Unfortunately, this is not supported for our robots. 

In your case, I do not think there is a way to grab a box if you don't know its size to set the correct position.

Sorry for the inconvenience,
Felix

goncalo-carrola-silva, 2021-06-15:
Hi @felixmaisonneuve,

Appreciate the reply. I think I found a way that partially solves the problem. For each step, the manipulator will close 2% of its gripper until the finger position does not change more than 0.5% from the previous step, which means that effectively the fingers will already be in contact with the box and able to grasp and lift the object. 

I realize it might not be the most elegant way to do it, but it works.

Compliments,
Gonçalo

felixmaisonneuve, 2021-06-15:
I am glad you found a workaround.

I will close this issue

Best,
Felix

closed 2021-06-15 by felixmaisonneuve