Kinovarobotics/ros_kortex #157: v2.3.0 gen3 fail to execute trajectory plan

fi6, 2021-06-14:
Tried to checkout back to v2.2.2 and everything works. In v2.3.0, running the following command for driver and the example will cause problem when executing trajectory.

`roslaunch kortex_driver  kortex_driver.launch dof:=7 gripper:=robotiq_2f_85`
`roslaunch kortex_examples moveit_example.launch`

[ INFO] [1623688518.391460883]: Joint Trajectory Goal is accepted.
[ERROR] [1623688518.392914886]: Kortex exception while validating the trajectory
[ERROR] [1623688518.392937621]: Error code: ERROR_PROTOCOL_SERVER

[ERROR] [1623688518.392950537]: Error sub code: UNSUPPORTED_METHOD

[ERROR] [1623688518.392962369]: Error description: Server protocol error, Error sub type=UNSUPPORTED_METHOD => <srv: 2, fct: 235, msgType: 3>
description: Non-parsable payload received from device.

[ INFO] [1623688518.393067387]: Controller gen3_joint_trajectory_controller successfully finished
[ WARN] [1623688518.393131953]: Controller handle gen3_joint_trajectory_controller reports status ABORTED
[ INFO] [1623688518.393168973]: Completed trajectory execution with status ABORTED ...
[ERROR] [1623688519.443046068]: Kortex exception while sending the trajectory
[ERROR] [1623688519.443182644]: Error code: ERROR_PROTOCOL_SERVER

[ERROR] [1623688519.443252275]: Error sub code: UNSUPPORTED_METHOD

[ERROR] [1623688519.443334416]: Error description: Server protocol error, Error sub type=UNSUPPORTED_METHOD => <srv: 2, fct: 226, msgType: 3>
description: Non-parsable payload received from device.

[ERROR] [1623688519.443424012]: To transition to an aborted state, the goal must be in a preempting or active state, it is currently in state: 4

alexvannobel, 2021-06-14:
Hi @fi6 ,

Did you update your arm's firmware to release 2.3.0 firmware?

Best,
Alex

fi6, 2021-07-17:
Updated and it works! However everytime I restart the arm the actions in web app will be cleaned(and I'll have to import them again), any workaround for this?

felixmaisonneuve, 2021-07-19:
Hi @fi6,

Restarting the arm should reset the actions within the webapp unless you also do a factory reset.
Do you have more information about this? Can you provide the steps you are doing when creating actions and reseting the arm to make sure everything is done correctly? 

Regards,
Felix

fi6, 2021-07-20:
Hi @felixmaisonneuve,

I'm currently using Kinova Gen3 with Robotiq 2f 85, firmware 2.3.0. I haven't test if normally shutting down the arm will cause this problem or not, but every time when I stopped the arm with emergency button and then restart it, all the saved actions were lost.

felixmaisonneuve, 2021-07-26:
Hi @fi6,

Have you tried to shut down the arm normally to see if the problem persist?
Also, it is possible to trigger a factory reset (that will wipe all your actions) if you hold down the button for too long. 
Could that be your case? 

Regards,
Felix

felixmaisonneuve, 2021-08-20:
Hi @fi6,

Have you managed to figure out your issue?
Do you still need support? I will close this issue in a couple of weeks if I don't get an answer.

Regards,
Felix

felixmaisonneuve, 2022-01-04:
Since I did not receive any answer, I will close this issue.
Feel free to comment if you still experience the same problem and I will re-open it. 

Best,
Felix

closed 2022-01-04 by felixmaisonneuve

fi6, 2022-01-18:
> Since I did not receive any answer, I will close this issue. Feel free to comment if you still experience the same problem and I will re-open it.
> 
> Best, Felix

Hi Felix,

Sorry for the late reply. I'm currently using another kinova gen3 and the profiles persists across reboots. Still unable to figure out what happens but it works for me now. Thanks!

Best,
Yida