Kinovarobotics/ros_kortex #158: Error while running MoveIt! example: ImportError: undefined symbol

Jaroan, 2021-06-15:
When I try to run the example to use the Python MoveIt! API to move the arm based on [#moveit-example](https://github.com/Kinovarobotics/ros_kortex/tree/kinetic-devel/kortex_examples#moveit-example) 
I get the following error:  `ImportError: /opt/ros/melodic/lib/libmoveit_planning_scene.so.1.0.8: undefined symbol: _ZN8octomath6Pose6DC1ERKS0_`
The full log is below. 
Any help to fix the error is appreciated.
 Thank you!

```
:~/catkin_ws$  roslaunch kortex_examples moveit_example.launch
... logging to /home/jasprism/.ros/log/16f1a5ce-cde1-11eb-8708-ac120395af92/roslaunch-jasprism-31206.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jasprism:33917/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES
  /my_gen3/
    moveit_example (kortex_examples/example_move_it_trajectories.py)

ROS_MASTER_URI=http://localhost:11311

process[my_gen3/moveit_example-1]: started with pid [31221]
Traceback (most recent call last):
  File "/home/jasprism/catkin_ws/src/ros_kortex/kortex_examples/src/move_it/example_move_it_trajectories.py", line 46, in <module>
    import moveit_commander
  File "/opt/ros/melodic/lib/python2.7/dist-packages/moveit_commander/__init__.py", line 4, in <module>
    from .move_group import *
  File "/opt/ros/melodic/lib/python2.7/dist-packages/moveit_commander/move_group.py", line 51, in <module>
    from moveit_ros_planning_interface import _moveit_move_group_interface
ImportError: /opt/ros/melodic/lib/libmoveit_planning_scene.so.1.0.8: undefined symbol: _ZN8octomath6Pose6DC1ERKS0_
[my_gen3/moveit_example-1] process has died [pid 31221, exit code 1, cmd bash -c sleep 0; $0 $@ /home/jasprism/catkin_ws/src/ros_kortex/kortex_examples/src/move_it/example_move_it_trajectories.py __name:=moveit_example __log:=/home/jasprism/.ros/log/16f1a5ce-cde1-11eb-8708-ac120395af92/my_gen3-moveit_example-1.log].
log file: /home/jasprism/.ros/log/16f1a5ce-cde1-11eb-8708-ac120395af92/my_gen3-moveit_example-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

felixmaisonneuve, 2021-06-15:
Hi @Jaroan,

The easiest way to fix this I think would be to delete moveit from your machine
```
sudo apt purge ros-melodic-moveit-*
```
then reinstall everything from your catkin workspace using 
```
 rosdep install --from-paths src --ignore-src -y
```
and finally rebuild your catkin workspace
```
rm -rf build/ devel/
catkin_make
```

Let me know if this works,
Felix

Jaroan, 2021-06-15:
Hi @felixmaisonneuve  ,
Thank you for the help. I did as told above and now I get this error
```
process[my_gen3/moveit_example-1]: started with pid [10712]
[ERROR] [1623775414.455499854]: Robot model parameter not found! Did you remap 'robot_description'?
RobotInterfacePython: invalid robot model
[ERROR] [1623775414.462146]: The example encountered an error.
[my_gen3/moveit_example-1] process has finished cleanly
log file: /home/jasprism/.ros/log/7b573724-cdf8-11eb-8708-ac120395af92/my_gen3-moveit_example-1*.log
all processes on machine have died, roslaunch will exit
```
Do I need to check the model description?
Thanks again!

Jaroan, 2021-06-15:
Seems like this launch command enables moveit automatically?
```
$ roslaunch kortex_gazebo spawn_kortex_robot.launch
... logging to /home/jasprism/.ros/log/3735e32a-cdfc-11eb-8708-ac120395af92/roslaunch-jasprism-23267.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://jasprism:33573/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /my_gen3/gen3_joint_trajectory_controller/action_monitor_rate: 25
 * /my_gen3/gen3_joint_trajectory_controller/constraints/goal_time: 1.0
 * /my_gen3/gen3_joint_trajectory_controller/constraints/stopped_velocity_tolerance: 0.5
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_1/d: 2.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_1/i: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_1/i_clamp_max: 100.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_1/i_clamp_min: -100.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_1/p: 3000.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_2/d: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_2/i: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_2/i_clamp_max: 5.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_2/i_clamp_min: -5.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_2/p: 50000.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_3/d: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_3/i: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_3/i_clamp_max: 1.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_3/i_clamp_min: -1.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_3/p: 3000.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_4/d: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_4/i: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_4/i_clamp_max: 1.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_4/i_clamp_min: -1.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_4/p: 50000.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_5/d: 0.2
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_5/i: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_5/i_clamp_max: 1.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_5/i_clamp_min: -1.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_5/p: 750.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_6/d: 1.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_6/i: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_6/i_clamp_max: 1.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_6/i_clamp_min: -1.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_6/p: 5000.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_7/d: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_7/i: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_7/i_clamp_max: 0.1
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_7/i_clamp_min: -0.1
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_7/p: 100.0
 * /my_gen3/gen3_joint_trajectory_controller/joints: ['joint_1', 'join...
 * /my_gen3/gen3_joint_trajectory_controller/state_publish_rate: 25
 * /my_gen3/gen3_joint_trajectory_controller/stop_trajectory_duration: 1.0
 * /my_gen3/gen3_joint_trajectory_controller/type: effort_controller...
 * /my_gen3/home_the_arm/robot_name: my_gen3
 * /my_gen3/joint_1_position_controller/joint: joint_1
 * /my_gen3/joint_1_position_controller/pid/d: 2.0
 * /my_gen3/joint_1_position_controller/pid/i: 0.0
 * /my_gen3/joint_1_position_controller/pid/p: 3000.0
 * /my_gen3/joint_1_position_controller/type: effort_controller...
 * /my_gen3/joint_2_position_controller/joint: joint_2
 * /my_gen3/joint_2_position_controller/pid/d: 0.0
 * /my_gen3/joint_2_position_controller/pid/i: 0.0
 * /my_gen3/joint_2_position_controller/pid/p: 50000.0
 * /my_gen3/joint_2_position_controller/type: effort_controller...
[... 335 lines cut ...]
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1623776882.767085466, 0.350000000]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1623776882.767101447, 0.350000000]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1623776883.743568663, 1.324000000]: Ready to take commands for planning group arm.
[ INFO] [1623776884.045062159, 1.625000000]: The Kortex driver has been initialized correctly!
[INFO] [1623776884.056093, 1.636000]: Sending the robot home...
[INFO] [1623776886.750733, 4.326000]: Received ACTION_END notification
[INFO] [1623776886.753440, 4.329000]: The Gazebo initialization executed without fail.
[my_gen3/home_the_arm-10] process has finished cleanly
log file: /home/jasprism/.ros/log/3735e32a-cdfc-11eb-8708-ac120395af92/my_gen3-home_the_arm-10*.log

```

felixmaisonneuve, 2021-06-16:
Are you sure all your packages are updated?
ROS GPG Key recently expired, when you do `sudo apt-get update`, are you sure ros is updated correctly?

alexvannobel, 2021-06-16:
Hi @Jaroan ,

Yes, the [spawn_kortex_robot](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_gazebo/launch/spawn_kortex_robot.launch) script launches the MoveIt server on its own, precisely [here](https://github.com/Kinovarobotics/ros_kortex/blob/kinetic-devel/kortex_gazebo/launch/spawn_kortex_robot.launch#L140-L151).

Do you still have the error with your MoveIt Python client?

Best,
Alex

Jaroan, 2021-06-16:
Hi @felixmaisonneuve 
I still get the same error when I try `$ roslaunch kortex_examples moveit_example.launch`
`ImportError: /opt/ros/melodic/lib/libmoveit_planning_scene.so.1.0.8: undefined symbol: _ZN8octomath6Pose6DC1ERKS0_`

However the ` $ roslaunch kortex_gazebo spawn_kortex_robot.launch` enabled the Gazebo simulator and the MoveIt RViz Motion Planning plugin similar to these [tutorials ](http://docs.ros.org/en/melodic/api/moveit_tutorials/html/doc/quickstart_in_rviz/quickstart_in_rviz_tutorial.html).

For 
>  ROS GPG Key recently expired, when you do `sudo apt-get update`, are you sure ros is updated correctly?

I had reinstalled the ros_kortex package and repeated the full installation to get it working with ` $ roslaunch kortex_gazebo spawn_kortex_robot.launch`  again

felixmaisonneuve, 2021-06-16:
Hi @Jaroan,

Your `$ roslaunch kortex_gazebo spawn_kortex_robot.launch` is working as expected. The problem comes from `$ roslaunch kortex_examples moveit_example.launch`, which raise an error with `/opt/ros/melodic/lib/libmoveit_planning_scene.so.1.0.8`.

This is comes from a librairy in ROS packages. I don't think the problem comes from our ros_kortex package directly, which is why I want to make sure that your ROS packages are correctly updated. I have found [this](https://answers.ros.org/question/236573/undefined-symbol-in-rviz/) which is a similar issue to yours. It says this might be caused by a partial update and recommand using `sudo apt-get update && sudo apt-get upgrade`. (He actually recommends `dist-upgrade` but I have trust issues with those "smart" system removing packages, even if I think it would not break anything)

You already seem to have the correct version, this is why I recommeded to remove moveit packages with `sudo apt purge ros-melodic-moveit-*` then reinstalling it, not just our ros_kortex package. 

If updating doesn't work, you might have a better shot on the moveit [github](https://github.com/ros-planning/moveit). I saw a couple similar issues, but they use ROS Neotic on Ubuntu 20.04, which we do not support yet. For example [this one](https://github.com/ros-planning/moveit/issues/1850) which has been answered.

Like I mentionned, I don't think this comes from our ros_kortex package as I am unable to reproduce your specific error with Ubuntu 18.04 and ROS Melodic

Best,
Felix

Jaroan, 2021-06-17:
Thank you so much @felixmaisonneuve .
> You already seem to have the correct version, this is why I recommeded to remove moveit packages with `sudo apt purge ros-melodic-moveit-*` then reinstalling it, not just our ros_kortex package.

As suggested before, I've done this and the result remained the same unfortunately. I'll shift the issue to the moveit github issues.
Thank you once again!

closed 2021-06-17 by Jaroan

Jaroan, 2022-06-05:
Hi @fishbotics, unfortunately, I was not able to resolve it