Kinovarobotics/ros_kortex #161: Compiler error on base_services.cpp.p

jabbershort, 2021-06-21:
I've been trying to install kortex on a number of machines to try and get a working configuration. The closest I've gotten so far is using Ubuntu 18.04 & ROS Melodic on an x86 machine, but I'm still hitting an error compiling base_services.cpp.o.

I've copied my error below, has anyone else hit something like this?
[image]
[image]

jabbershort, 2021-06-21:
Also, is there any way to install/make on a raspberry pi? I tried setting the target architecture, but couldn't get any working configuration.

felixmaisonneuve, 2021-06-21:
Hi @jabbershort,

What version of cmake do you have on your machine? 
Your conan profile looks fine to me. Did you get anything unusual while installing all the required dependencies with `rosdep install --from-paths src --ignore-src -y`?

For your raspberry pi, we have three different ARM configurations available : `Artik 710`, `IMX6` and `Nvidia Jetson`. You could try using any of them to see if it works. I am not an expert in this, and I don't really know how this works, but I looked in the code and it seems to me the only difference is the settings used in `ros_kortex/kortex_driver/CmakeLists.txt`. Although the readme says it is for ARMv8, I think IMX6 is only for ARMv7. Also, in `ros_kortex/kortex_driver/CmakeLists.txt`, every option has `kortex_api_cpp:arch=armv7`, so you might try to change it to `kortex_api_cpp:arch=armv8` and see if it works. But again, I don't really know what this does and how it works. 

Looking at the [commit](https://github.com/Kinovarobotics/ros_kortex/commit/6b2f985d883f415ab53c1a19866ea6e6e309a3f5) when this feature was added, the only changes made were in `ros_kortex/kortex_driver/CmakeLists.txt`, so you might as well add your own configuration or change one of them to match your Pi configuration. 

Something like this maybe? 
```
elseif("${CONAN_TARGET_PLATFORM}" STREQUAL "pi")
    conan_cmake_run(CONANFILE conanfile.py
                    UPDATE
                    BASIC_SETUP CMAKE_TARGETS
                    NO_OUTPUT_DIRS
                    SETTINGS kortex_api_cpp:arch=armv8 kortex_api_cpp:compiler=gcc kortex_api_cpp:compiler.version=7 kortex_api_cpp:compiler.libcxx=libstdc++11
                    OPTIONS kortex_api_cpp:target=pi)
```

Regards,
Felix

jabbershort, 2021-06-22:
Hi Felix,

Thanks so much for your quick response. 
I had cmake 3.10.2, and tried upgrading to 3.20, but got the same error. I didn't get anything unusual during the install step. 

On the raspberry pi front, I've had some more success. I found that one of my pip packages was out of date and when upgrading it, I got all the way to 98%. It has hit an error with a missing file, so I'm just going to try downloading the source directly, not using conan. I will report back.

felixmaisonneuve, 2021-06-22:
I suppose you started from a clean catkin workspace and tried to run `catkin_make` without changing any file in ros_kortex? 
Did you get any error when you ran `rosdep install --from-paths src --ignore-src -y`?

Also, do you have other errors in your output after running `catkin_make`, or the error from your first screenshot is the only one?

jabbershort, 2021-06-23:
Hi Felix,

To make the jump in progress, I updated all my pip packages, and re-installed gazebo, which seemed to get me a step closer. Each time I've started with a fresh workspace, re-cloning the git and re-installing before I make. Still nothing odd coming back from rosdep install step.

So I did a bit more testing last night, on x86, I still get the same error, thats the only text I get. I'm going to have a go today going back to a completely fresh install of ubuntu and starting again.

On Raspberry PI, I'm still getting up to 98% and it seems to fail on a different part each time, I've had it fail building on srv __init__.py, example_full_arm_movement_cpp, example_actuator_configuration_cpp, example_waypoint_action_client_cpp. It seems that the actual problem is occuring a few builds above and it always seems to be missing ActuatorConfig.pb.h & Common.pb.h.

I haven't had a chance to try downloading the source and building myself without Conan, but will try and give that a go today. 

[image]
[image]

alexvannobel, 2021-06-23:
Hi @jabbershort ,

Can you provide the full log for the failed build? Not finding this file is basically due to not finding Kortex API, which Conan fetches at the start of the build. We would see if there is an error with Conan at the start of the build then.

Thanks,
Alex

jabbershort, 2021-06-28:
Hi both, 

Thanks for your input so far and sorry it took me so long to get back to you. I've been doing quite a bit of testing and managed to get things working. Firstly, I did try using Docker instead for x86 build and it worked flawlessly. When compiling on low-powered / single threaded processers I did find you needed to use the -j1 flag for catkin_make (i.e 'catkin_make -j1') to prevent hanging.

For raspberry pi, I've tried a number of things, but finally got one working solution. I'm going to carry on trying with Docker as well, as this would be a better end goal for my platform, but to get it working I had to do the following:

Starting with completely fresh ubuntu 18.04.5 server build for Raspberry Pi (mine is a 3b+):
- Fully update the system
- Ensure there is at least 2GB of swap available.
- Install ROS melodic, but only the base (i.e. `sudo apt-get install ros-melodic-ros-base`)
- Initalise ROS and follow the inital steps to setup the ros_kortex build (up to, and including, cloning the repo)
- Copy in the source code provided for linux arm builds, I used the jetson flavour, but I haven't tried any of the others.
- Then run `rosdep install --from-paths src --ignore-src -y`
- Finally running catkin_make without using conan and using the -j1 flag i.e. `catkin_make -j1 -DUSE_CONAN=OFF`
- I found the build took around 5-6 hours as well, with some steps taking upwards of 30 mins.
- Then sourcing it like normal. 

I'll try and give an update when/if I get it working on Docker on rpi.

felixmaisonneuve, 2021-06-28:
That is good news. I'm glad you got it working using Docker. Still weird it didn't using the x86 build. 

If I remember corectly, the `-jX` flag specifies how many threads are going to be used to build. In a single threaded processor, it makes sense you need to use `-j1`. This also explains why it took so much time (I think we use `-j8` by default), this is a painfully long time. 

I will keep this thread open so you can inform us on your future progress with your Raspbarry Pi

Best,
Felix

jabbershort, 2021-06-29:
Yeah, doing it one thread at a time is really snails pace compared to running it on a multi-core processor! I haven't tested it, but it might be that you can run it on `-j2` instead, but at least there is a stable solution. 

I've got it running on Docker (on RPi) now as well. Following exactly the same logic, but ensuring the host system has the swap space available (as the containers just share that).

Happy for you to close now, I think this comment and my previous give a good enough solution to anyone else who comes up against this problem.

closed 2021-06-29 by felixmaisonneuve