Kinovarobotics/ros_kortex #163: kortex_arm_simulation crashes if the arm is mounted on a mobile robot

civerachb-cpr, 2021-06-24:
A common use case (for us anyway) is mounting the Gen3 or Gen3 Lite onto a mobile robot platform (e.g. Clearpath's Jackal, Dingo, or Husky).  Unfortunately, it appears to be impossible to run a complete simulation of the arm and mobile platform because of some issues inside kortex_arm_simulation.cpp:

1) The robot_description path is hard-coded.  When mounting the arm on the real robot, we modify the base robot's URDF to include the arm macro.  For example on a Dingo with a Gen3 Lite arm we would use https://github.com/dingo-cpr/dingo_manipulation/blob/main/dingo_kinova_description/urdf/dingo_gen3_lite_description.urdf.xacro to attach the arm to the robot.  This results in the robot description parameter being /robot_description, but https://github.com/Kinovarobotics/ros_kortex/blob/eef6758e901d408df5656bb16edc5bd00534d6f9/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp#L60 hard-codes the path to be `/<arm-name>/robot_description` which does not exist in this use-case, and there's no way to modify this path without needing to recompile the driver.

2) Even if we do change the `model.initParam` line in kortex_arm_simulator.cpp, the fact that this class _also_ tries to launch MoveIt and find planning groups causes the driver to crash anyway.  We don't _want_ to launch Moveit as part of the simulated driver; we just want to be able to get the joint states publishing correctly.  We can launch Moveit ourselves (e.g. using https://github.com/dingo-cpr/dingo_manipulation/blob/main/dingo_gen3_lite_moveit_config/launch/move_group.launch) -- again, there doesn't appear to be any way to disable auto-launching Moveit without making some major changes to the driver and recompiling it.

Is there a different node I should be launching to launch a simulated arm driver that _just_ launches the core driver with a configurable robot_description path and without automatically starting Moveit planning groups?  The launch file I'm using to start the arm is this:

```
<launch>
  <!-- Arm configuration -->
  <arg name="arm" default="$(optenv DINGO_ARM_MODEL gen3_lite)"/>
  <arg name="dof" default="$(optenv DINGO_ARM_DOF 6)"/>
  <arg name="vision" default="false"/>
  <arg name="use_hard_limits" default="false"/>
  <arg name="gripper" default="$(optenv DINGO_ARM_GRIPPER gen3_lite_2f)"/>

  <!-- Namespace -->
  <arg name="robot_name" default="$(optenv DINGO_ARM_PREFIX arm)"/>
  <arg name="prefix" default="$(optenv DINGO_ARM_PREFIX arm)_"/>

  <!-- Kortex API options -->
  <arg name="ip_address" default="$(optenv DINGO_ARM_HOST 192.168.131.40)"/>
  <arg name="username" default="admin"/>
  <arg name="password" default="admin"/>
  <arg name="cyclic_data_publish_rate" default="100"/> <!--Hz-->
  <arg name="api_rpc_timeout_ms" default="2000"/> <!--milliseconds-->
  <arg name="api_session_inactivity_timeout_ms" default="35000"/> <!--milliseconds-->
  <arg name="api_connection_inactivity_timeout_ms" default="20000"/> <!--milliseconds-->
  <arg name="sim" default="false" />

  <!-- Action server params -->
  <arg name="default_goal_time_tolerance" default="0.5"/> <!--seconds-->
  <arg name="default_goal_tolerance" default="0.005"/> <!--radians-->

  <group ns="$(arg robot_name)">
      <!-- Start the kortex_driver node -->
      <node name="$(arg robot_name)_driver" pkg="kortex_driver" type="kortex_arm_driver" output="screen"> <!--launch-prefix="gdb -ex run args"-->
          <param name="sim" value="$(arg sim)"/>
          <param name="ip_address" value="$(arg ip_address)"/>
          <param name="username" value="$(arg username)"/>
          <param name="password" value="$(arg password)"/>
          <param name="cyclic_data_publish_rate" value="$(arg cyclic_data_publish_rate)"/>
          <param name="api_rpc_timeout_ms" value="$(arg api_rpc_timeout_ms)"/>
          <param name="api_session_inactivity_timeout_ms" value="$(arg api_session_inactivity_timeout_ms)"/>
          <param name="api_connection_inactivity_timeout_ms" value="$(arg api_connection_inactivity_timeout_ms)"/>
          <param name="default_goal_time_tolerance" value="$(arg default_goal_time_tolerance)"/>
          <param name="default_goal_tolerance" value="$(arg default_goal_tolerance)"/>
          <param name="arm" value="$(arg arm)"/>
          <param name="gripper" value="$(arg gripper)"/>
          <param name="dof" value="$(arg dof)"/>
          <param name="use_hard_limits" value="$(arg use_hard_limits)"/>
          <param name="robot_name" value="$(arg robot_name)"/>
          <param name="prefix" value="$(arg prefix)"/>
          <rosparam command="load" file="$(find kortex_description)/arms/$(arg arm)/$(arg dof)dof/config/joint_limits.yaml" subst_value="true"/>
          <!-- If there is a gripper, load the active joint names for it -->
          <rosparam command="load" file="$(find kortex_description)/grippers/$(arg gripper)/config/joint_limits.yaml" unless="$(eval not arg('gripper'))" subst_value="true"/>
      </node>
  </group>

  <node name="joint_state_relay" type="relay" pkg="topic_tools" args="/$(arg robot_name)/base_feedback/joint_state /joint_states" />
</launch>
```
with the `sim` parameter set to `true`.  This same launch file works properly on a physical robot (with the `sim` parameter set to `false`).  My goal is to be able to have the simulation and real robot work as-similarly as possible.  If I need to use a different node than the `kortex_driver/kortex_arm_driver` instead, what node should I use instead?

felixmaisonneuve, 2021-06-25:
Hi @civerachb-cpr,

I am not sure I understand everything in your question, so I will try to answer it the best I can.

First, you seem to have trouble getting the robot_description. In kortex_arm_simulation.cpp, the robot_description is found using the hardcoded path `"/" + m_robot_name + "/robot_description"`. This is because in our launchfile (kortex_driver.launch) we have something like 
```
<param name="robot_description" command="$(find xacro)/xacro --inorder $(find kortex_description)/robots/$(arg arm).xacro dof:=$(arg dof) vision:=$(arg vision) sim:=false"
```
defined within the robot_name namespace (`<group ns="$(arg robot_name)">`). This will publish the robot_description in `"/m_robot_name/robot_description"`. 
I don't see where you publish your robot_description using your xacro file. I think you should have something like 
```
 <group ns="$(arg robot_name)">

      <!-- Load robot description -->
      <param name="robot_description" command="$(find xacro)/xacro ...

      <!-- Start the kortex_driver node -->
      <node name="$(arg robot_name)_driver" pkg="kortex_driver" type="kortex_arm_driver" output="screen"> <!--launch-prefix="gdb -ex run args"-->
          <param name="sim" value="$(arg sim)"/>
          <param name="ip_address" value="$(arg ip_address)"/>
...
```

Second, you mentionned kortex_arm_simulation.cpp is starting Moveit, which is not the case (or at least I don't see anything related to moveit in kortex_arm_simulation.cpp). 
In our launch file (kortex_driver.launch) however, we do have 2 arguments (`start_rviz` and `start_moveit`) that are set to `true` by default. These two will start the simulation by launching ...$(arm)_moveit_config/launch/move_group.launch. So, unless you set those to false when launching the driver (e.g. `roslaunch kortex_driver kortex_driver.launch start_rviz:=false start_moveit:=false`) it will automatically launch the simulation. 

I am not sure what you are trying to achieve with the `sim` argument.

civerachb-cpr, 2021-06-25:
Sorry, let me clarify:
I am using a Clearpath Dingo with a Gen3 Lite arm mounted to it.  We've previously done this successfully on a physical robot, using the process outlined here: http://www.clearpathrobotics.com/assets/guides/melodic/dingo/manipulation.html, and are now looking into supporting this entirely within Gazebo using a simulated robot.

Basically, we're using the Dingo as the main robot description, and adding the arm using the URDF Extras environment variable we support.  You can see the Dingo description here: https://github.com/dingo-cpr/dingo/blob/melodic-devel/dingo_description/urdf/dingo-d.urdf.xacro.

Specifically, at the end of the file (see: https://github.com/dingo-cpr/dingo/blob/06cdeee63212374608d0add4a780f49fca5d6ecc/dingo_description/urdf/dingo-d.urdf.xacro#L246) we can load a URDF stub to add additional links and joints to the base platform.  In our setup, that environment variable points to the location of [this file](https://github.com/dingo-cpr/dingo_manipulation/blob/main/dingo_kinova_description/urdf/dingo_gen3_lite_description.urdf.xacro) on-disk, which simply connects the Gen3 Lite xacro macro to one of the mounting points of the base platform.

The model, viewed in Rviz looks like this:

[image]

When we configure the Dingo's URDF as described above and launch the simulation the model loads correctly, but the arm is completely slack and clips through the robot model.  This, I assume, is because there is no ROS node running that is publishing the joint states of the arm joints.  Echoing `/tf` there's nothing publishing any of the `arm_*` joints, which is what I would expect in this situation; the base platform node for Dingo is running, but the Kinova driver is not.

[image]

My goal is to be able to get the kortex_driver running to control the arm inside Gazebo and have the joint states properly published & simulated.

If I launch the kortex_driver without setting the `sim` argument, it attempts to connect to a physical arm, which is obviously not what I want to happen in this case.  My assumption is that launching the driver in simulation mode should let me control the simulated arm.  But when I try launching the driver in sim mode with my simulated robot configured as-described above, the node crashes.

> I don't see where you publish your robot_description using your xacro file. I think you should have something like `<param name="robot_description" command="$(find xacro)/xacro ...`

The robot description is provided by the dingo_description package: https://github.com/dingo-cpr/dingo/blob/melodic-devel/dingo_description/launch/description.launch

When working with a physical arm mounted to a physical robot this is sufficient, and the robot works correctly.  But when I launch the kortex_driver with the `sim` flag set to true, I get an exception that the robot_description parameter is not found in /arm/robot_description.  But on the physical robot, under the same configuration conditions, the driver does not expect that parameter.

On the real robot, if we attempt to launch the kortex_driver using the standard launch file included in the kortex_driver package, the Dingo's robot_description parameter gets overwritten by Kinova's, and the robot model winds up in a broken state.  We want to use the complete robot_description parameter published by dingo_description, which appears to be possible with the non-simulated version of kortex_driver.

If I edit https://github.com/Kinovarobotics/ros_kortex/blob/eef6758e901d408df5656bb16edc5bd00534d6f9/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp#L60 to use `/robot_description`, which is the description parameter we use when working on the physical robot, I can successfully load the parameter, but then the node crashes when looking up the move groups used by moveit.

For reference, here's the output of `rosparam dump` when my Gazebo simulation is running
```
$ rosparam dump
arm:
  arm_driver:
    api_connection_inactivity_timeout_ms: 20000
    api_rpc_timeout_ms: 2000
    api_session_inactivity_timeout_ms: 35000
    arm: gen3_lite
    cyclic_data_publish_rate: 100
    default_goal_time_tolerance: 0.5
    default_goal_tolerance: 0.005
    dof: 6
    gripper: gen3_lite_2f
    gripper_joint_limits_max: [-0.09]
    gripper_joint_limits_min: [0.96]
    gripper_joint_names: [arm_right_finger_bottom_joint]
    ip_address: 192.168.131.40
    joint_names: [arm_joint_1, arm_joint_2, arm_joint_3, arm_joint_4, arm_joint_5,
      arm_joint_6]
    maximum_accelerations: [1.0, 0.5, 0.4, 1.0, 10.0, 10.0]
    maximum_velocities: [0.5, 0.5, 0.5, 0.5, 0.5, 0.5]
    password: admin
    prefix: arm_
    robot_name: arm
    sim: true
    use_hard_limits: false
    username: admin
  is_initialized: false
bluetooth_teleop:
  joy_node: {autorepeat_rate: 20, deadzone: 0.1}
  teleop_twist_joy: {axis_angular: 0, axis_linear: 1, enable_button: 4, enable_turbo_button: 5,
    scale_angular: 1.4, scale_linear: 0.4, scale_linear_turbo: 2.0}
dingo_joint_publisher: {publish_rate: 50, type: joint_state_controller/JointStateController}
dingo_velocity_controller:
  angular:
    z: {has_acceleration_limits: true, has_velocity_limits: true, max_acceleration: 2.0,
      max_velocity: 4.0}
  cmd_vel_timeout: 0.25
  enable_odom_tf: false
  k_l: 0.1
[... 345 lines cut ...]
  - {name: e_stop, priority: 255, timeout: 0.0, topic: e_stop}
  topics:
  - {name: joy, priority: 10, timeout: 0.5, topic: joy_teleop/cmd_vel}
[... cut ...] on the next line, appear to be part of the Moveit client that the simulator is launching.

At this point I'm not sure what my next logical step is.  I'd rather not go through kortex_arm_simulation.cpp and manually remove all of the Moveit client code and recompile.  But is that the correct course of action?  Or am I missing something and there's a different/better launch file I should be using to launch the Kortex driver simulation while preserving the Dingo URDF and robot_description parameter?

felixmaisonneuve, 2021-06-25:
Ok, I understand your problem, and I will try my best to help you.  

When you start the driver with `sim:=true`, it launches kortex_arm_simulation.cpp, which tries to read the robot description from the ros param /arm/robot_description and crashes because it is not there. This does not happen with a real arm because it communicates with the actual robot and [verify the configuration](https://github.com/Kinovarobotics/ros_kortex/blob/eef6758e901d408df5656bb16edc5bd00534d6f9/kortex_driver/src/non-generated/driver/kortex_arm_driver.cpp#L328) from there.

If you change `model.initParam("/" + m_robot_name + "/robot_description");` to `model.initParam("/robot_description");` it crashes when trying to create the default actions in moveit. I think this is not essential and you could simply try to comment the lines below (involving moveit) in kortex_arm_simulation.cpp (this includes line 149): 
https://github.com/Kinovarobotics/ros_kortex/blob/eef6758e901d408df5656bb16edc5bd00534d6f9/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp#L141-L149

This should allow you to finsh to initialization of the kortex_arm_driver.cpp. 
If it does, you should see `is_initialized: true` instead of `is_initialized: false` in your `rosparam dump`

civerachb-cpr, 2021-06-25:
Ok, thanks.  I'll try that out and let you know how it goes.

Assuming it does work I may submit a PR to add some additional ROS params to make this easier to configure without needing to recompile the driver.

civerachb-cpr, 2021-07-15:
I made the changes you suggested, and it seems like I'm making progress in the right direction.  However, I'm noticing more inconsistencies between the real arm driver and the simulated one that seem to be causing further issues.

After starting the kortex_driver with the simulation flag set to `true` the arm remains slack in Gazebo.  It appears nothing is actually publishing the state of the arm joints.

When using the real arm, the joint states are published on `/${robot_name}/base_feedback/joint_state`, but no such topic seems to exist when I run in simulation.  This has a trickle-down effect, as normally I would use a topic_tools/relay to republish the base_feedback/joint_state data to /joint_states, which is subscribed to by the main robot_state_publisher.

The only topics published by the arm driver when I run in simulation are:

```
Node [/arm/arm_driver]
Publications: 
 * /arm/action_topic [kortex_driver/ActionNotification]
 * /arm/arm_gen3_lite_joint_trajectory_controller/follow_joint_trajectory/cancel [actionlib_msgs/GoalID]
 * /arm/arm_gen3_lite_joint_trajectory_controller/follow_joint_trajectory/goal [control_msgs/FollowJointTrajectoryActionGoal]
 * /arm/arm_joint_1_position_controller/command [std_msgs/Float64]
 * /arm/arm_joint_2_position_controller/command [std_msgs/Float64]
 * /arm/arm_joint_3_position_controller/command [std_msgs/Float64]
 * /arm/arm_joint_4_position_controller/command [std_msgs/Float64]
 * /arm/arm_joint_5_position_controller/command [std_msgs/Float64]
 * /arm/arm_joint_6_position_controller/command [std_msgs/Float64]
 * /arm/arm_state_topic [kortex_driver/ArmStateNotification]
 * /arm/configuration_change_topic [kortex_driver/ConfigurationChangeNotification]
 * /arm/control_configuration_topic [kortex_driver/ControlConfigurationNotification]
 * /arm/control_mode_topic [kortex_driver/ControlModeNotification]
 * /arm/controller_topic [kortex_driver/ControllerNotification]
 * /arm/factory_topic [kortex_driver/FactoryNotification]
 * /arm/kortex_error [kortex_driver/KortexError]
 * /arm/mapping_info_topic [kortex_driver/MappingInfoNotification]
 * /arm/network_topic [kortex_driver/NetworkNotification]
 * /arm/operating_mode_topic [kortex_driver/OperatingModeNotification]
 * /arm/protection_zone_topic [kortex_driver/ProtectionZoneNotification]
 * /arm/robot_event_topic [kortex_driver/RobotEventNotification]
 * /arm/safety_topic [kortex_driver/SafetyNotification]
 * /arm/sequence_info_topic [kortex_driver/SequenceInfoNotification]
 * /arm/servoing_mode_topic [kortex_driver/ServoingModeNotification]
 * /arm/user_topic [kortex_driver/UserNotification]
 * /rosout [rosgraph_msgs/Log]
```

What topic should the joint states of the simulated arm be published on?

felixmaisonneuve, 2021-07-15:
Hi @civerachb-cpr,

I noticed you are using a prefix when launching the driver. I think this might be the problem here. 
I was able to get similar output to yours while using a prefix. 

I recently (last week) realised that something is not working when using a prefix and I haven't looked into it yet. I noticed this while looking at [this PR](https://github.com/Kinovarobotics/ros_kortex/pull/162).
This is a major inconvenience and I don't know if it was working at some point (probably) and broke or if it never worked in the first place. This is probably something I will look into soon

If it is possible, you might try to not use any prefix. 
In my case, I am able to get the complete list of topics when I am not using any prefix, but, when I use the `prefix` argument, I get basically the same topics for the driver node and the simulation is not working.

Hopefully it is possible for you to use `prefix=""` in your launch file and it will work.
Sorry for the inconveniance,
Felix

felixmaisonneuve, 2021-07-19:
After further inspection, I don't think it is related to the `prefix` argument. 
I tried again and `/arm/base_feedback` and `/arm/base_feedback/joint_state` are published correctly, so I must have missed them last week. 

Those two topics are supposed to be published at the end of the driver initialization 
https://github.com/Kinovarobotics/ros_kortex/blob/ee05de829fb4806ae1b36826feea76993ebae19c/kortex_driver/src/non-generated/driver/kortex_arm_driver.cpp#L47-L56

Can you make sure everything has been initialized correctly with the param `is_initialized` that should be `true` (for example by doing a `rosparam dump`). 

If everything looks good, this might indicate the problem is coming form `GetFeedback()` : 
https://github.com/Kinovarobotics/ros_kortex/blob/ee05de829fb4806ae1b36826feea76993ebae19c/kortex_driver/src/non-generated/driver/kortex_arm_driver.cpp#L617

Also, looking at your output, I get 2 topics related to the gripper that you do not have : `/arm/arm_gen3_lite_2f_gripper_controller/gripper_cmd/cancel` and `/arm/arm_gen3_lite_2f_gripper_controller/gripper_cmd/goal`.
Can you make sure you set `gripper:=gen3_lite_2f` correctly in your launch file? 

Finally, I see that one of your topic is `/arm/control_mode_topic [kortex_driver/ControlModeNotification]`. `ControlModeNotification` has been renamed to `Base_ControlModeNotification` in Kortex [release 2.3](https://github.com/Kinovarobotics/ros_kortex/releases/tag/v2.3.0). 
I do not think it will fix your problem, but it might be a good idea to update your repo if it not too much trouble

felixmaisonneuve, 2021-08-20:
Hi @civerachb-cpr,

Do you have any update on this? 
I will wait for a couple of weeks and close this issue if I don't get an answer.

Regards,
Felix

civerachb-cpr, 2021-08-23:
Hi, sorry for the delay replying to this.  I'll get straight into answering your questions:

1- Yes, the gripper parameter is set to `gen3_lite_2f`
2- I tried setting `prefix=""` in the launch file, but that didn't help; when I tried launching the arm driver I got the following error:
```
...
[ INFO] [1629726247.428386962, 1654.489000000]: -------------------------------------------------
[ INFO] [1629726247.437425354, 1654.496000000]: Simulating arm with following characteristics:
[ INFO] [1629726247.437461693, 1654.496000000]: Arm type : gen3_lite
[ INFO] [1629726247.437475851, 1654.496000000]: Gripper type : gen3_lite_2f
[ INFO] [1629726247.437487258, 1654.496000000]: Arm namespace : kinova_arm
[ INFO] [1629726247.437503305, 1654.496000000]: URDF prefix : 
[ERROR] [1629726247.441029520, 1654.498000000]: Failed to extract kinematic chain from parsed tree!
terminate called after throwing an instance of 'std::runtime_error'
  what():  Failed to extract kinematic chain from parsed tree!
```

I suspect this may be due to a mismatch in prefixes between the URDF and the driver launch file?  In order to keep the joint names unique across the robot we do use the `prefix` parameter of the `load_robot` URDF macro.

Launching with the prefix argument set normally appears to run without any errors, but `is_initialized` is still `false` in my rosparam dump output:

Roslaunch output:
```
SUMMARY
========

PARAMETERS
 * /kinova_arm/kinova_arm_driver/api_connection_inactivity_timeout_ms: 20000
 * /kinova_arm/kinova_arm_driver/api_rpc_timeout_ms: 2000
 * /kinova_arm/kinova_arm_driver/api_session_inactivity_timeout_ms: 35000
 * /kinova_arm/kinova_arm_driver/arm: gen3_lite
 * /kinova_arm/kinova_arm_driver/cyclic_data_publish_rate: 40
 * /kinova_arm/kinova_arm_driver/default_goal_time_tolerance: 0.5
 * /kinova_arm/kinova_arm_driver/default_goal_tolerance: 0.005
 * /kinova_arm/kinova_arm_driver/dof: 6
 * /kinova_arm/kinova_arm_driver/gripper: gen3_lite_2f
 * /kinova_arm/kinova_arm_driver/gripper_joint_limits_max: [-0.09]
 * /kinova_arm/kinova_arm_driver/gripper_joint_limits_min: [0.96]
 * /kinova_arm/kinova_arm_driver/gripper_joint_names: ['kinova_arm_righ...
 * /kinova_arm/kinova_arm_driver/ip_address: 192.168.1.10
 * /kinova_arm/kinova_arm_driver/joint_names: ['kinova_arm_join...
 * /kinova_arm/kinova_arm_driver/maximum_accelerations: [1.0, 0.5, 0.4, 1...
 * /kinova_arm/kinova_arm_driver/maximum_velocities: [0.5, 0.5, 0.5, 0...
 * /kinova_arm/kinova_arm_driver/password: admin
 * /kinova_arm/kinova_arm_driver/prefix: kinova_arm_
 * /kinova_arm/kinova_arm_driver/robot_name: kinova_arm
 * /kinova_arm/kinova_arm_driver/sim: True
 * /kinova_arm/kinova_arm_driver/use_hard_limits: False
 * /kinova_arm/kinova_arm_driver/username: admin
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES
  /kinova_arm/
    joint_state_relay (topic_tools/relay)
    kinova_arm_driver (kortex_driver/kortex_arm_driver)

ROS_MASTER_URI=http://localhost:11311

process[kinova_arm/kinova_arm_driver-1]: started with pid [3776]
process[kinova_arm/joint_state_relay-2]: started with pid [3777]
[ INFO] [1629726259.594322042]: -------------------------------------------------
[ INFO] [1629726259.594951547]: Initializing Kortex Driver's services...
[ INFO] [1629726261.745428358, 1665.215000000]: Kortex Driver's services initialized correctly.
[ INFO] [1629726261.745473183, 1665.215000000]: -------------------------------------------------
[ INFO] [1629726261.754778920, 1665.223000000]: Simulating arm with following characteristics:
[ INFO] [1629726261.754809371, 1665.223000000]: Arm type : gen3_lite
[ INFO] [1629726261.754824325, 1665.223000000]: Gripper type : gen3_lite_2f
[ INFO] [1629726261.754836299, 1665.223000000]: Arm namespace : kinova_arm
[ INFO] [1629726261.754848030, 1665.223000000]: URDF prefix : kinova_arm_
```

```
$ rosparam dump|grep is_initialized
  is_initialized: false
```

I do not see anything published on the `base_feedback/joint_state` nor `base_feedback` topics, likely because the initialization is failing.

I just pulled the latest `melodic-devel` branch this morning and tested again; same results.  Is there a different branch I should try that hasn't been merged yet?

felixmaisonneuve, 2021-08-23:
Hi @civerachb-cpr,

Did you delete some of the rows in your roslaunch output? I get way more stuff than you. I was wondering if you did it on purpose to make more readable. 

Also, your simulation driver does not seem to initialize correctly and I am not sure why. 
Do you get any error message in your roslaunch output?

Can you also give the output of `rostopic list`. I see the driver is looking for topic `/<robot_name>/joint_states` : 
https://github.com/Kinovarobotics/ros_kortex/blob/a334ce2a9fcbec12b27012f77183a81a75ec6e6e/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp#L158

If it is not there, I think it might explain why all your topics are empty.

To answer your question, there is no other branch you should try it on.

Regards,
Felix

yj-Tang, 2021-12-22:
@civerachb-cpr Hi, have you solved this problem? I am integrating Husky and Kinova Gen3 Lite in Gazebo, and have the same problem. Could you give any updates on how you solved them?

yj-Tang, 2022-01-03:
Hi @felixmaisonneuve, 
Thanks for the reply! I somehow have the kortex arm and the mobile robot simulated together. But the kortex driver is always died after I launched the simulation (```kinova_arm_driver-8] process has died [pid 1502665, exit code -11```), which confused me.  The output of the roslaunch is shown below:
```
.. logging to /home/yujie/.ros/log/be55b2e2-6ca6-11ec-9974-4d8c6a873cd5/roslaunch-ricks-Alienware-m15-R6-1502516.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
WARN: unrecognized 'param' tag in <include> tag
WARN: unrecognized 'param' tag in <include> tag
started roslaunch server http://ricks-Alienware-m15-R6:38451/

SUMMARY
========

PARAMETERS
 * /ekf_localization/base_link_frame: base_link
 * /ekf_localization/frequency: 50
 * /ekf_localization/imu0: imu/data
 * /ekf_localization/imu0_config: [False, False, Fa...
 * /ekf_localization/imu0_differential: True
 * /ekf_localization/imu0_queue_size: 10
 * /ekf_localization/imu0_remove_gravitational_acceleration: True
 * /ekf_localization/odom0: husky_velocity_co...
 * /ekf_localization/odom0_config: [False, False, Fa...
 * /ekf_localization/odom0_differential: False
 * /ekf_localization/odom0_queue_size: 10
 * /ekf_localization/odom_frame: odom
 * /ekf_localization/two_d_mode: True
 * /ekf_localization/world_frame: odom
 * /gazebo/enable_ros_network: True
 * /gazebo_ros_control/pid_gains/kinova_arm_left_finger_bottom_joint/d: 0.0
 * /gazebo_ros_control/pid_gains/kinova_arm_left_finger_bottom_joint/i: 0.0
 * /gazebo_ros_control/pid_gains/kinova_arm_left_finger_bottom_joint/p: 10.0
 * /gazebo_ros_control/pid_gains/kinova_arm_left_finger_tip_joint/d: 0.0
 * /gazebo_ros_control/pid_gains/kinova_arm_left_finger_tip_joint/i: 0.0
 * /gazebo_ros_control/pid_gains/kinova_arm_left_finger_tip_joint/p: 1.0
 * /gazebo_ros_control/pid_gains/kinova_arm_right_finger_bottom_joint/d: 0.0
 * /gazebo_ros_control/pid_gains/kinova_arm_right_finger_bottom_joint/i: 0.0
 * /gazebo_ros_control/pid_gains/kinova_arm_right_finger_bottom_joint/p: 10.0
 * /gazebo_ros_control/pid_gains/kinova_arm_right_finger_tip_joint/d: 0.0
 * /gazebo_ros_control/pid_gains/kinova_arm_right_finger_tip_joint/i: 0.0
 * /gazebo_ros_control/pid_gains/kinova_arm_right_finger_tip_joint/p: 1.0
 * /husky_joint_publisher/publish_rate: 50
 * /husky_joint_publisher/type: joint_state_contr...
 * /husky_velocity_controller/angular/z/has_acceleration_limits: True
 * /husky_velocity_controller/angular/z/has_velocity_limits: True
 * /husky_velocity_controller/angular/z/max_acceleration: 6.0
 * /husky_velocity_controller/angular/z/max_velocity: 2.0
 * /husky_velocity_controller/base_frame_id: base_link
 * /husky_velocity_controller/cmd_vel_timeout: 0.25
 * /husky_velocity_controller/enable_odom_tf: False
 * /husky_velocity_controller/estimate_velocity_from_position: False
 * /husky_velocity_controller/left_wheel: ['front_left_whee...
 * /husky_velocity_controller/linear/x/has_acceleration_limits: True
 * /husky_velocity_controller/linear/x/has_velocity_limits: True
 * /husky_velocity_controller/linear/x/max_acceleration: 3.0
 * /husky_velocity_controller/linear/x/max_velocity: 1.0
 * /husky_velocity_controller/pose_covariance_diagonal: [0.001, 0.001, 0....
 * /husky_velocity_controller/publish_rate: 50
 * /husky_velocity_controller/right_wheel: ['front_right_whe...
 * /husky_velocity_controller/twist_covariance_diagonal: [0.001, 0.001, 0....
 * /husky_velocity_controller/type: diff_drive_contro...
 * /husky_velocity_controller/velocity_rolling_window_size: 2
 * /husky_velocity_controller/wheel_radius_multiplier: 1.0
 * /husky_velocity_controller/wheel_separation_multiplier: 1.875
 * /joy_teleop/joy_node/autorepeat_rate: 20
 * /joy_teleop/joy_node/deadzone: 0.1
 * /joy_teleop/joy_node/dev: /dev/input/ps4
[... 629 lines cut ...]
/place/cancel
/place/feedback
/place/goal
/place/result
/place/status
/planning_scene
/planning_scene_world
/protection_zone_topic
/robot_event_topic
/rosout
/rosout_agg
/safety_topic
/sequence_info_topic
/servoing_mode_topic
/set_pose
/tf
/tf_static
/trajectory_execution_event
/twist_marker_server/cmd_vel
/twist_marker_server/feedback
/twist_marker_server/update
/twist_marker_server/update_full
/user_topic
/velodyne_points
```

felixmaisonneuve, 2022-01-04:
Looking at the `rosparam dump` output of @civerachb-cpr, the `robot_description` is not in the relative param directory (`/arm/arm_driver`), it is directly in the root param directory, so I think changing this line 
https://github.com/Kinovarobotics/ros_kortex/blob/037233668ee5610d62ece15159fe767259922803/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp#L61

to 
```
model.initParam("/robot_description");
```

might allow the simulation driver to find the `robot_description` so it will initialize correctly and possibly fix the issue.

civerachb-cpr, 2022-01-04:
Sorry about the late reply; just got back from vacation. No, I have not yet been able to fix this problem yet either.

yj-Tang, 2022-01-06:
> Looking at the `rosparam dump` output of @civerachb-cpr, the `robot_description` is not in the relative param directory (`/arm/arm_driver`), it is directly in the root param directory, so I think [...]

Hi, @felixmaisonneuve. Thanks for your quick reply. I have changed the scripts as you recommend. But it still makes no difference. The output of ```param``` is shown below
```
/camera_description
/ekf_localization/base_link_frame
/ekf_localization/frequency
/ekf_localization/imu0
/ekf_localization/imu0_config
/ekf_localization/imu0_differential
/ekf_localization/imu0_queue_size
/ekf_localization/imu0_remove_gravitational_acceleration
/ekf_localization/odom0
/ekf_localization/odom0_config
/ekf_localization/odom0_differential
/ekf_localization/odom0_queue_size
/ekf_localization/odom_frame
/ekf_localization/two_d_mode
/ekf_localization/world_frame
/gazebo/auto_disable_bodies
/gazebo/cfm
/gazebo/contact_max_correcting_vel
/gazebo/contact_surface_layer
/gazebo/enable_ros_network
/gazebo/erp
/gazebo/gravity_x
/gazebo/gravity_y
/gazebo/gravity_z
/gazebo/max_contacts
/gazebo/max_update_rate
/gazebo/sor_pgs_iters
/gazebo/sor_pgs_precon_iters
/gazebo/sor_pgs_rms_error_tol
/gazebo/sor_pgs_w
/gazebo/time_step
/gazebo_ros_control/pid_gains/kinova_arm_left_finger_bottom_joint/antiwindup
/gazebo_ros_control/pid_gains/kinova_arm_left_finger_bottom_joint/d
/gazebo_ros_control/pid_gains/kinova_arm_left_finger_bottom_joint/i
/gazebo_ros_control/pid_gains/kinova_arm_left_finger_bottom_joint/i_clamp_max
/gazebo_ros_control/pid_gains/kinova_arm_left_finger_bottom_joint/i_clamp_min
/gazebo_ros_control/pid_gains/kinova_arm_left_finger_bottom_joint/p
/gazebo_ros_control/pid_gains/kinova_arm_left_finger_tip_joint/antiwindup
/gazebo_ros_control/pid_gains/kinova_arm_left_finger_tip_joint/d
/gazebo_ros_control/pid_gains/kinova_arm_left_finger_tip_joint/i
/gazebo_ros_control/pid_gains/kinova_arm_left_finger_tip_joint/i_clamp_max
/gazebo_ros_control/pid_gains/kinova_arm_left_finger_tip_joint/i_clamp_min
/gazebo_ros_control/pid_gains/kinova_arm_left_finger_tip_joint/p
/gazebo_ros_control/pid_gains/kinova_arm_right_finger_bottom_joint/antiwindup
/gazebo_ros_control/pid_gains/kinova_arm_right_finger_bottom_joint/d
/gazebo_ros_control/pid_gains/kinova_arm_right_finger_bottom_joint/i
/gazebo_ros_control/pid_gains/kinova_arm_right_finger_bottom_joint/i_clamp_max
/gazebo_ros_control/pid_gains/kinova_arm_right_finger_bottom_joint/i_clamp_min
/gazebo_ros_control/pid_gains/kinova_arm_right_finger_bottom_joint/p
/gazebo_ros_control/pid_gains/kinova_arm_right_finger_tip_joint/antiwindup
/gazebo_ros_control/pid_gains/kinova_arm_right_finger_tip_joint/d
/gazebo_ros_control/pid_gains/kinova_arm_right_finger_tip_joint/i
/gazebo_ros_control/pid_gains/kinova_arm_right_finger_tip_joint/i_clamp_max
/gazebo_ros_control/pid_gains/kinova_arm_right_finger_tip_joint/i_clamp_min
/gazebo_ros_control/pid_gains/kinova_arm_right_finger_tip_joint/p
/husky_joint_publisher/publish_rate
/husky_joint_publisher/type
/husky_velocity_controller/angular/z/has_acceleration_limits
/husky_velocity_controller/angular/z/has_velocity_limits
/husky_velocity_controller/angular/z/max_acceleration
/husky_velocity_controller/angular/z/max_velocity
/husky_velocity_controller/base_frame_id
/husky_velocity_controller/cmd_vel_timeout
/husky_velocity_controller/enable_odom_tf
/husky_velocity_controller/estimate_velocity_from_position
/husky_velocity_controller/left_wheel
[... 344 lines cut ...]
/robot_description_planning/joint_limits/kinova_arm_left_finger_bottom_joint/has_acceleration_limits
/robot_description_planning/joint_limits/kinova_arm_left_finger_bottom_joint/has_velocity_limits
/robot_description_planning/joint_limits/kinova_arm_left_finger_bottom_joint/max_acceleration
/robot_description_planning/joint_limits/kinova_arm_left_finger_bottom_joint/max_velocity
/robot_description_planning/joint_limits/kinova_arm_left_finger_tip_joint/has_acceleration_limits
/robot_description_planning/joint_limits/kinova_arm_left_finger_tip_joint/has_velocity_limits
/robot_description_planning/joint_limits/kinova_arm_left_finger_tip_joint/max_acceleration
/robot_description_planning/joint_limits/kinova_arm_left_finger_tip_joint/max_velocity
/robot_description_planning/joint_limits/kinova_arm_right_finger_bottom_joint/has_acceleration_limits
/robot_description_planning/joint_limits/kinova_arm_right_finger_bottom_joint/has_velocity_limits
/robot_description_planning/joint_limits/kinova_arm_right_finger_bottom_joint/max_acceleration
/robot_description_planning/joint_limits/kinova_arm_right_finger_bottom_joint/max_velocity
/robot_description_planning/joint_limits/kinova_arm_right_finger_tip_joint/has_acceleration_limits
/robot_description_planning/joint_limits/kinova_arm_right_finger_tip_joint/has_velocity_limits
/robot_description_planning/joint_limits/kinova_arm_right_finger_tip_joint/max_acceleration
/robot_description_planning/joint_limits/kinova_arm_right_finger_tip_joint/max_velocity
/robot_description_semantic
/rosdistro
/roslaunch/uris/host_ricks_alienware_m15_r6__46821
/rosversion
/run_id
/twist_mux/locks
/twist_mux/topics
/use_sim_time
```

felixmaisonneuve, 2022-01-07:
At that point, I am pretty sure it gets stuck at that line : 
https://github.com/Kinovarobotics/ros_kortex/blob/4f7e716deb087c4157ed9a58342e494d8aa5f576/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp#L60

because, based on your `roslaunch` output, it never reaches this section (couple lines below) : 
https://github.com/Kinovarobotics/ros_kortex/blob/4f7e716deb087c4157ed9a58342e494d8aa5f576/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp#L110-L114

 I would try changing the line to different variations like
```
model.initParam("robot_description");
```
and
```
model.initParam("~robot_description"); // this one I doubt very much
```

Keep in mind you need to rebuild your catkin workspace with `catkin_make` each time you change some in the cpp files for the changes to take effect. Forgot to specify last time, so if you didn't do it, you might want to try again with 
```
model.initParam("/robot_description");
```

yj-Tang, 2022-01-10:
I tried different variations, but it still doesn't work. However, I can simulate the arm in gazebo and use moveit to control it without the driver. So I am wondering what is the function of this driver? Can we simulate the arm without the driver?

felixmaisonneuve, 2022-01-10:
The driver simulates some API calls (listed [here](https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel/kortex_examples#the-ways-to-use-a-kortex-arm-with-ros)) and add ros subscribers to certain topics that uses those API calls. It also initialize the 3 pre-defined arm positions in MoveIt (home, vertical, zero), so they might not work. Finally, it creates the ros publisher that publishes the arm's feedback on the `base_feedback` topic.  
From what I see that is pretty much what `KortexArmSimulation()` does.

If your use case doesn't use these and you are able to not use the driver, I think you could get away without using it, but some features may be broken, I am not sure to want extent it is a necessity. 

Out of curiosity, what did you remove exactly? What modifications did you do?

The next step to debug this issue would be to add `ROS_INFO` everywhere between lines 54 and 108 in [kortex_arm_simulation.cpp](https://github.com/Kinovarobotics/ros_kortex/blob/noetic-devel/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp) and trying to find out what line is blocking the simulation driver initialization and **why** this happens.

yj-Tang, 2022-01-10:
Actually, we didn't remove anything. We just combine the arm and our mobile base in one urdf file. Then create a moveit setup for the whole robot. The controllers can run correctly, but the driver of the arm always died. 

We can control the arm after all the stuffs are loaded in Gazebo. However, the initial pose of the robot can not be controlled (maybe it is influenced by the death of the driver ) which is quite annoying. 

I will try to debug with ```ROS_INFO``` then give the updates here.

yj-Tang, 2022-01-10:
After adding the ```ROS_INFO```, I think the error comes after the first loop of line 70. The outputs in terminals are:
```
[ INFO] [1641834504.399191400]: -------------------------------------------------
[ INFO] [1641834504.399563320]: line 56-robot_name 
[ INFO] [1641834504.399740919]: line 57-prefix
[ INFO] [1641834504.401829537]: line 62-robot_description
[ INFO] [1641834504.402037072]: line 63-robot_description
[ INFO] [1641834504.402875995]: line 67-arm_acceleration_max_limits
[ INFO] [1641834504.402892335]: line 69 completed
[ INFO] [1641834504.402905719]: line 70 completed, m_arm_joint_names : joint_1
[kinova_arm_driver-8] process has died [pid 353696, exit code -11, cmd /home/yujie/robot_pushing/pushing_training/noetic_husky_2/devel/lib/kortex_driver/kortex_arm_driver __name:=kinova_arm_driver __log:=/home/yujie/.ros/log/e9896e72-7237-11ec-8294-97f7c0019c07/kinova_arm_driver-8.log].
log file: /home/yujie/.ros/log/e9896e72-7237-11ec-8294-97f7c0019c07/kinova_arm_driver-8*.log
Topic [///joint_cmd] is not valid.
Service [///joint_cmd_req] is not valid.
```

Our robot namespace is "/", so maybe the error comes from something related to this topic name [///joint_cmd] ?

felixmaisonneuve, 2022-01-11:
So turns out I was wrong and the driver can find the model just fine. It is crashing when trying to get the arm joints at 
https://github.com/Kinovarobotics/ros_kortex/blob/fa072f13b500b399e109bb146891c3e9ebb830cc/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp#L71

You printed the value `m_arm_joint_names[i]` and, to me, `joint_1` seems correct.

Can you copy the output of `rosparam get /robot_description` and `rosparam get /kinova_arm_driver/joint_names`?
If your Husky uses a joint named `joint_1`, the driver will find this joint instead of the arm's one. If that's the case, we will see it in the `robot_description`. You could use a `prefix` when launching your arm's driver. It will rename your joints to something like `prefix_joint_1`, it will solve the conflict (if that is the issue).

I am not sure what the `///joint_cmd` topic and the `///joint_cmd_req` service are. I have never seen these and I don't see any reference to them in ros_kortex.

dfalveargOT, 2022-04-05:
Hi @yj-Tang, @civerachb-cpr, and @felixmaisonneuve, I have the same issue. Have you solved the problem?.
I'm trying to simulate Dingo-O with a gen-3 arm, then looking at the comments I followed all the suggestions without success.
Debugging the Kortex-arm-simulation I found:
- The rosparam dump shows: **is_initialized: false**
- Running **roslaunch kortex_driver kortex_driver.launch** shows **Action client not connected: gen3_joint_trajectory_controller/follow_joint_trajectory**.
- I found that the code got stuck in `m_follow_joint_trajectory_action_client->waitForServer();`
https://github.com/Kinovarobotics/ros_kortex/blob/fa072f13b500b399e109bb146891c3e9ebb830cc/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp#L189-L192
- Gazebo is not updating the joint's position.
Thanks

civerachb-cpr, 2022-04-06:
@dfalveargOT unfortunately I haven't had much chance to dig deeper into this in a long time.  Last time I tried I was still encountering these issues though, so it sounds like not much has changed between when I opened this ticket and now if you're still running into this.

RX-00, 2022-10-06:
@civerachb-cpr Hello! I'm also having the same problems detailed by everyone above. I haven't been able to find/create a fix either. Are there any updates/progress from the Clearpath side?

Thanks!

civerachb-cpr, 2022-10-12:
@RX-00, I've been swamped with other work lately and haven't had much time to revisit this.  Last time I tried in May or June I was still having the same problems.

civerachb-cpr, 2022-12-20:
Sorry about the delay in replying.  I'd forgotten about this ticket as I haven't had to do any work with arms in a long time.

I just tried it again; still seeing the same issues I previously reported.  The situation appears unchanged since my last update six months ago.

DavidPaulBenjamin, 2024-07-07:
We have this issue, too. We have a mobile robot with two gen3 arms. I changed line 60 in kortex_arm_simulation.cpp as you said, and I commented out the moveit startup lines 141 to 149. Now both arm drivers are running. They both say is_initialized is false. This is because kortex_arm_simulation.cpp has hung at line 192, which is:

m_follow_joint_trajectory_action_client->waitForServer

and never returns to kortex_arm_driver.cpp which sets is_initialized to true, as well as registering the simulation handlers to the simulation will run properly. Something appears to be wrong with the communication between the action clients and the action servers. Perhaps a namespace problem?