Kinovarobotics/ros_kortex #164: External force/torque on all axes

rustlluk, 2021-06-28:
Hi,
Is there a possibility to see external forces/torques on all axes of the Kinova Gen3? I know I can check torques/forces on the wrench from `BaseFeedback/external_tool_wrench_torque*`, however I would like to have it for all axes. Also, the wrench forces don't look really external, there is some torque even when the robot is not moving and nothing is applying force on it (after some movements, it looks like it corresponds to the changes of gravity).

I have tested to use a physics simulator to compute dynamic torques on the robot, but it's not really precise and also slow.

Thanks

felixmaisonneuve, 2021-06-28:
Hi @rustlluk,
The external tool wrench force represents an approximation of the force applied on the end-effector. By default only gravity will be taken into account as an external force (otherwise the arm would fall on the ground). 

What are you trying to acheive exactly? I am not sure I understand your question

Best,
Felix

rustlluk, 2021-06-28:
Usually, one can express the torque as dynamic + external part, where the dynamic part contains gravity, Coriolis + centrifugal force and acceleration (inertia matrix). From this equation, gravity should not be taken as "external force", so I would expect that `external_tool_wrench_torque` should return zero when there is no other force is applied.

However, even if gravity is taken as an external force, I would like to know this information expressed in each joint/axis. I know, I can compute it using the Jacobian from the wrench force, however I think it is unnecessary when the robot has the force/torque sensors in each joint and, I guess, the driver should compute these things internally.

My desired outcome is to have for example:
joint:                                J_1  J_2  J_3   J_4 J_5  J_6    J_7
torque[Nm] (magnitude): 0.1   0.5   0.3    0     0.1    0.3    0

In an ideal case, this output would show in case something touches the robot and otherwise, the vector would be full of zeros. But with your conception of "external torque", there would be also added gravity. 

So, can I read something like this from the robot?

Best,
Lukas

felixmaisonneuve, 2021-06-28:
If you are only looking for the torque values of each joint, you could use [JointTorques](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/messages/Base/JointTorques.md) that comes from the Base (There is also JointAngles, JointSpeeds,... available).

You might also be interested in [ActuatorFeedback](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/messages/BaseCyclic/ActuatorFeedback.md) that could give more advanced informations for a specific joint.

Is that what you are loooking for?

rustlluk, 2021-06-29:
Yes, I know I can get these values. But these are directly from the torque sensors, and as I said, I need only values of external torque, i.e. values which does not include dynamics of the robot and gravity. 

From you answers it seems there is no such possibility, so I will probably need to compute this by myself.

felixmaisonneuve, 2021-06-29:
Unfortunately, we do not provide such values. Like you said, all the values you will get will come directly from the torque sensors. 

I think you will have to compute it yourself.

I will close this issue, but feel free to open it again if you encounter something else.
Sorry for the inconvenience,
Felix

closed 2021-06-29 by felixmaisonneuve