Kinovarobotics/ros_kortex #166: Build fail on Noetic

RaduCorcodel, 2021-07-06:
Fresh install of Ubuntu20 and fresh install of ROS Noetic. I cloned the repo and switched to noetic-devel branch, then followed the build instructions exactly. Tried to build ros_kortex and during build I get the following sets of errors:

```
 [ 98%] Building CXX object ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/src/generated/robot/deviceconfig_ros_converter.cpp.o
In file included from /usr/include/ignition/math6/ignition/math/Quaternion.hh:20,
                 from /usr/include/ignition/math6/ignition/math/Pose3.hh:20,
                 from /usr/include/sdformat-9.5/sdf/Actor.hh:23,
                 from /usr/include/sdformat-9.5/sdf/sdf.hh:2,
                 from /usr/include/gazebo-11/gazebo/common/Plugin.hh:33,
                 from /homes/corcodel/ws_kortex/src/ros_kortex/third_party/roboticsgroup_gazebo_plugins/include/roboticsgroup_gazebo_plugins/disable_link_plugin.h:33,
                 from /homes/corcodel/ws_kortex/src/ros_kortex/third_party/roboticsgroup_gazebo_plugins/src/disable_link_plugin.cpp:23:
/usr/include/ignition/math6/ignition/math/Helpers.hh:40:13: warning: variable templates only available with ‘-std=c++14’ or ‘-std=gnu++14’
   40 | constexpr T IGN_MASSMATRIX3_DEFAULT_TOLERANCE = T(10);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/ignition/math6/ignition/math/Helpers.hh: In function ‘std::chrono::_V2::steady_clock::duration ignition::math::v6::stringToDuration(const string&)’:
/usr/include/ignition/math6/ignition/math/Helpers.hh:990:28: error: ‘chrono_literals’ is not a namespace-name
  990 |       using namespace std::chrono_literals;
      |                            ^~~~~~~~~~~~~~~
/usr/include/ignition/math6/ignition/math/Helpers.hh: In function ‘std::chrono::_V2::steady_clock::time_point ignition::math::v6::stringToTimePoint(const string&)’:
/usr/include/ignition/math6/ignition/math/Helpers.hh:1032:28: error: ‘chrono_literals’ is not a namespace-name
 1032 |       using namespace std::chrono_literals;
      |                            ^~~~~~~~~~~~~~~
/usr/include/ignition/math6/ignition/math/Helpers.hh:1033:56: error: unable to find numeric literal operator ‘operator""s’
 1033 |       std::chrono::steady_clock::time_point timePoint{-1s};
      |                                                        ^~
/usr/include/ignition/math6/ignition/math/Helpers.hh:1033:56: note: use ‘-fext-numeric-literals’ to enable more built-in suffixes
/usr/include/ignition/math6/ignition/math/Helpers.hh:1033:58: error: no matching function for call to ‘std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long int, std::ratio<1, 1000000000> > >::time_point(<brace-enclosed initializer list>)’
 1033 |       std::chrono::steady_clock::time_point timePoint{-1s};
      |                                                          ^
In file included from /usr/include/c++/9/mutex:39,
                 from /usr/include/boost/system/detail/std_interoperability.hpp:13,
                 from /usr/include/boost/system/error_code.hpp:959,
                 from /usr/include/boost/system/system_error.hpp:11,
                 from /usr/include/boost/thread/exceptions.hpp:22,
                 from /usr/include/boost/thread/pthread/mutex.hpp:14,
                 from /usr/include/boost/thread/mutex.hpp:16,
                 from /opt/ros/noetic/include/ros/publisher.h:36,
                 from /opt/ros/noetic/include/ros/node_handle.h:32,
                 from /opt/ros/noetic/include/ros/ros.h:45,
                 from /homes/corcodel/ws_kortex/src/ros_kortex/third_party/roboticsgroup_gazebo_plugins/include/roboticsgroup_gazebo_plugins/disable_link_plugin.h:27,
                 from /homes/corcodel/ws_kortex/src/ros_kortex/third_party/roboticsgroup_gazebo_plugins/src/disable_link_plugin.cpp:23:
/usr/include/c++/9/chrono:643:14: note: candidate: ‘template<class _Dur2, class> constexpr std::chrono::time_point<_Clock, _Dur>::time_point(const std::chrono::time_point<_Clock, _Dur2>&)’
  643 |    constexpr time_point(const time_point<clock, _Dur2>& __t)
      |              ^~~~~~~~~~
/usr/include/c++/9/chrono:643:14: note:   template argument deduction/substitution failed:
/usr/include/c++/9/chrono:636:21: note: candidate: ‘constexpr std::chrono::time_point<_Clock, _Dur>::time_point(const duration&) [with _Clock = std::chrono::_V2::steady_clock; _Dur = std::chrono::duration<long int, std::ratio<1, 1000000000> >; std::chrono::time_point<_Clock, _Dur>::duration = std::chrono::duration<long int, std::ratio<1, 1000000000> >]’
  636 |  constexpr explicit time_point(const duration& __dur)
      |                     ^~~~~~~~~~
/usr/include/c++/9/chrono:636:21: note:   conversion of argument 1 would be ill-formed:
/usr/include/c++/9/chrono:633:12: note: candidate: ‘constexpr std::chrono::time_point<_Clock, _Dur>::time_point() [with _Clock = std::chrono::_V2::steady_clock; _Dur = std::chrono::duration<long int, std::ratio<1, 1000000000> >]’
  633 |  constexpr time_point() : __d(duration::zero())
      |            ^~~~~~~~~~
/usr/include/c++/9/chrono:633:12: note:   candidate expects 0 arguments, 1 provided
/usr/include/c++/9/chrono:626:14: note: candidate: ‘constexpr std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long int, std::ratio<1, 1000000000> > >::time_point(const std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long int, std::ratio<1, 1000000000> > >&)’
  626 |       struct time_point
      |              ^~~~~~~~~~
/usr/include/c++/9/chrono:626:14: note:   conversion of argument 1 would be ill-formed:
/usr/include/c++/9/chrono:626:14: note: candidate: ‘constexpr std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long int, std::ratio<1, 1000000000> > >::time_point(std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long int, std::ratio<1, 1000000000> > >&&)’
/usr/include/c++/9/chrono:626:14: note:   conversion of argument 1 would be ill-formed:
In file included from /usr/include/ignition/math6/ignition/math/Quaternion.hh:20,
                 from /usr/include/ignition/math6/ignition/math/Pose3.hh:20,
                 from /usr/include/sdformat-9.5/sdf/Actor.hh:23,
                 from /usr/include/sdformat-9.5/sdf/sdf.hh:2,
                 from /usr/include/gazebo-11/gazebo/common/Plugin.hh:33,
                 from /homes/corcodel/ws_kortex/src/ros_kortex/third_party/roboticsgroup_gazebo_plugins/include/roboticsgroup_gazebo_plugins/mimic_joint_plugin.h:36,
                 from /homes/corcodel/ws_kortex/src/ros_kortex/third_party/roboticsgroup_gazebo_plugins/src/mimic_joint_plugin.cpp:23:
/usr/include/ignition/math6/ignition/math/Helpers.hh:40:13: warning: variable templates only available with ‘-std=c++14’ or ‘-std=gnu++14’
   40 | constexpr T IGN_MASSMATRIX3_DEFAULT_TOLERANCE = T(10);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/ignition/math6/ignition/math/Helpers.hh: In function ‘std::chrono::_V2::steady_clock::duration ignition::math::v6::stringToDuration(const string&)’:
[... 317 lines cut ...]
make[2]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/build.make:102: ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/src/generated/robot/actuatorcyclic_proto_converter.cpp.o] Interrupt
make[2]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/build.make:141: ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/src/generated/robot/base_ros_converter.cpp.o] Interrupt
make[1]: *** [CMakeFiles/Makefile2:27985: ros_kortex/kortex_examples/CMakeFiles/example_full_arm_movement_cpp.dir/all] Interrupt
make[2]: *** [ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/build.make:206: ros_kortex/kortex_driver/CMakeFiles/kortex_driver_generated_files.dir/src/generated/robot/common_ros_converter.cpp.o] Interrupt
make[1]: *** [CMakeFiles/Makefile2:3544: ros_kortex/third_party/gazebo-pkgs/gazebo_version_helpers/CMakeFiles/gazebo_version_helpers.dir/all] Interrupt
make[1]: *** [CMakeFiles/Makefile2:4206: ros_kortex/third_party/roboticsgroup_gazebo_plugins/CMakeFiles/roboticsgroup_gazebo_mimic_joint_plugin.dir/all] Interrupt
make[1]: *** [CMakeFiles/Makefile2:4125: ros_kortex/third_party/roboticsgroup_gazebo_plugins/CMakeFiles/roboticsgroup_gazebo_disable_link_plugin.dir/all] Interrupt
[... cut ...]
  File "/opt/ros/noetic/bin/catkin_make", line 249, in main
    run_command(cmd, make_path)
  File "/opt/ros/noetic/lib/python3/dist-packages/catkin/builder.py", line 241, in run_command
    proc.wait()
  File "/usr/lib/python3.8/subprocess.py", line 1083, in wait
    return self._wait(timeout=timeout)
  File "/usr/lib/python3.8/subprocess.py", line 1806, in _wait
    (pid, sts) = self._try_wait(0)
  File "/usr/lib/python3.8/subprocess.py", line 1764, in _try_wait
    (pid, sts) = os.waitpid(self.pid, wait_flags)
KeyboardInterrupt
```

felixmaisonneuve, 2021-07-06:
Hi @RaduCorcodel,

Can you `git pull` your noetic-devel branch and try again. I just merged the modifications needed in [this PR](https://github.com/Kinovarobotics/ros_kortex/pull/165). 

I created the branch a few days ago from the ROS Melodic branch, but only merge its actual content today. 

It should now work and you should be able to build with the latest changes.
Sorry about that.  

Best,
Felix

RaduCorcodel, 2021-07-07:
Hi @felixmaisonneuve,

No worries, I wasn't sure if the problem was on my end because the computer had a slight hiccup when installing ROS. Installed package was broken but I thought I fixed it. But with the latest commit, everything builds just fine so it's resolved.

I have the identical problem when I tried to build the ROS package from Robotiq (we're also using other grippers from Robotiq and controlling them through a separate HW controller). Do you have a hint of what causes this error in the first place? Maybe it's an easy fix since their repo doesn't have a Noetic branch yet.
Thank you in advance. I'll close this now.

closed 2021-07-07 by RaduCorcodel

felixmaisonneuve, 2021-07-07:
Yes, it is because of the 2 repos that we are using in the "third_party" folder. They are both not compatible with the newest version of Gazebo.

First, we were previously using roboticsgroup_gazebo_plugins. This repo is outdated and if you go on its [github page](https://github.com/roboticsgroup/roboticsgroup_gazebo_plugins) you will find the link to the newest (and working) [repo](https://github.com/roboticsgroup/roboticsgroup_upatras_gazebo_plugins). 

Also (and I think this is where you error comes from), the third_party/gazebo-pkgs is also outdated and defines the compiler version to c++11 in the CMakelists.txt (which is not compatible with the math6 librairy or something). You can make it work by using c++17 instead (if I remember correctly) or by using the latest version of its [repo](https://github.com/JenniferBuehler/gazebo-pkgs) (like I did). To change the compiler version, you would need something like that in the CMakelists.txt files within the package :
```
set(CMAKE_CXX_FLAGS "-std=c++17")
```
instead of 
```
set(CMAKE_CXX_FLAGS "-std=c++11")
```

You can find hints about this in the huge output you provided : 
```
/usr/include/ignition/math6/ignition/math/Capsule.hh:91:15: note: ‘std::optional’ is only available from C++17 onwards
   91 |       public: std::optional< MassMatrix3<Precision> > MassMatrix() const;
```
```
/usr/include/ignition/math6/ignition/math/Ellipsoid.hh:77:15: note: ‘std::optional’ is only available from C++17 onwards
   77 |       public: std::optional< MassMatrix3<Precision> > MassMatrix() const;
```
...

Hope this answers your question
Best,
Felix

RaduCorcodel, 2021-07-07:
Hi @felixmaisonneuve,

This is perfect. I did see the compiler hinting to use C++17 and I did set it in the metapackage hoping it will apply to the entire build, but now that I think of it I think it gets overwritten by the sub-project's CMakeLists. I'll take a second pass at it and fingers crossed it works. Thanks again for the tips, it's exactly what I need to get myself started.

Best,
Radu