Kinovarobotics/ros_kortex #167: Error in launching kortex driver from "roslaunch kortex_driver kortex_driver.launch"

GitNzino, 2021-07-07:
## Description

I followed all the instructions provided on the GitHub Readme for the installation, but when i try to launch the driver using the command on terminal _**roslaunch kortex_driver kortex_driver.launch arm:=gen3_lite**_. At a certain point of the process in the terminal appear an error of failure that I cannot solve and and I was not able to find information about on internet.

The two error are reported in the following picture, at the end of the issue there is the complete terminal output.

[image]
[image]

I strongly hope that you can help me, thank you in advice

## Version

ROS distribution : Melodic with version 1.14.11;
Robot: Kinova Kortex Gen 3 Lite;

Branch and commit you are using : The lastest

## Steps to reproduce

1. *step 1: start the driver.*
2. *step 2: move the robot with an example script.*

## Terminal Output

 ```
lorenzo@ubuntu18-04:~$ roslaunch kortex_driver kortex_driver.launch 
... logging to /home/lorenzo/.ros/log/9773bb52-df29-11eb-aca0-3c950922ab5a/roslaunch-ubuntu18-04-31406.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://ubuntu18-04:36699/

SUMMARY
========

PARAMETERS
 * /my_gen3_lite/hztest_test_base_feedback/hz: 40
 * /my_gen3_lite/hztest_test_base_feedback/hzerror: 5.0
 * /my_gen3_lite/hztest_test_base_feedback/test_duration: 30.0
 * /my_gen3_lite/hztest_test_base_feedback/topic: base_feedback
 * /my_gen3_lite/hztest_test_base_feedback/wait_time: 10.0
 * /my_gen3_lite/hztest_test_driver_joint_state/hz: 40
 * /my_gen3_lite/hztest_test_driver_joint_state/hzerror: 5.0
 * /my_gen3_lite/hztest_test_driver_joint_state/test_duration: 30.0
 * /my_gen3_lite/hztest_test_driver_joint_state/topic: base_feedback/joi...
 * /my_gen3_lite/hztest_test_driver_joint_state/wait_time: 10.0
 * /my_gen3_lite/hztest_test_joint_state_publisher_joint_states/hz: 40
 * /my_gen3_lite/hztest_test_joint_state_publisher_joint_states/hzerror: 5.0
 * /my_gen3_lite/hztest_test_joint_state_publisher_joint_states/test_duration: 30.0
 * /my_gen3_lite/hztest_test_joint_state_publisher_joint_states/topic: joint_states
 * /my_gen3_lite/hztest_test_joint_state_publisher_joint_states/wait_time: 10.0
 * /my_gen3_lite/joint_state_publisher/rate: 40
 * /my_gen3_lite/joint_state_publisher/source_list: ['base_feedback/j...
 * /my_gen3_lite/joint_state_publisher/use_gui: False
 * /my_gen3_lite/kortex_driver_tests/api_connection_inactivity_timeout_ms: 20000
 * /my_gen3_lite/kortex_driver_tests/api_rpc_timeout_ms: 2000
 * /my_gen3_lite/kortex_driver_tests/api_session_inactivity_timeout_ms: 35000
 * /my_gen3_lite/kortex_driver_tests/arm: gen3_lite
 * /my_gen3_lite/kortex_driver_tests/cyclic_data_publish_rate: 40
 * /my_gen3_lite/kortex_driver_tests/default_goal_time_tolerance: 0.5
 * /my_gen3_lite/kortex_driver_tests/default_goal_tolerance: 0.005
 * /my_gen3_lite/kortex_driver_tests/dof: 6
 * /my_gen3_lite/kortex_driver_tests/gripper: gen3_lite_2f
 * /my_gen3_lite/kortex_driver_tests/gripper_joint_limits_max: [-0.09]
 * /my_gen3_lite/kortex_driver_tests/gripper_joint_limits_min: [0.96]
 * /my_gen3_lite/kortex_driver_tests/gripper_joint_names: ['right_finger_bo...
[... 265 lines cut ...]
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...
[ INFO] [1625665598.016614120]: 

********************************************************
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1625665598.017394643]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1625665598.017498398]: MoveGroup context initialization complete

You can start planning now!

```

RaduCorcodel, 2021-07-07:
Hi,

Did you updated the firmware to 2.3.0 as well? Perhaps that's what causes it.

Best,
Radu

GitNzino, 2021-07-07:
Thank you for the feedback, i use the conan tool to download automatically the API and I think that conan should download also the firmware. 

Do you suggest that I have to download the firmware from my own?

alexvannobel, 2021-07-07:
Hi @GitNzino ,

Yes you have to download the firmware on your own and also upload it to the arm following the steps in the User Guide.

The [Kortex release 2.3.0 page](https://github.com/Kinovarobotics/kortex/releases/tag/v2.3.0) has all the links you need.

Best,
Alex

GitNzino, 2021-07-07:
Now it works!! Thank you very much!!

But if I try to move the robot with riviz I obtain an error regarding the same error, I'm putting an example in the picture down:
[image]
[image]

And all the output code:

```
You can start planning now!

[ WARN] [1625671620.647168248]: Could not identify parent group for end-effector 'end_effector'
[ WARN] [1625671620.666898995]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[ WARN] [1625671621.600928564]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[ INFO] [1625671658.178027600]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ INFO] [1625671658.178232726]: Starting state is just outside bounds (joint 'joint_3'). Assuming within bounds.
[ INFO] [1625671658.179844719]: Planner configuration 'arm[RRTConnect]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1625671658.181078485]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1625671658.181187036]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1625671658.181278808]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1625671658.182367153]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1625671658.194374505]: arm[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1625671658.194991865]: arm[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1625671658.195315681]: arm[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1625671658.197496191]: arm[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1625671658.197743428]: ParallelPlan::solve(): Solution found by one or more threads in 0.017119 seconds
[ INFO] [1625671658.198122262]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1625671658.198699312]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1625671658.200500330]: arm[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1625671658.201186215]: arm[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1625671658.201417303]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1625671658.201522478]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1625671658.203816490]: arm[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1625671658.204268212]: arm[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1625671658.204428096]: ParallelPlan::solve(): Solution found by one or more threads in 0.006391 seconds
[ INFO] [1625671658.204667788]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1625671658.204800570]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1625671658.207254030]: arm[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1625671658.207336936]: arm[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1625671658.207523025]: ParallelPlan::solve(): Solution found by one or more threads in 0.002937 seconds
[ INFO] [1625671658.213358457]: SimpleSetup: Path simplification took 0.005701 seconds and changed from 3 to 2 states
[ INFO] [1625671658.214467776]: Planning adapters have added states at index positions: [ 0 ]
[ INFO] [1625671667.377734667]: Execution request received
[ INFO] [1625671667.403545381]: New goal received.
[ INFO] [1625671667.403588385]: Joint Trajectory Goal is accepted.
[ERROR] [1625671667.416323365]: Joint Trajectory failed validation in the arm.
[ERROR] [1625671667.416469681]: Error 1 : Invalid Joint Position - Provided joint position for actuator (2) exceeds the joint limits
[ERROR] [1625671667.416602825]: Error 2 : Invalid Joint Position - Provided joint position for actuator (2) exceeds the joint limits
[ERROR] [1625671667.416686601]: Error 3 : Invalid Joint Position - Provided joint position for actuator (2) exceeds the joint limits
[ERROR] [1625671667.416803471]: Error 4 : Invalid Joint Position - Provided joint position for actuator (2) exceeds the joint limits
[ INFO] [1625671667.417575801]: Controller gen3_lite_joint_trajectory_controller successfully finished
[ WARN] [1625671667.417850014]: Controller handle gen3_lite_joint_trajectory_controller reports status ABORTED
[ INFO] [1625671667.418083385]: Completed trajectory execution with status ABORTED ...
[ INFO] [1625671667.418624421]: Execution completed: ABORTED
```

felixmaisonneuve, 2021-07-07:
Hi @GitNzino,

This is because Rviz has limits and it will fail whenever your robot is outside of those limits. 
Since your robot "fell" when it was turned off, it is now outside of those limits. 
If you manually move the robot with the joystick just a bit it should get back within those limits and everything should work fine.

Regards,
Felix

GitNzino, 2021-07-07:
Thank you again, once i have moved from the "fell" position it works perfectly.
I'm closing the issue

closed 2021-07-07 by GitNzino