Kinovarobotics/ros_kortex #168: Kortex messages missing

jabbershort, 2021-07-07:
I have built the kortex_driver up and launched it successfully, however, now I've got the kortex_driver.launch running, not all the messages are showing in `rosmsg list`, only the following:

kortex_driver/FollowCartesianTrajectoryAction
kortex_driver/FollowCartesianTrajectoryActionFeedback
kortex_driver/FollowCartesianTrajectoryActionGoal
kortex_driver/FollowCartesianTrajectoryActionResult
kortex_driver/FollowCartesianTrajectoryFeedback
kortex_driver/FollowCartesianTrajectoryGoal
kortex_driver/FollowCartesianTrajectoryResult

However, even these aren't available within the autocomplete following `rosmsg show....`. All the python files are present under /catkin_workspace/devel/lib/python2.7/dist-packages/kortex_driver/msg/. 

I'm assuming I've missed something within the launch file or setup (I've run source /catkin_workspace/devel/setup.bash)?

felixmaisonneuve, 2021-07-07:
have you sourced the new devel/setup.bash in your catkin_workspace?

jabbershort, 2021-07-07:
When you say the new devel/setup.bash, I've run the one produced by the catkin_make command, is that the one you mean?

felixmaisonneuve, 2021-07-07:
yes, when you do `catkin_make`, it generates a `build` and a `devel` folder. You then need to source the newly created setup.bash : `source devel/setup.bash`

jabbershort, 2021-07-07:
Yes I've done that. It seems to have worked, because I can run the launch files fine, just it seems that the messages haven't worked.

felixmaisonneuve, 2021-07-07:
Ok, 

What ROS distro and ubuntu are you using? 
On what ros_kortex branch are you on? 
Do you have multiple catkin workspaces in different locations (a `catkin_make clean`, deleting `devel` and `build` folders and building again with `catkin_make` can fix you conflicting catkin workspaces)?  

What commands did you do to get this output?

jabbershort, 2021-07-07:
So, I'm running Melodic straight from Docker Hub (ros:melodic-ros-core, I believe thats based on 18.04), which I've updated and followed the instructions on the readme to setup, install dependancies and make kortex package. I should be on the default branch, whichever is pulled by git. I'm just trying a rebuild based on melodic-devel instead, will post when thats finished building.

I've only got one catkin_workspace. 

In order to recreate this, I'm starting my container, which has pre-installed and made ros_kortex, then running `roslaunch kortex_driver kortex_driver.launch` and in a separate window, running `rosmsg list` once the core is up and running. I'm not physically connected to an arm at the moment, but a similar thing happens using the gazebo spawn. When using the kortex_gazebo spawn_kortex_robot.launch, I can get the rostopic echos to work for /my_gen3/joints_states, /my_gen3/base_feedback and /my_gen3/kortex_error, two of which are obviously custom message types, but they still don't appear in the rosmsg list.

jabbershort, 2021-07-07:
So I've tested the build using melodic-devel instead, but with no success, I still get all the same messages. Would this not work whilst not physically connected to the robot? I'm back with one tomorrow, so will give it a test then as well.

alexvannobel, 2021-07-07:
Hi @jabbershort ,

Can you send the full log of your `roslaunch kortex_driver kortex_driver.launch` command?

Alex

jabbershort, 2021-07-08:
Here is the log from starting roslaunch:
root@99ede36994fe:/# roslaunch kortex_driver kortex_driver.launch dof:=6 start_rviz:=false start_moveit:=false gripper:=robotiq_2f_85
... logging to /root/.ros/log/c2263292-dfd0-11eb-9c76-0242ac110002/roslaunch-99ede36994fe-341.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://99ede36994fe:40121/

SUMMARY
========

PARAMETERS
 * /my_gen3/hztest_test_base_feedback/hz: 40
 * /my_gen3/hztest_test_base_feedback/hzerror: 5.0
 * /my_gen3/hztest_test_base_feedback/test_duration: 30.0
 * /my_gen3/hztest_test_base_feedback/topic: base_feedback
 * /my_gen3/hztest_test_base_feedback/wait_time: 10.0
 * /my_gen3/hztest_test_driver_joint_state/hz: 40
 * /my_gen3/hztest_test_driver_joint_state/hzerror: 5.0
 * /my_gen3/hztest_test_driver_joint_state/test_duration: 30.0
 * /my_gen3/hztest_test_driver_joint_state/topic: base_feedback/joi...
 * /my_gen3/hztest_test_driver_joint_state/wait_time: 10.0
 * /my_gen3/hztest_test_joint_state_publisher_joint_states/hz: 40
 * /my_gen3/hztest_test_joint_state_publisher_joint_states/hzerror: 5.0
 * /my_gen3/hztest_test_joint_state_publisher_joint_states/test_duration: 30.0
 * /my_gen3/hztest_test_joint_state_publisher_joint_states/topic: joint_states
 * /my_gen3/hztest_test_joint_state_publisher_joint_states/wait_time: 10.0
 * /my_gen3/joint_state_publisher/rate: 40
 * /my_gen3/joint_state_publisher/source_list: ['base_feedback/j...
 * /my_gen3/joint_state_publisher/use_gui: False
 * /my_gen3/kortex_driver_tests/api_connection_inactivity_timeout_ms: 20000
 * /my_gen3/kortex_driver_tests/api_rpc_timeout_ms: 2000
 * /my_gen3/kortex_driver_tests/api_session_inactivity_timeout_ms: 35000
 * /my_gen3/kortex_driver_tests/arm: gen3
 * /my_gen3/kortex_driver_tests/cyclic_data_publish_rate: 40
 * /my_gen3/kortex_driver_tests/default_goal_time_tolerance: 0.5
 * /my_gen3/kortex_driver_tests/default_goal_tolerance: 0.005
 * /my_gen3/kortex_driver_tests/dof: 6
 * /my_gen3/kortex_driver_tests/gripper: robotiq_2f_85
 * /my_gen3/kortex_driver_tests/gripper_joint_limits_max: [0.8]
 * /my_gen3/kortex_driver_tests/gripper_joint_limits_min: [0.0]
 * /my_gen3/kortex_driver_tests/gripper_joint_names: ['finger_joint']
 * /my_gen3/kortex_driver_tests/ip_address: 192.168.1.10
 * /my_gen3/kortex_driver_tests/joint_names: ['joint_1', 'join...
 * /my_gen3/kortex_driver_tests/maximum_accelerations: [1.0, 1.0, 1.0, 1...
 * /my_gen3/kortex_driver_tests/maximum_velocities: [0.8727, 0.8727, ...
 * /my_gen3/my_gen3_driver/api_connection_inactivity_timeout_ms: 20000
 * /my_gen3/my_gen3_driver/api_rpc_timeout_ms: 2000
 * /my_gen3/my_gen3_driver/api_session_inactivity_timeout_ms: 35000
 * /my_gen3/my_gen3_driver/arm: gen3
 * /my_gen3/my_gen3_driver/cyclic_data_publish_rate: 40
 * /my_gen3/my_gen3_driver/default_goal_time_tolerance: 0.5
 * /my_gen3/my_gen3_driver/default_goal_tolerance: 0.005
 * /my_gen3/my_gen3_driver/dof: 6
 * /my_gen3/my_gen3_driver/gripper: robotiq_2f_85
 * /my_gen3/my_gen3_driver/gripper_joint_limits_max: [0.8]
 * /my_gen3/my_gen3_driver/gripper_joint_limits_min: [0.0]
 * /my_gen3/my_gen3_driver/gripper_joint_names: ['finger_joint']
 * /my_gen3/my_gen3_driver/ip_address: 192.168.1.10
 * /my_gen3/my_gen3_driver/joint_names: ['joint_1', 'join...
 * /my_gen3/my_gen3_driver/maximum_accelerations: [1.0, 1.0, 1.0, 1...
 * /my_gen3/my_gen3_driver/maximum_velocities: [0.8727, 0.8727, ...
 * /my_gen3/my_gen3_driver/password: admin
 * /my_gen3/my_gen3_driver/prefix: 
 * /my_gen3/my_gen3_driver/robot_name: my_gen3
 * /my_gen3/my_gen3_driver/sim: False
 * /my_gen3/my_gen3_driver/use_hard_limits: False
 * /my_gen3/my_gen3_driver/username: admin
[... 20 lines cut ...]
[ INFO] [1625737343.198839660]: Session created successfully for UDP services
[ INFO] [1625737343.207974918]: -------------------------------------------------
[ INFO] [1625737343.209358007]: Scanning all devices in robot... 
[ INFO] [1625737343.229108826]: Base device was found on device identifier 0
[ INFO] [1625737343.230404362]: Vision device was found on device identifier 8
[ INFO] [1625737343.232108479]: Interconnect device was found on device identifier 7
[ INFO] [1625737343.263333061]: Actuator device of type SMALL_ACTUATOR was found on device identifier 6
[ INFO] [1625737343.275494096]: Actuator device of type SMALL_ACTUATOR was found on device identifier 5
[ INFO] [1625737343.276914662]: Actuator device of type BIG_ACTUATOR was found on device identifier 1
[ INFO] [1625737343.278421245]: Actuator device of type BIG_ACTUATOR was found on device identifier 2
[ INFO] [1625737343.279995472]: Actuator device of type BIG_ACTUATOR was found on device identifier 3
[ INFO] [1625737343.281468995]: Actuator device of type SMALL_ACTUATOR was found on device identifier 4
[ INFO] [1625737343.287786990]: -------------------------------------------------
[ INFO] [1625737343.603085067]: State changed from INITIALIZING to IDLE

[ INFO] [1625737343.696570385]: State changed from INITIALIZING to IDLE

[ INFO] [1625737343.757717279]: -------------------------------------------------
[ INFO] [1625737343.758587074]: Initializing Kortex Driver's services...
process[my_gen3/joint_state_publisher-3]: started with pid [415]
[ INFO] [1625737346.944230164]: Kortex Driver's services initialized correctly.
[ INFO] [1625737346.946533341]: -------------------------------------------------
[ INFO] [1625737346.973570015]: The Kortex driver has been initialized correctly!
process[my_gen3/robot_state_publisher-4]: started with pid [458]
[ WARN] [1625737350.169307438]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.

jabbershort, 2021-07-08:
I managed to use the messages, e.g. `rostopic pub /my_gen3/in/joint_velocity kortex_driver/Base_JointSpeeds .....`, but haven't managed to reference them in code.

felixmaisonneuve, 2021-07-08:
Looking further, I found that ros has an issue opened and it does not list messages in subfolders. Since all our messages are in subfolders (either `generated` or `non_generated`), it is possible they do not show when running `rosmsg list`. I will link this issue [here](https://github.com/ros/ros_comm/issues/1021). You can also find that link in the official ROS documention [here](https://wiki.ros.org/rosmsg#rosmsg_list). 

This should not affect the comportment and everything should work fine. This is why you are able to use the message while publishing on a topic with `rostopic pub /my_gen3/in/joint_velocity kortex_driver/Base_JointSpeeds .....`.

There is probably something else in your code that would explain why you haven't been able to use it yet. 
Can you show me a code sample of what you are doing?

jabbershort, 2021-07-08:
Hi Felix, thanks for linking that, its very helpful. 

The bit of code I'm trying to run is a simple server endpoint, where I'm sending the ROS messages to Unity. 

```py
import rospy

from ros_tcp_endpoint import TcpServer, RosPublisher, RosSubscriber, RosService, UnityService
from sensor_msgs.msg import JointState
from std_msgs.msg import String
from kortex_driver.msg import KortexError

def main():
    ros_node_name = rospy.get_param("/TCP_NODE_NAME", 'TCPServer')
    buffer_size = rospy.get_param("/TCP_BUFFER_SIZE", 1024)
    connections = rospy.get_param("/TCP_CONNECTIONS", 10)
    tcp_server = TcpServer(ros_node_name, buffer_size, connections)
    rospy.init_node(ros_node_name, anonymous=True)

    tcp_server.start({
        'joints':RosSubscriber('/my_gen3/joint_state',JointState,tcp_server),
        'errors':RosSubscriber('/my_gen3/kortex_error',KortexError,tcp_server),
    })
    rospy.spin()

if __name__ == "__main__":
    main()
```

And I'm getting an error on `from kortex_driver.msg import KortexError`, which is: `ImportError: No module named kortex_driver.msg`. By the sounds of it, I should be able to get this to work, but I've likely incorrectly referenced. What should it be?

felixmaisonneuve, 2021-07-09:
I tried quickly and it seems to work for me. 

Where is your python file located? Is it inside the kortex package?
What command do you use? You should run you script using `rosrun` like so : `rosrun kortex_examples your_script.py` if your script is in `kortex_examples` package (although `./your_script.py` and `python your_script.py` should work the same).
In order for this to work, you need to add `#!/usr/bin/env python` on top of your script and make it executable with `chmod +x your_script.py`.

Also, please make sure you did a `catkin_make` after creating your script and you did `source devel/setup.bash` 

Can you also give the full log of the error you get?

jabbershort, 2021-07-13:
Hi Felix,

Thanks again for being so helpful. I've just given it another go and found that my issue was that I was running a script on another node which didn't have kortex built on it. So that explains it! I've managed to port the message types to my other node and now it works perfectly.

closed 2021-07-13 by jabbershort