Kinovarobotics/ros_kortex #169: Capability to access the forward kinematics scripts

Jaroan, 2021-07-13:
Hello,
I'm trying to use the `example_cartesian_poses_with_notifications.py` to send my robot to separate cartesian positions. I want to know the actual positions that my robot moved after sending the command.

I am able to obtain the join states from the topic `/my_gen3/joint_state` and I thought I could use forward kinematics to check the actual position of the arm's end effector. Is there an explicit script I could use?

Any suggestions and help are appreciated.
Thank you!

felixmaisonneuve, 2021-07-13:
Hi @Jaroan,

We provide examples on how to use forward and inverse kinematics for our Kortex api repo. You can find th C++ example [here](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/examples/111-Kinematics/01-compute-kinematics.cpp) and the python example [here](https://github.com/Kinovarobotics/kortex/blob/master/api_python/examples/111-kinematics/01-compute-kinematics.py).

These services have been generated in our ros repo and you can find them here : 
https://github.com/Kinovarobotics/ros_kortex/blob/72fc06e52b0524dd36155c7ce9be40650ca8b7d6/kortex_driver/src/generated/robot/base_services.cpp#L230-L231

You can use them like any other service. 
To do so, you would need the init the service
```python
compute_forward_kinematics_full_name = '/' + self.robot_name + '/base/compute_forward_kinematics'
rospy.wait_for_service(compute_forward_kinematics_full_name)
self.compute_forward_kinematics = rospy.ServiceProxy(compute_forward_kinematics_full_name, ComputeForwardKinematics)
```
then you will be able to call the service with 
```python
self.compute_forward_kinematics(joint_angles)
```
where `joint_angles` is equivalent to the topic `/my_gen3/joint_state` (I think). 
You can learn more about the `JointAngles` object (needed to compute forward kinematics) [here](https://github.com/Kinovarobotics/kortex/blob/master/api_python/doc/markdown/messages/Base/JointAngles.md#).

Computing inverse kinematics would be the same, except it takes a `Pose` object and return a `JointAngles` object. 

I haven't tried any of this, so it might not be 100% acurate. 
If it doesn't work, I will be glad to help you more, but this should be a good start. 

Regards,
Felix

felixmaisonneuve, 2021-07-13:
You can also look at the topic `/gen3/base_feedback`, which gives the position and orientation of the end effector (amongst other stuff).

Jaroan, 2021-07-13:
Thank you very much for this @felixmaisonneuve .
I will try these out!

Jaroan, 2021-07-14:
Hi @felixmaisonneuve, I had a few observations and doubts.
The message types of `JointState` and  `JointAngles` had different representations. I was a little confused by the way to populate it.
I tried it in this way based on this [example](https://github.com/Kinovarobotics/kortex/blob/762803927987c6cd08367e8f1755db1c193de73e/api_python/examples/111-kinematics/01-compute-kinematics.py#L40) as:
```
joint_angs = JointAngles()
compute_forward_kinematics_full_name = '/' + robot_name + '/base/compute_forward_kinematics'
rospy.wait_for_service(compute_forward_kinematics_full_name)
compute_forward_kinematics = rospy.ServiceProxy(compute_forward_kinematics_full_name, ComputeForwardKinematics)
# print(joint_angs.joint_angles)
for joint_angle in joint_angs.joint_angles:
	joint_angle.joint_identifier = joint_angle
	joint_angle.value= data.position[i]*180.0/np.pi
compute_forward_kinematics(joint_angs)
```
However, I don't get any input, and seem to be accessing the message object's members incorrectly?

I also wanted to know if the `/gen3/base_feedback` topic provides the end effector's Cartesian Pose. I got the following information and found this information and wanted to know if this is the end effector's pose?:
```
  tool_pose_x: 0.419379293919
  tool_pose_y: -0.152442455292
  tool_pose_z: 0.348679870367
  tool_pose_theta_x: 122.467559814
  tool_pose_theta_y: -0.0650764852762
  tool_pose_theta_z: 69.8309936523
```
`$ rostopic echo /my_gen3/base_feedback`
```
frame_id: 0
base: 
  active_state_connection_identifier: 0
  active_state: 0
  arm_voltage: 0.0
  arm_current: 0.0
  temperature_cpu: 0.0
  temperature_ambient: 0.0
  imu_acceleration_x: 0.0
  imu_acceleration_y: 0.0
  imu_acceleration_z: 0.0
  imu_angular_velocity_x: 0.0
  imu_angular_velocity_y: 0.0
  imu_angular_velocity_z: 0.0
  tool_pose_x: 0.419379293919
  tool_pose_y: -0.152442455292
  tool_pose_z: 0.348679870367
  tool_pose_theta_x: 122.467559814
  tool_pose_theta_y: -0.0650764852762
  tool_pose_theta_z: 69.8309936523
  tool_twist_linear_x: 0.0
  tool_twist_linear_y: 0.0
  tool_twist_linear_z: 0.0
  tool_twist_angular_x: 0.0
  tool_twist_angular_y: 0.0
  tool_twist_angular_z: 0.0
  tool_external_wrench_force_x: 0.0
  tool_external_wrench_force_y: 0.0
  tool_external_wrench_force_z: 0.0
  tool_external_wrench_torque_x: 0.0
  tool_external_wrench_torque_y: 0.0
  tool_external_wrench_torque_z: 0.0
  fault_bank_a: 0
  fault_bank_b: 0
  warning_bank_a: 0
  warning_bank_b: 0
  commanded_tool_pose_x: 0.419379293919
  commanded_tool_pose_y: -0.152442455292
  commanded_tool_pose_z: 0.348679870367
  commanded_tool_pose_theta_x: 122.467559814
  commanded_tool_pose_theta_y: -0.0650764852762
  commanded_tool_pose_theta_z: 69.8309936523
actuators: 
  - 
    command_id: 0
    status_flags: 0
    jitter_comm: 0
    position: 20.2384338379
    velocity: -48.4700889587
    torque: 0.0
    current_motor: 0.0
    voltage: 0.0
    temperature_motor: 0.0
    temperature_core: 0.0
    fault_bank_a: 0
    fault_bank_b: 0
    warning_bank_a: 0
    warning_bank_b: 0
.
.
.
  - 
    command_id: 0
    status_flags: 0
    jitter_comm: 0
    position: 89.9528198242
    velocity: -0.0849969312549
    torque: 0.0
    current_motor: 0.0
    voltage: 0.0
    temperature_motor: 0.0
    temperature_core: 0.0
    fault_bank_a: 0
    fault_bank_b: 0
    warning_bank_a: 0
    warning_bank_b: 0
interconnect: 
  feedback_id: 
    identifier: 0
  status_flags: 0
  jitter_comm: 0
  imu_acceleration_x: 0.0
  imu_acceleration_y: 0.0
  imu_acceleration_z: 0.0
  imu_angular_velocity_x: 0.0
  imu_angular_velocity_y: 0.0
  imu_angular_velocity_z: 0.0
  voltage: 0.0
  temperature_core: 0.0
  fault_bank_a: 0
  fault_bank_b: 0
  warning_bank_a: 0
  warning_bank_b: 0
  oneof_tool_feedback: 
    gripper_feedback: 
      - 
        feedback_id: 
          identifier: 0
        status_flags: 0
        fault_bank_a: 0
        fault_bank_b: 0
        warning_bank_a: 0
        warning_bank_b: 0
        motor: 
          - 
            motor_id: 0
            position: 0.0
            velocity: 0.0
            current_motor: 0.0
            voltage: 0.0
            temperature_motor: 0.0
---

```

felixmaisonneuve, 2021-07-15:
Hi @Jaroan,

For your second question, yes, `tool_pose_(xyz)` is the position of the end effector (in meters) and `tool_pose_theta_(xyz)` is the end effector orientation (in degrees). You can also read the end effector linear and angular speed (`tool_twist_linear_x` and `tool_twist_angular_x`).

For your first question, `joint_state` and `JointAngles` are different (my mistake). If you want to get the `JointAngles` (wich is just a list of joint ID and joint angle value for each joint), the easiest way would be to call the service `base/get_measured_joint_angles` here : 
https://github.com/Kinovarobotics/ros_kortex/blob/72fc06e52b0524dd36155c7ce9be40650ca8b7d6/kortex_driver/src/generated/robot/base_services.cpp#L162

You would have to do exactly the same thing you did to use the service to compute kinematics and you would probably end up with something similar to this : 
```
compute_forward_kinematics_full_name = '/' + robot_name + '/base/compute_forward_kinematics'
rospy.wait_for_service(compute_forward_kinematics_full_name)
compute_forward_kinematics = rospy.ServiceProxy(compute_forward_kinematics_full_name, ComputeForwardKinematics)
get_joint_angles_full_name = '/' + robot_name + '/base/get_measured_joint_angles'
rospy.wait_for_service(get_joint_angles_full_name)
get_joint_angles = rospy.ServiceProxy(get_joint_angles_full_name, GetMeasuredJointAngles)
joint_angs = get_joint_angles()
pose = compute_forward_kinematics(joint_angs)
```
(but you could probably create your own `joint_angs` by using the info given by `/my_gen3/joint_state`)

Anyway, I think using the output of `/gen3/base_feedback` and filtering the end effector pose is what you need and it is probably the simplest way to get it. 

Let me know how it goes,
Felix

Jaroan, 2021-07-15:
Yes, this clarifies my doubts!
Thank you very much @felixmaisonneuve!

closed 2021-07-15 by Jaroan