Kinovarobotics/ros_kortex #170: catkin_make error

RolandZhu, 2021-07-13:
## Description

Hi, thanks for providing the fantastic package for Kinova! I followed the instruction to build the package but got stuck at the end. 
Please see the attached log for more details. It'll be great if you can help me to solve this issue.
BTW. I've installed moveit through `sudo apt install ros-melodic-moveit`

## Version

ROS distribution : Melodic

Branch and commit you are using : melodic-devel

## Steps to reproduce

1. catkin_make

[logs.txt](https://github.com/Kinovarobotics/ros_kortex/files/6811838/logs.txt)

felixmaisonneuve, 2021-07-13:
Hi @RolandZhu,

Thank you for providing your full log. This really looks like the problem is coming from your moveit packages. 
Can you make sure your system is correctly update.
Also, make sure you have `ros-melodic-trac-ik` also installed (`sudo apt-get install ros-melodic-trac-ik`). You did not mentionned this package so I think it might be this.

Regards,
Felix

RolandZhu, 2021-07-13:
Hi Felix,

Many thanks for your reply, it works after installing the `ros-melodic-trac-ik`.

Best,
Xinghao

closed 2021-07-13 by felixmaisonneuve