Kinovarobotics/ros_kortex #172: End effector not found in example moveit

GitNzino, 2021-07-20:
## Description

Problem on definition of end effector in kortex_example moveit.
I was running the example on python with the kinova kortex driver with a gen3 ite arm. I passed as argument the correct robot_name, but the gripper does not initialize and does not move how it should be in the example code

## Version

ROS distribution : Melodic  (ubuntu 18.04 LTS)

Branch and commit you are using : latest

## Steps to reproduce
I want to close and open the gripper as shown in code example

## Code example (if necessary)

```
class ExampleMoveItTrajectories(object):
  """ExampleMoveItTrajectories"""
  def __init__(self):

    # Initialize the node
    super(ExampleMoveItTrajectories, self).__init__()
    moveit_commander.roscpp_initialize(sys.argv)
    rospy.init_node('example_move_it_trajectories')

    try:
      self.is_gripper_present = rospy.get_param(rospy.get_namespace() + "is_gripper_present", False)
      if self.is_gripper_present:
        gripper_joint_names = rospy.get_param(rospy.get_namespace() + "gripper_joint_names", [])
        self.gripper_joint_name = gripper_joint_names[0]
      else:
        gripper_joint_name = ""
      self.degrees_of_freedom = rospy.get_param(rospy.get_namespace() + "degrees_of_freedom", 7)

      # Create the MoveItInterface necessary objects
      arm_group_name = "arm"
      self.robot = moveit_commander.RobotCommander("robot_description")
      self.scene = moveit_commander.PlanningSceneInterface(ns=rospy.get_namespace())
      self.arm_group = moveit_commander.MoveGroupCommander(arm_group_name, ns=rospy.get_namespace())
      self.display_trajectory_publisher = rospy.Publisher(rospy.get_namespace() + 'move_group/display_planned_path',
                                                    moveit_msgs.msg.DisplayTrajectory,
                                                    queue_size=20)

      if self.is_gripper_present:
        gripper_group_name = "gripper"
        self.gripper_group = moveit_commander.MoveGroupCommander(gripper_group_name, ns=rospy.get_namespace())

      rospy.loginfo("Initializing node in namespace " + rospy.get_namespace())
    except Exception as e:
      print (e)
      self.is_init_success = False
    else:
      self.is_init_success = True 
```

## Terminal output

```
[ INFO] [1626773892.257120602]: Loading robot model 'gen3_lite_gen3_lite_2f'...
[ INFO] [1626773892.261895752]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1626773892.458213776]: Could not identify parent group for end-effector 'end_effector'
[ WARN] [1626773892.478357468]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[ INFO] [1626773893.480244553]: Ready to take commands for planning group arm.
[ INFO] [1626773894.238282431]: Ready to take commands for planning group gripper.
[INFO] [1626773894.241618]: Initializing node in namespace /my_gen3_lite/

```
Thank you for the help in advice!

felixmaisonneuve, 2021-07-20:
Hi @GitNzino,

I tried the example with a gen3 and everything is working correctly for me, so I expect it should work with a gen3_lite too. I will try on a gen3_lite tomorrow.
I get the two warning messages, but everything is working fine and I was still able to open and close the gripper. 

Can you give the output of your driver initialization (`roslaunch kortex_driver kortex_driver.launch arm:=gen3_lite ...`)?

You could make sure the driver has correctly detected the gripper with `rosparam get /my_gen3_lite/is_gripper_present`. It should return `true`. The example is looking at this param to know if the arm has a gripper or not. 

You also need to make sure the example you are testing is within the correct namespace (`/my_gen3_lite/`). If you are running the example provided from its original directory, it should be fine.

Regards,
Felix

felixmaisonneuve, 2021-07-20:
I just tried with a Gen3 lite and everything is working correctly on that arm too

GitNzino, 2021-07-21:
This is my terminal output

```
[INFO] [1626851338.317473]: Opening the gripper...
[ INFO] [1626851338.825768880]: Ready to take commands for planning group gripper.
False
[INFO] [1626851338.838715]: Closing the gripper 50%...
False
[my_gen3_lite/constrain_path-1] process has finished cleanly
log file: /home/lorenzo/.ros/log/5a2a9800-e9f2-11eb-b115-3c950922ab5a/my_gen3_lite-constrain_path-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
And this is my initialization

<launch>

	
    <arg name="robot_name" default="my_gen3_lite"/>
    <arg name="start_delay_seconds" default="0"/>

  	<node name="constrain_path" pkg="kortex_planner" type="constrain_path.py" ns="$(arg robot_name)" output="screen" launch-prefix="bash -c 'sleep $(arg start_delay_seconds); $0 $@' ">
  	</node>

```
</launch><launch>

	<!-- Robot namespace -->
    <arg name="robot_name" default="my_gen3_lite"/>
    <arg name="start_delay_seconds" default="0"/>

  	<node name="constrain_path" pkg="kortex_planner" type="constrain_path.py" ns="$(arg robot_name)" output="screen" launch-prefix="bash -c 'sleep $(arg start_delay_seconds); $0 $@' ">
  	</node>

</launch>
```

I have also putted 6 as degree of freedom as showed in picture below
[image]

GitNzino, 2021-07-21:
### Update

I switched off all the system and the turn on and sometimes it works properly, sometimes not

felixmaisonneuve, 2021-07-21:
When you run the example, you must initialize the arm driver in a first terminal using `roslaunch kortex_driver kortex_driver.launch arm:=gen3_lite ...`. Then, you can run the example in a second terminal window with `rosrun kortex_example moveit_example robot_name:=/my_gen3_lite/`.

What is the full output in your first terminal after initializing the arm driver?

Also, when your example fail and some functions return false, like you have : 
```
[INFO] [1626851338.317473]: Opening the gripper...
[ INFO] [1626851338.825768880]: Ready to take commands for planning group gripper.
False
[INFO] [1626851338.838715]: Closing the gripper 50%...
False
```

Some error messages should appear in your first terminal window (the one that is running the driver). 
Can you add those errors too?

Regards,
Felix

GitNzino, 2021-07-21:
These are the only warnings for the output `roslaunch kortex_driver kortex_driver.launch arm:=gen3_lite` 
[image]

for now the gripper works and i cannot send the error related to the not working gripper on the kortex_driver.launch

felixmaisonneuve, 2021-07-21:
Are you saying everything is working correctly right now?

Other than that, the only thing I see is your namespace, which does not contain '/'. I think I had to put them in the robot_name to make it work. I would need to double check, but I think I used `robot_name:=/my_gen3_lite/` with the '/' at the beginning and at the end. 
In your launch file, it seems to be `robot_name:=my_gen3_lite`. But again, I might be wrong. I don't remember exactly what I used

Other than that, I don't know what could have caused this problem.

GitNzino, 2021-07-21:
Thank you again,
With `robot_name:=/my_gen3_lite` the launch file not works at all meanwhile witout "/" kortex_driver complete the launch procedure.

I'm closing the issue hoping that yesterday not worked for some error in Ubuntu.

Thank you for patience.

closed 2021-07-21 by GitNzino