Kinovarobotics/ros_kortex #176: Node or topic that shows robot input

GitNzino, 2021-07-21:
## Summary

I want to known if there is any node or topic that shows the input given to the robot in terms of joint position and joint velocity

## Use case

I am using moveit planner to obtain some advanced feature in robot movement and collision avoiding, but i do not known what is the input that the planner give to the robot. I can also obtain the feedback that in the case of velocity is with some noise. 

Having a topic with input velocities and position is better to recognize the type of input that the joint is following

felixmaisonneuve, 2021-07-21:
Hi @GitNzino,

If I understand correctly, you want to know the joint velocity and position that is given by the moveit planner. 
`/my_gen3_lite/base_feedback` and `/my_gen3_lite/joint_states` are **not** what you are looking for because it is the actual values measured by the robot and they contain noise, is that correct?

If so, I think you are looking for `/my_gen3_lite/gen3_lite_joint_trajectory_controller/follow_joint_trajectory/goal`. When you "execute" a planning sequence with moveit, it will send a goal. This goal contains a sequence of points for the robot to reach that you can find in goal->trajectory->points (in `/my_gen3_lite/gen3_lite_joint_trajectory_controller/follow_joint_trajectory/goal`). 

The list of points is the sequence the arm will follow. For each point, you will find the position, velocity and acceleration of each joint. 

Now, I am not an expert on how moveit motion planning works. I found this using "Topic monitor" in rqt and I think this is what correspond the most to what you are looking for. Here's a screenshot of what it looks like in rqt : 
[image]

Regards,
Felix

GitNzino, 2021-07-22:
Thank you so much!

I hoped that Gen 3 Lite could give as topic the input message that send to the robot!

closed 2021-07-26 by felixmaisonneuve