Kinovarobotics/ros_kortex #177: errors happen when i roslaunch kortex_examples vision_configuration_cpp.launch

weakenleg, 2021-07-30:
## Description
when i roslaunch kortex_examples vision_configuration_cpp.launch,i meet such errors
[ERROR] [1627628178.329730590]: Failed to call GetIntrinsicParameters
[ERROR] [1627628178.330526905]: Failed to call GetExtrinsicParameters
[ERROR] [1627628178.330947459]: Failed to call GetSensorSettings
[ERROR] [1627628178.331243355]: Failed to call SetSensorSettings
[ERROR] [1627628178.332239464]: Failed to call GetOptionValue

## Version

ROS distribution : melodic

Branch and commit you are using : melodic-devel

## Steps to reproduce

1. *step 1...*
2. *step 2...*

## Code example (if necessary)

```
    # paste code here #
```

## Expected behavior

Describe the expected behavior (in regard to the bug described above).

## Any other information

Any other information you believe the maintainers need to know.

felixmaisonneuve, 2021-07-30:
Hi @weakenleg,

Can you confirm that you are using a Gen3 with a vision module? 
Also, you need to make sure the vision driver has been initialized. 

Can you copy the full log you get after running `roslaunch kortex_driver kortex_driver ...`

Regards,
Felix

weakenleg, 2021-08-01:
I am a novice,I just running roslaunch kortex_driver kortex_driver and change nothing.
the full log
[image]

felixmaisonneuve, 2021-08-02:
Ok,

If you are a novice I would **highly** recommend going throught the multiple readme.md files in the repo. It gives explanations and examples to help you understand how everything works.

If you go in the `ros_kortex/kortex_driver` readme, you will find a [usage](https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel/kortex_driver#usage) section that gives indications on how to launch the driver properly. There is multiple arguments you need to use to specify your robot configuration. 

For example, with a 6dof (6 joints on the robot) without a camera and with a robotiq 85 gripper, you need to use `roslaunch kortex_driver kortex_driver dof:=6 vision:=false gripper:=robotiq_2f_85 ...`

The default launch file considers a gen3 7dof with a vision module and no gripper, so unless this is your configuration, you need to add arguments to your roslaunch command. 

In your case, your driver does not launch correctly (and your arm is not connected). You can see an error message in the left pane of your terminal : 
> The node could not connect to the arm. Did you specify the right IP address and is the arm powered on?

Which brings me to the following question : Did you powered on your arm?
If so, you probably need to add the `ip_address` argument when launching your driver : `roslaunch kortex_driver kortex_driver ip_address:=*your_arm_ip_address*`

Hope this helps,
Felix

weakenleg, 2021-08-02:
Initially,I want to study kortex_examples in the gazebo,so I run `roslaunch kortex_gazebo spawn_kortex_robot.launch`,then I run `roslaunch kortex_examples vision_configuration_cpp.launch`to learn the Vision module configuration examples.According to the readme.md of  kortex_examples, I think I can see the process in the simulation(gazebo),but I find that it seems that the launch file(roslaunch kortex_gazebo spawn_kortex_robot.launch)  does not set the argument of vision,so I meet the errors just like those in the right pane of terminal.
According to your first answer,I think that I should run roslaunch kortex_driver kortex_driver ,then roslaunch kortex_examples vision_configuration_cpp.launch,so i just do so.
Now,according to your second answer, I guess that your meaning is that I should have a real arm then  I can see the process ,or can  I could do some changes in the roslaunch kortex_gazebo spawn_kortex_robot.launch to see the  vision example ,right?

felixmaisonneuve, 2021-08-02:
`roslaunch kortex_driver kortex_driver` does need a real. This command launches the driver and establishes communication with a real robot. 
However, you don't need a real robot when using `roslaunch kortex_gazebo spawn_kortex_robot.launch`. This command creates a gazebo simulation. 

Because of how the simulation driver works, the vision module is never initialized, which is why you cannot run `roslaunch kortex_examples vision_configuration_cpp.launch`.

Now, even if I changed the driver to initialize the vision module, I am pretty sure it wouldn't work because the services related to the vision modules are probably not implemented in the simulation (more details on how the simulation works [here](https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel/kortex_examples#the-ways-to-use-a-kortex-arm-with-ros)). And even if they were implemented, I am not sure what would happen when trying to "take a picture" of a simulated environment. 

Why do you want to use the vision module inside a simulation? 
The vision example analyses the specs of your camera (frame rate, bit rate, resolution, focal point, depth perception, etc). I don't see why you would need those inside a simulation.

Best,
Felix

weakenleg, 2021-08-03:
Thank you for your answer,I totally understand.While,I just made a try to use the vision module inside a simulation.According to the readme.md ,I just thought that vision module could be used in a simulation and I did not watched the file clearly,now, I know the function of it.

felixmaisonneuve, 2021-08-03:
Perfect,

I will assume this answers your question and close this issue. If it is not the case, feel free to reopen it or create a new one

Regards,
Felix

closed 2021-08-03 by felixmaisonneuve