Kinovarobotics/ros_kortex #178: Getting error when building ros_kortex against NVIDIA Jetson- Ubuntu 18/ROS melodic

mgoli-cav, 2021-08-04:
Hello,

I've built the ros_kortex in Ubunut 18/ROS melodic before successfully, however recently when I try to build it again in a different Catkin workspace it gives me the following error:

ERROR: kortex_api_cpp/2.3.0-r.34@kortex/stable: option 'target' doesn't exist
Possible options are ['shared']
CMake Error at ros_kortex/kortex_driver/cmake/conan.cmake:399 (message):
  Conan install failed='1'
Call Stack (most recent call first):
  ros_kortex/kortex_driver/cmake/conan.cmake:485 (conan_cmake_install)
  ros_kortex/kortex_driver/CMakeLists.txt:92 (conan_cmake_run)

-- Configuring incomplete, errors occurred!
See also "/home/largerover/kinova_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/largerover/kinova_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

Could you please help me resolve this issue.
Thank you

felixmaisonneuve, 2021-08-04:
Hi @mgoli-cav,

We now have ROS Noetic support for Ubuntu 20.04. Can you make sure you are using the melodic-devel branch and not the noetic-devel one?

Also, can you make sure your system is fully updated with `sudo apt-get update && sudo apt-get upgrade`?

If I understand correctly, you now have 2 catkin workspaces and the newer one. It might be a good idea to do a `catkin_make clean` before trying to build with `catkin_make` to avoid conflicts between the two workspaces.

Regards,
Felix

mgoli-cav, 2021-08-04:
Thank you for your quick response. I tried your recommended commands, however, I still get the same error, even the catkin_make clean gives me the very same error the catkin_make gave me which is:

ERROR: Missing prebuilt package for 'kortex_api_cpp/2.3.0-r.34@kortex/stable'
Try to build from sources with '--build=kortex_api_cpp'
Use 'conan search <reference> --table table.html'
Or read 'http://docs.conan.io/en/latest/faq/troubleshooting.html#error-missing-prebuilt-package'

CMake Error at ros_kortex/kortex_driver/cmake/conan.cmake:399 (message):
  Conan install failed='1'
Call Stack (most recent call first):
  ros_kortex/kortex_driver/cmake/conan.cmake:485 (conan_cmake_install)
  ros_kortex/kortex_driver/CMakeLists.txt:72 (conan_cmake_run)

-- Configuring incomplete, errors occurred!
See also "/home/largerover/kinova_workspace/build/CMakeFiles/CMakeOutput.log".
See also "/home/largerover/kinova_workspace/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

-- Configuring incomplete, errors occurred!
See also "/home/largerover/kinova_workspace/build/CMakeFiles/CMakeOutput.log".
See also "/home/largerover/kinova_workspace/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

It seems to be an error with Conan.

felixmaisonneuve, 2021-08-05:
Yes. And does any of these commands give an error message?
```
    sudo apt install python3 python3-pip
    sudo python3 -m pip install conan
    conan config set general.revisions_enabled=1
    conan profile new default --detect > /dev/null
    conan profile update settings.compiler.libcxx=libstdc++11 default

    cd catkin_workspace
    rosdep install --from-paths src --ignore-src -y
```

If you have conan already installed, you might need to update it
```
pip3 install --upgrade conan
```

jyang-cpr, 2021-08-05:
Hi,

I am encountering the same issue on a Jetson Xavier platform with Ubuntu 18/ROS Melodic. 
```
ERROR: kortex_api_cpp/2.3.0-r.34@kortex/stable: option 'target' doesn't exist
Possible options are ['shared']
CMake Error at ros_kortex/kortex_driver/cmake/conan.cmake:399 (message):
  Conan install failed='1'
Call Stack (most recent call first):
  ros_kortex/kortex_driver/cmake/conan.cmake:485 (conan_cmake_install)
  ros_kortex/kortex_driver/CMakeLists.txt:92 (conan_cmake_run)

-- Configuring incomplete, errors occurred!
See also "/home/administrator/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/administrator/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

However, reverting to commit `996ffaffd33c135c6888ae624271aa2742fd83d4` allows the driver to be built successfully.

felixmaisonneuve, 2021-08-06:
What version of gcc, cmake and conan are you using?

Also, can one of you give the full output of your `catkin_make` command?

felixmaisonneuve, 2021-08-06:
@jyang-cpr, can you try to change this line and build again : https://github.com/Kinovarobotics/ros_kortex/blob/b11f8d01b8f273909ff9f2cddd6aa7a9c6c47418/kortex_driver/CMakeLists.txt#L97

to 
```
ENV TARGET=jetson
```

@mgoli-cav, if you are trying to build on a different platform than jetson you will want to change the correct line. Basically, you want to change `OPTIONS kortex_api_cpp:target=<target>` to `ENV TARGET=<target>` in the corresponding elseif statement

Let me know how it goes,
Felix

jyang-cpr, 2021-08-06:
@felixmaisonneuve 

Changing that line in `CMakeLists.txt` worked. I'm able to build the driver now. 

Also FYI, if this is still useful, my versions are:
- conan 1.39.0
- cmake 3.10.2
- gcc 7.5.0

Thanks!

felixmaisonneuve, 2021-08-06:
Thank you @jyang-cpr for your help. 

I will update our repo with the modifications and close this issue when it will be merged. 

Felix

mgoli-cav, 2021-08-10:
I just got some time today to try the proposed fix and modifying the CMakeLists.txt worked for me as well. Thank you for the quick resolution. 
The only issue I am facing now is that when I try to test the ros_kortex by running the following command
roslaunch kortex_driver kortex_driver.launch gripper:=robotiq_2f_140
the Kortex_driver execute Okay, but when I try to use the MoveIt plugin to (adding the MotionPlanning in RVIZ) and try to plan and execute a trajectory it gives me the following warnings and errors:
You can start planning now!

[ INFO] [1628615491.489677233]: Kortex Driver's services initialized correctly.
[ INFO] [1628615491.490529117]: -------------------------------------------------
[ INFO] [1628615491.503860260]: The Kortex driver has been initialized correctly!
[ WARN] [1628615499.347121515]: Could not identify parent group for end-effector 'gripper'
[ WARN] [1628615499.536619170]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[ WARN] [1628615500.600971541]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
[ INFO] [1628615621.025165359]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1628615621.026041531]: Planning attempt 1 of at most 1
[ INFO] [1628615621.032057228]: Planner configuration 'arm[RRTConnect]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1628615621.034656175]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1628615621.034922941]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1628615621.035289391]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1628615621.035436087]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1628615621.057782305]: arm[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1628615621.062828705]: arm[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1628615621.064524951]: arm[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1628615621.066154185]: arm[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1628615621.066868237]: ParallelPlan::solve(): Solution found by one or more threads in 0.033146 seconds
[ INFO] [1628615621.067589266]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1628615621.067813597]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1628615621.072308929]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1628615621.076231079]: arm[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1628615621.077142997]: arm[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1628615621.078110982]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1628615621.094119472]: arm[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1628615621.094665388]: arm[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1628615621.095215336]: ParallelPlan::solve(): Solution found by one or more threads in 0.027780 seconds
[ INFO] [1628615621.095792485]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1628615621.096210682]: arm[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1628615621.103381445]: arm[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1628615621.105886724]: arm[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1628615621.106521604]: ParallelPlan::solve(): Solution found by one or more threads in 0.010916 seconds
[ INFO] [1628615621.134536558]: SimpleSetup: Path simplification took 0.027542 seconds and changed from 4 to 2 states
[ INFO] [1628615621.139504265]: Disabling trajectory recording
[ INFO] [1628615621.161566406]: New goal received.
[ INFO] [1628615621.161696748]: Joint Trajectory Goal is accepted.
[ERROR] [1628615621.197838385]: Kortex exception while validating the trajectory
[ERROR] [1628615621.198025659]: Error code: ERROR_PROTOCOL_SERVER

[ERROR] [1628615621.198102335]: Error sub code: UNSUPPORTED_METHOD

[ERROR] [1628615621.198223685]: Error description: Server protocol error, Error sub type=UNSUPPORTED_METHOD => <srv: 2, fct: 235, msgType: 3>
description: Non-parsable payload received from device.

[ INFO] [1628615621.201893855]: Controller gen3_joint_trajectory_controller successfully finished
[ WARN] [1628615621.203096732]: Controller handle gen3_joint_trajectory_controller reports status ABORTED
[ INFO] [1628615621.203299462]: Completed trajectory execution with status ABORTED ...
[ERROR] [1628615622.245758906]: Kortex exception while sending the trajectory
[ERROR] [1628615622.245939555]: Error code: ERROR_PROTOCOL_SERVER

[ERROR] [1628615622.246053385]: Error sub code: UNSUPPORTED_METHOD

[ERROR] [1628615622.246262035]: Error description: Server protocol error, Error sub type=UNSUPPORTED_METHOD => <srv: 2, fct: 226, msgType: 3>
description: Non-parsable payload received from device.

[ERROR] [1628615622.246727947]: To transition to an aborted state, the goal must be in a preempting or active state, it is currently in state: 4
Can you clarify what has been affected in your latest updates?

Thank you

felixmaisonneuve, 2021-08-10:
Hi @mgoli-cav,

Informations about the 2.3 release can be  found [here](https://github.com/Kinovarobotics/ros_kortex/releases/tag/v2.3.0)

About your error, can you move the arm using ros topics?
For example, can you run : 
```
rostopic pub /my_gen3/in/joint_velocity kortex_driver/JointSpeeds "joint_speeds:
- joint_identifier: 0
value: 0.57
duration: 0" 
```

Best,
Felix

mgoli-cav, 2021-08-12:
This works in sending a speed to specific joints(Base_JointSpeeds):

rostopic pub /my_gen3/in/joint_velocity kortex_drive/Base_JointSpeeds "joint_speeds:
- joint_identifier: 6
  value: 0.6
  duration: 0
duration: 0" 

So yes, I can communicate with the arm using ros topics.
Thank you,

felixmaisonneuve, 2021-08-12:
What firmware version are you using on your arm?

I think you might be using incompatible versions between the ros repo, the arm firmware and/or possibly external packages (rviz, ros, ...)

mgoli-cav, 2021-08-12:
I attached a screenshot of the web app version page. If you need other information please let me know where to find the information you need and I will post it here. I am using Ubuntu 18.04 on a Jetson Xavier AGX and the melodic-devel branch for ros_kortex. I did not used to have this issue before. Please see if the following warnings and errors tells you anything about the root cause of the issue. When I add the MoveIt MotionPlanning plugin to RVIZ I get the following warnings:

_[ WARN] [1628792665.481393565]: Could not identify parent group for end-effector 'gripper'
[ WARN] [1628792665.684014032]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[ WARN] [1628792666.771201155]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)_

It plans for a new trajectory just Okay, but when I try to execute the planned trajectory it gives me the following errors:

 _[ INFO] [1628792811.457724436]: Execution request received
[ INFO] [1628792811.498339612]: New goal received.
[ INFO] [1628792811.498638603]: Joint Trajectory Goal is accepted.
[ERROR] [1628792811.503693439]: Kortex exception while validating the trajectory
[ERROR] [1628792811.504008430]: Error code: ERROR_PROTOCOL_SERVER

[ERROR] [1628792811.504313053]: Error sub code: UNSUPPORTED_METHOD

[ERROR] [1628792811.504467748]: Error description: Server protocol error, Error sub type=UNSUPPORTED_METHOD => <srv: 2, fct: 235, msgType: 3>
description: Non-parsable payload received from device.

[ INFO] [1628792811.505719233]: Controller gen3_joint_trajectory_controller successfully finished
[ WARN] [1628792811.506236890]: Controller handle gen3_joint_trajectory_controller reports status ABORTED
[ INFO] [1628792811.506431331]: Completed trajectory execution with status ABORTED ...
[ INFO] [1628792811.506911258]: Execution completed: ABORTED
[ERROR] [1628792812.552674732]: Kortex exception while sending the trajectory
[ERROR] [1628792812.552988763]: Error code: ERROR_PROTOCOL_SERVER

[ERROR] [1628792812.553804130]: Error sub code: UNSUPPORTED_METHOD

[ERROR] [1628792812.554013932]: Error description: Server protocol error, Error sub type=UNSUPPORTED_METHOD => <srv: 2, fct: 226, msgType: 3>
description: Non-parsable payload received from device.

[ERROR] [1628792812.554545286]: To transition to an aborted state, the goal must be in a preempting or active state, it is currently in state: 4_

Thank you,

[image]

felixmaisonneuve, 2021-08-12:
Ok, your arm is using the firmware 2.2. That's the problem. 

The waypoint system (used to move the arm) has been added with the 2.3 release. So your sending waypoints (because your ros_kortex package is up to date with release 2.3) but your arm does not have this logic implemented, so you get a UNSUPPORTED_METHOD error.

You have 2 options.

1. You can upload the firmware 2.3 on your arm. The swu file you need is available on our website in the ressources [section](https://www.kinovarobotics.com/en/resources/gen3-technical-resources). Here's the [link](https://drive.google.com/file/d/1WTveCJwUMup6ZdkuID9yA_Etss0wK2Qm/view?usp=sharing) of the swu file. You can upload it on your arm from the webapp page in your screenshot. (recommended)

2. You can revert your ros_kortex package to any commit before the release 2.3 on github. `996ffaffd33c135c6888ae624271aa2742fd83d4` is the latest commit before releasing 2.3.

Regards,
Felix