Kinovarobotics/ros_kortex #182: ABORTED:Solution found but controller failed during execution

heiyumiao, 2021-08-13:
I only used the kinova_driver package in  version 2.3,however,after a period of testing,this problem occurs again,which means this problem haven’t been solved.Do I need to use the full package in version 2.3?

felixmaisonneuve, 2021-08-13:
Hi @heiyumiao,

Took me some time to figure out what you were talking about, but I found the issue https://github.com/Kinovarobotics/ros_kortex/issues/131

From what you are saying, you are using an up-to-date branch (probably melodic-devel)? Or you are at least on a commit after the 2.3 release (commit `f0218097604a902cf8bfc65db64ef3aa5eefb455`).

Are you also using the 2.3 firmware version on your arm? 

Finally, can you give a full log of the error message you get?

Regards,
Felix

heiyumiao, 2021-08-15:
Hi @felixmaisonneuve ,
I have update my firmware version to 2.3,the last problem has disappeared,but a new problem occurs:
The details of the problem are shown as follows:

**My code**:

 vertical_angle =[-0.0656953904023,0.69399547873,-3.01732023606,
-0.539136722662,3.11725455702,1.87381106194,-1.35760388995]
 arm_group = rm.arm_group
 success = True
 joint_positions = arm_group.get_current_joint_values()
 rm.arm_group.set_goal_joint_tolerance(0.01)
 for i in range(0,7):
       joint_positions[i] = vertical_angle[i]
 arm_group.set_joint_value_target(joint_positions)
 success &= arm_group.go(wait=True)

**The errors occur**:

### this is the error in my demo file

[ INFO] [1629031922.047139176]: ABORTED: Timeout reached. 

### this is the errors occur in kinova_driver.launch

[ERROR] [1629031927.649204642]: Notification mismatch : received ACTION_PREPROCESS_ABORT but we are in PRE_PROCESSING_PENDING
[ERROR] [1629031941.543043084]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 16.134000 seconds). Stopping trajectory.
[ INFO] [1629031941.543079096]: Cancelling execution for gen3_joint_trajectory_controller
[ INFO] [1629031941.543106240]: Completed trajectory execution with status TIMED_OUT ...
[ INFO] [1629031944.712994939]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1629031944.713063153]: Planning attempt 1 of at most 1
[ INFO] [1629031944.713504991]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1629031944.713668603]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1629031944.726403911]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1629031944.726435745]: Solution found in 0.012839 seconds
[ INFO] [1629031944.744197233]: SimpleSetup: Path simplification took 0.017720 seconds and changed from 4 to 2 states
[ INFO] [1629031944.748858803]: Interpolated time exceeds original trajectory (quitting), original: 7.02185 final interpolated time: 7.022
[ INFO] [1629031944.748887888]: Uniform sampling, resample duraction: 0.001 input traj. size: 71 output traj. size: 7023
[ INFO] [1629031944.980206308]: New goal received.
[ WARN] [1629031944.980273082]: There is already an active goal. It is being cancelled.
[ INFO] [1629031944.980287969]: Calling Stop on the robot.
[ERROR] [1629031945.026178421]: Notification mismatch : received ACTION_ABORT but we are in PRE_PROCESSING_PENDING
[ INFO] [1629031946.023566972]: Joint Trajectory Goal is accepted.
[ INFO] [1629031946.537541163]: State changed from PRE_PROCESSING_PENDING to PRE_PROCESSING_PENDING

felixmaisonneuve, 2021-08-16:
Hi @heiyumiao,

When you run your code, does your arm move towards the destination (very slowly) or does it stay motionless?
Also, can you provide the full log of your `kinova_driver.launch`? 

Regards,
Felix

heiyumiao, 2021-08-17:
Hi @felixmaisonneuve ,

Firstly,when I run my code,my arm stay motionless!Before I update the firmware version,my arm can reach this pose without any problems.

Secondly,the full log of my kinova_driver.launch is show as follows:

You can start planning now!

[ INFO] [1629213527.692679070]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1629213527.693061070]: Planning attempt 1 of at most 1
[ INFO] [1629213527.695442443]: Using a sample_duration value of 0.001
[ INFO] [1629213527.699854896]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1629213527.702483207]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1629213527.716841236]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1629213527.716874038]: Solution found in 0.014619 seconds
[ INFO] [1629213527.730599296]: SimpleSetup: Path simplification took 0.013642 seconds and changed from 4 to 2 states
[ INFO] [1629213527.735318119]: Interpolated time exceeds original trajectory (quitting), original: 7.14344 final interpolated time: 7.144
[ INFO] [1629213527.735351964]: Uniform sampling, resample duraction: 0.001 input traj. size: 71 output traj. size: 7145
[ INFO] [1629213527.970369075]: New goal received.
[ INFO] [1629213528.992635350]: Joint Trajectory Goal is accepted.
[ INFO] [1629213529.656507229]: State changed from IDLE to PRE_PROCESSING_PENDING

[ERROR] [1629213530.166702557]: Notification mismatch : received ACTION_PREPROCESS_ABORT but we are in PRE_PROCESSING_PENDING
[ERROR] [1629213544.256953453]: Controller is taking too long to execute trajectory (the expected upper bound for the trajectory execution was 16.288000 seconds). Stopping trajectory.
[ INFO] [1629213544.257009723]: Cancelling execution for gen3_joint_trajectory_controller
[ INFO] [1629213544.257078019]: Completed trajectory execution with status TIMED_OUT ...
[ INFO] [1629213553.504559332]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1629213553.504625018]: Planning attempt 1 of at most 1
[ INFO] [1629213553.505053597]: Planner configuration 'arm' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1629213553.505201699]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1629213553.528463843]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1629213553.528517032]: Solution found in 0.023379 seconds
[ INFO] [1629213553.538952130]: SimpleSetup: Path simplification took 0.010392 seconds and changed from 4 to 2 states
[ INFO] [1629213553.544238640]: Interpolated time exceeds original trajectory (quitting), original: 7.02739 final interpolated time: 7.028
[ INFO] [1629213553.544269138]: Uniform sampling, resample duraction: 0.001 input traj. size: 71 output traj. size: 7029
[ INFO] [1629213553.785874493]: New goal received.
[ WARN] [1629213553.785994004]: There is already an active goal. It is being cancelled.
[ INFO] [1629213553.786013155]: Calling Stop on the robot.
[ERROR] [1629213553.817518608]: Notification mismatch : received ACTION_ABORT but we are in PRE_PROCESSING_PENDING
[ INFO] [1629213554.815744264]: Joint Trajectory Goal is accepted.
[ INFO] [1629213555.478724692]: State changed from PRE_PROCESSING_PENDING to PRE_PROCESSING_PENDING

Regards,
Heiyumiao

felixmaisonneuve, 2021-08-17:
Hi @heiyumiao,

What commit and what branch are you using? 
Can you make sure your branch is up to date with `git pull`? Also, make sure you recompile the changes with `catkin_make` in your catkin workspace.

heiyumiao, 2021-08-18:
Hi @felixmaisonneuve ,

I have tried to use  ros_kortex-kinetic-devel package and ros_kortex-2.3.0 package,and I am sure that I have recompile the changes with catkin_make in my catkin workspace.

What’s more,I found another problem.When I use the follow-up control,the arm can’t move when it reach to a certain pose,,which mean I can’t move the arm to the ideal pose even by follow-up control.The pose I mentioned about can be reached before I update the firmware version.Is there any limits in firmware version 2.3?

JingKangSun, 2021-08-18:
Hi @felixmaisonneuve 

I am the colleague of @heiyumiao ,after the firmware version update to 2.3.0,we have met a lot of problems!

Firstly,as @heiyumiao said,we can't use the kinova_driver to reanch to a lot poses.Not only the pose he listed above,but also some poses we predefined before.

Secondly,the follow-up control do have some problems.When we push the follow-up button,but don't exert force on it,the arm will move downward.What’s more,the follow-up button will change to the left one after the arm reach to a certain pose,before we update the firmware version,the follow-up button always be the right one.For these problems,we record two videoes.This is the link of our videoes:https://github.com/JingKangSun/video. 

Thirdly, the rgbd driver of the camera on the arm have some problem.We need to try reboot some times or wait for a long time to link the depth stream and color stream.There is also a picture of the problem:https://github.com/JingKangSun/video.

Regards,
JingKang

felixmaisonneuve, 2021-08-18:
For me to try to reproduce your error, I will need to know 
- The commit your branch is on (I now know you are using kinetic-devel)
- The firmware version that is on your robot

You say the arm falls when you press the follow-up button. This might be because of the gripper. In the webapp, can you make sure the gripper is correctly configured (in the *Configuration->Robot* page). I feel like the arm does not know it has a gripper and does not compensate for the weight of said gripper, so the arm is falling. 

For the poses you can't seem to reach, do you have some protection zones configured on the arm? In the webapp, it is possible to set protection zones that the arm will avoid. 

I think what you could do is a factory reset. To do that, you need to hold the power button for 10-15s. It will erase everything in your arm's database, so be sure to export your work (actions and protection zones) if you want to add them again, otherwise you will have to recreate everything. Also, if you do that, you will need to configure the gripper again (as well as any other configurations that you have previously made). 
[image]

About the vision module, you need to use our other repository to use all of its features : https://github.com/Kinovarobotics/ros_kortex_vision

you have to add this ROS package within ros_kortex and recompile everything with `catkin_make` (instructions are detailed in the main readme). If you already have that repo, can you tell me what branch you are on for it too?

Regards,
Felix

RaduCorcodel, 2021-08-19:
Hi @felixmaisonneuve,

I confirm that the vision module on our robot behaves the same. After shutting down the kinova_vision launcher and relaunching, it happens more than often that it cannot start the depth and color streams. Some times it finally starts the streams after the 6-7th attempt. Other times we need to restart the robot for the vision to work. Any advice on how to fix this would be great. I looked at the source code but nothing stands out to what might be causing it

Best,
Radu

JingKangSun, 2021-08-19:
Hi @felixmaisonneuve ,

Under your guidance,I did some attempts.

Firstly,for the problem of kinova_driver.launch,I tried to recover the firmware version to 2.2.0,though the arm can't move to the ideal pose,this version provides a clear error log:

[image]

I tried to change the speed limits to max,then I can use the arm normally.Inspired by version 2.2.0,I did the same thing in version 2.3.0,this change looks useful!

Secondly,for the  kinova_vision launcher,I found that the reason of can't link the color stream and depth stream maybe  the network proxy has been modified.However,as @RaduCorcodel  said , after shutting down the kinova_vision launcher ,the kinova_vision launcher can't be relaunching directly,some times it finally starts the streams after the 10-11th attempt in my environment.

Thirdly,for the problem of follow-up control,I checked the configuration in webapp,and I thought the gripper is correctly configured,the picture blow is my configuration:

[image]

However,when I push the follow-up button,the arm moves up automatically.The gripper parameters seem like do nothing:

[image]

I also tried to do a factory reset,but it didn't work.

Regards,
JingKang

felixmaisonneuve, 2021-08-19:
Please do a `git log` inside your ros_kortex directory and tell me what the hash is.
[image]

JingKangSun, 2021-08-19:
Hi @felixmaisonneuve 

[image]

[image]

[image]

[image]

[image]

[image]

[image]

felixmaisonneuve, 2021-08-19:
Thank you @JingKangSun,

You are using an old version of ROS that is not compatible with the 2.3 firmware that is on your robot. 

I now need to know two things : 
1. What version of Ubuntu are you using? (16.04, 18.04, 20.04?)
2. What version of ROS are you using? (Kinetic, Melodic, Noetic?)

heiyumiao, 2021-08-19:
Hi, @felixmaisonneuve 

I use Ubuntu16.04 and Ros kinetic

JingKangSun, 2021-08-19:
Hi @felixmaisonneuve,

I use Ubuntu16.04 and Ros kinetic

felixmaisonneuve, 2021-08-19:
Thank you, 

The next step is to update your ros_kortex. To do that, go in your ros_kortex folder and do : 
```
git pull
git checkout melodic-devel
```
This will change your current git branch to `melodic-devel`, we also have a `kinetic-devel` branch, but `melodic-devel` can be used for ROS Kinetic without problems. 

After changing your branch, just to make sure everything is updated correctly, do another 
```
git pull
```

By that point, if you do a `git log`, the first hash on top should start with `2e4a57...`. 

After that, go into your catkin_workspace directory and do a `catkin_make`. 

Make sure your robot's firmware version is 2.3 and test again. I think if everything went well it should fix your issue

Regards,
Felix

JingKangSun, 2021-08-20:
Hi @felixmaisonneuve ,

I just follow your guidance , but when I do catkin_make in my workspace , there occur an error:

[image]

I wonder to know whether the new version doesn't need this .h file or  the new version doesn't contain this .h file .

Regards,
JingKang

felixmaisonneuve, 2021-08-20:
I think something went wrong in the git commands. 

What does `git status` says?

JingKangSun, 2021-08-21:
Hi @felixmaisonneuve ,

**git status** says:

[image]

[image]

...

[image]

**git log** says:

[image]

**git pull** says:

[image]

Regards,
JingKang

felixmaisonneuve, 2021-08-23:
Basically `git status` says your entire kortex_driver is deleted. `git checkout -- .` will bring it back. If you have moved your kortex_driver in another directory, I would suggest to delete it (unless you have made modifications on it) because it will probably be outdated. 

Other than that, your repo is using a recent commit (past 2.3 release), which is what we wanted. Once we get things working, it should fix a couple issue (related to uncompatible versions)

Just a recap, you should do : 
1. `git checkout -- .` (to bring back your an updated kortex_driver folder)
2. `catkin_make`
3. (if catkin_make builds correctly) try your examples and see ifit still gives errors

JingKangSun, 2021-08-23:
Hi @felixmaisonneuve ,

I checked out the file folder in the **ros_kortex**,and I found that the **kortex_driver** is not deleted.In fact，I have followed the hint of  **git pull** to delete some file in kortex_driver.

[image]

Also,I tried to follow your guidance to do  **git checkout -- .**,but this seems did nothing.

[image]

Regards,
JingKang

JingKangSun, 2021-08-24:
Hi @felixmaisonneuve ,

I have solved the problem which I mentioned above.And I have update my branch to **melodic-devel** successfully.

However , the problem of follow-up control doesn't be solved. When I push the follow-up button , the arm moves up automatically. The gripper parameters seem like do nothing .

Regards,
JingKang

felixmaisonneuve, 2021-08-24:
Hi @JingKangSun,

That's good.

For the follow-up control, can you juste make sure of 2 things in the Web application.
1. The gripper is correctly set
2. There is no payload configured

[image]

JingKangSun, 2021-08-24:
Hi @felixmaisonneuve ,

I am sure that I have satisfied the 2 things you mentioned about. And I cut some pictures about my **gripper config**,**firmware version** and **ros_kortex version**.

[image]

[image]

[image]

Regards,
JingKang

felixmaisonneuve, 2021-08-24:
Hi @JingKangSun,

Since you already have tried a factory reset, maybe check that the gravity vector is correctly set in the Web app (same page as the payload and gripper config). You should have -9.81 in the z-axis and 0 in both x-axis and y-axis. 

Other than that, I don't think this error is related to ros_kortex, it is probably something with your specific arm or a wrong Kortex configuration. 

I think you should contact our support team at support@kinova.ca, they will be in a better spot to help you out with this than me.
Make sure to include the link of your videos and maybe that last screenshot with all your firmware versions. 

Regards,
Felix

JingKangSun, 2021-08-25:
Hi @felixmaisonneuve ,

I have checked the gravity vector set in Web app,and the vector is set correctly.

I will try to contact the support team.Thank you so much for your help for so long.

Regards,
JingKang