Kinovarobotics/ros_kortex #188: Setting prefix breaks joint_state_publisher

phuicy, 2021-09-06:
## Description

If a prefix is set, the joint_state_publisher publishes all zeroes, base_feedback has correct values. joint_state_publisher seems to be confused about the joints to copy.

## Version

ROS distribution : Noetic

Branch and commit you are using : master

## Steps to reproduce

1. Set prefix in the launch
2. Launch and echo joint_state

## Expected behavior

joint_state_publisher to republish base_feedback/joint_state 

## Any other information

This breaks all moveit features.

phuicy, 2021-09-06:
Actually the prefix seems to be applied twice somewhere: 
[image]
The prefix in the lanuch ```<arg name="prefix" default="prefix_"/>```

phuicy, 2021-09-06:
https://github.com/Kinovarobotics/ros_kortex/blob/a334ce2a9fcbec12b27012f77183a81a75ec6e6e/kortex_driver/src/non-generated/driver/kortex_arm_driver.cpp#L655

and

https://github.com/Kinovarobotics/ros_kortex/blob/a334ce2a9fcbec12b27012f77183a81a75ec6e6e/kortex_driver/src/non-generated/driver/kortex_arm_driver.cpp#L666

These line seems to apply the prefix for a second time.

phuicy, 2021-09-06:
Ok, that has fixed the joint_state_publisher. but robot_state_publisher still not outputting tf.

felixmaisonneuve, 2021-09-07:
Hi @phuicy,

Using a prefix will break MoveIt!. If you want to use Moveit with a prefix, you will need to change its configuration and the according `.yaml` file. This is mentionned in the [readme](https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel/kortex_driver#usage) inside the `prefix` bullet point. 

This is annoying and this is something that would be nice to "fix", I just did not have the time to do it yet. 

However, I don't think I have seen the prefix applied twice before. Have you changed anything in your configs?
Also, the two lines you have mentionned in `kortex_arm_driver.cpp` are not duplicating the prefix. The first puts the prefix for the actuators joints and the second one is putting the prefix for the gripper motors.

Regards,
Felix

yukizakisoh, 2021-09-17:
I also faced the same symptom: the prefix appeared twice in base_feedback/joint_state.
Then I did the following modifications as @phuicy did, and it improved the situation.

> https://github.com/Kinovarobotics/ros_kortex/blob/a334ce2a9fcbec12b27012f77183a81a75ec6e6e/kortex_driver/src/non-generated/driver/kortex_arm_driver.cpp#L655
>
> and [...]

In my environment, robot_state_publisher now works correctly, and moveit also works correctly without changing .yaml files.

## Version
ROS distribution : melodic
Branch and commit you are using : melodic-devel
using a real robot

felixmaisonneuve, 2021-09-17:
Hi @phuicy and @yukizakisoh,

I tried the fix and I now understand what you meant. Like you both said, this will fix the prefix that is duplicated in `joint_state`. I created PR #194 about it. 
This was a known bug that was preventing moveit from working when using a prefix. I did not have to time to look into yet, so thank you for pointing it out to me. 

@phuicy you were also mentionning `robot_state_publisher` is still not outputting correctly. Can you elaborate on that?

Regards,
Felix

phuicy, 2021-09-18:
robot_state_publisher needed to pointed at the namespaced topic explicitly.

felixmaisonneuve, 2021-09-20:
Hi @phuicy,

Can you please explain a bit more? 

After changing `kortex_arm_driver` to remove duplicated prefix, is there still something wrong? Should I update something else in our arm driver? 

You seemed to have fix the issue, but I am not sure what the problem was in the first place.
Can you give more details?

Thank You,
Felix

phuicy, 2021-10-19:
https://github.com/Kinovarobotics/ros_kortex/blob/noetic-devel/kortex_driver/launch/kortex_driver.launch#L106

Needs to be adapted to point Parameters robot_description (urdf map) and tf_prefix to the correct names.