Kinovarobotics/ros_kortex #189: can't launch kortex_driver correctly

yuweiDu, 2021-09-10:
When I execute command line 'roslaunch kortex_driver kortex_driver.launch start_rviz:=false  arm:=gen3_lite', I always get this error:
[my_gen3_lite/my_gen3_lite_driver-2] process has died [pid 2388, exit code -6, cmd /home/dyw/catkin_ws/devel/lib/kortex_driver/kortex_arm_driver __name:=my_gen3_lite_driver __log:=/home/dyw/.ros/log/4516bf80-11dd-11ec-ac28-3aa1235eab48/my_gen3_lite-my_gen3_lite_driver-2.log].
log file: /home/dyw/.ros/log/4516bf80-11dd-11ec-ac28-3aa1235eab48/my_gen3_lite-my_gen3_lite_driver-2*.log

and kortex_examples can't run correctly. 

ROS distribution :  melodic
OS:ubuntu 18.04
evironment: miniconda base

felixmaisonneuve, 2021-09-10:
Hi @yuweiDu,

Can you provide the full output of `roslaunch kortex_driver kortex_driver.launch start_rviz:=false arm:=gen3_lite`?

Regards,
Felix

yuweiDu, 2021-09-11:
> Hi @yuweiDu,
> 
> Can you provide the full output of `roslaunch kortex_driver kortex_driver.launch start_rviz:=false arm:=gen3_lite`?
> 
> Regards,
> Felix

... logging to /home/dyw/.ros/log/7839287c-12b3-11ec-ac28-3aa1235eab48/roslaunch-dyw-X550JX-32231.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://dyw-X550JX:33733/

SUMMARY
========

PARAMETERS
 * /my_gen3_lite/hztest_test_base_feedback/hz: 40
 * /my_gen3_lite/hztest_test_base_feedback/hzerror: 5.0
 * /my_gen3_lite/hztest_test_base_feedback/test_duration: 30.0
 * /my_gen3_lite/hztest_test_base_feedback/topic: base_feedback
 * /my_gen3_lite/hztest_test_base_feedback/wait_time: 10.0
 * /my_gen3_lite/hztest_test_driver_joint_state/hz: 40
 * /my_gen3_lite/hztest_test_driver_joint_state/hzerror: 5.0
 * /my_gen3_lite/hztest_test_driver_joint_state/test_duration: 30.0
 * /my_gen3_lite/hztest_test_driver_joint_state/topic: base_feedback/joi...
 * /my_gen3_lite/hztest_test_driver_joint_state/wait_time: 10.0
 * /my_gen3_lite/hztest_test_joint_state_publisher_joint_states/hz: 40
 * /my_gen3_lite/hztest_test_joint_state_publisher_joint_states/hzerror: 5.0
 * /my_gen3_lite/hztest_test_joint_state_publisher_joint_states/test_duration: 30.0
 * /my_gen3_lite/hztest_test_joint_state_publisher_joint_states/topic: joint_states
 * /my_gen3_lite/hztest_test_joint_state_publisher_joint_states/wait_time: 10.0
 * /my_gen3_lite/joint_state_publisher/rate: 40
 * /my_gen3_lite/joint_state_publisher/source_list: ['base_feedback/j...
 * /my_gen3_lite/kortex_driver_tests/api_connection_inactivity_timeout_ms: 20000
 * /my_gen3_lite/kortex_driver_tests/api_rpc_timeout_ms: 2000
 * /my_gen3_lite/kortex_driver_tests/api_session_inactivity_timeout_ms: 35000
 * /my_gen3_lite/kortex_driver_tests/arm: gen3_lite
 * /my_gen3_lite/kortex_driver_tests/cyclic_data_publish_rate: 40
 * /my_gen3_lite/kortex_driver_tests/default_goal_time_tolerance: 0.5
 * /my_gen3_lite/kortex_driver_tests/default_goal_tolerance: 0.005
 * /my_gen3_lite/kortex_driver_tests/dof: 6
 * /my_gen3_lite/kortex_driver_tests/gripper: gen3_lite_2f
 * /my_gen3_lite/kortex_driver_tests/gripper_joint_limits_max: [-0.09]
 * /my_gen3_lite/kortex_driver_tests/gripper_joint_limits_min: [0.96]
 * /my_gen3_lite/kortex_driver_tests/gripper_joint_names: ['right_finger_bo...
 * /my_gen3_lite/kortex_driver_tests/ip_address: 192.168.1.10
 * /my_gen3_lite/kortex_driver_tests/joint_names: ['joint_1', 'join...
 * /my_gen3_lite/kortex_driver_tests/maximum_accelerations: [1.0, 0.5, 0.4, 1...
 * /my_gen3_lite/kortex_driver_tests/maximum_velocities: [0.5, 0.5, 0.5, 0...
 * /my_gen3_lite/move_group/allow_trajectory_execution: True
 * /my_gen3_lite/move_group/arm/default_planner_config: RRTConnect
 * /my_gen3_lite/move_group/arm/longest_valid_segment_fraction: 0.005
 * /my_gen3_lite/move_group/arm/planner_configs: ['SBL', 'EST', 'L...
 * /my_gen3_lite/move_group/arm/projection_evaluator: joints(joint_1,jo...
 * /my_gen3_lite/move_group/capabilities: 
 * /my_gen3_lite/move_group/controller_list: [{'default': True...
 * /my_gen3_lite/move_group/disable_capabilities: 
 * /my_gen3_lite/move_group/gripper/default_planner_config: RRTConnect
 * /my_gen3_lite/move_group/gripper/planner_configs: ['SBL', 'EST', 'L...
 * /my_gen3_lite/move_group/jiggle_fraction: 0.05
 * /my_gen3_lite/move_group/joint_state_controller/publish_rate: 50
 * /my_gen3_lite/move_group/joint_state_controller/type: joint_state_contr...
 * /my_gen3_lite/move_group/max_safe_path_cost: 1
 * /my_gen3_lite/move_group/moveit_controller_manager: moveit_simple_con...
 * /my_gen3_lite/move_group/moveit_manage_controllers: True
 * /my_gen3_lite/move_group/planner_configs/BFMT/balanced: 0
[... 219 lines cut ...]
Planning scene monitors started.
[ INFO] [1631332259.782373598]: Initializing OMPL interface using ROS parameters
[ INFO] [1631332259.804512893]: Using planning interface 'OMPL'
[ INFO] [1631332259.806992597]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1631332259.807396748]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1631332259.807682668]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1631332259.808029826]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1631332259.808343393]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1631332259.808663217]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1631332259.808724924]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1631332259.808789743]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1631332259.808819047]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1631332259.808846777]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1631332259.808870846]: Using planning request adapter 'Fix Start State Path Constraints'
[ WARN] [1631332264.823179799]: Waiting for gen3_lite_joint_trajectory_controller/follow_joint_trajectory to come up
[ WARN] [1631332270.823513731]: Waiting for gen3_lite_joint_trajectory_controller/follow_joint_trajectory to come up
[... cut ...]
[ WARN] [1631332281.853271821]: Waiting for gen3_lite_2f_gripper_controller/gripper_cmd to come up
[ WARN] [1631332287.853529505]: Waiting for gen3_lite_2f_gripper_controller/gripper_cmd to come up
[ERROR] [1631332293.853836119]: Action client not connected: gen3_lite_2f_gripper_controller/gripper_cmd
[ INFO] [1631332293.872302806]: Returned 0 controllers in list
[ INFO] [1631332293.883597281]: Trajectory execution is managing controllers
Loading 'move_group/ApplyPlanningSceneService'...
Loading 'move_group/ClearOctomapService'...
Loading 'move_group/MoveGro

yuweiDu, 2021-09-11:
> Hi @yuweiDu,
> 
> Can you provide the full output of `roslaunch kortex_driver kortex_driver.launch start_rviz:=false arm:=gen3_lite`?
> 
> Regards,
> Felix

Loading 'move_group/ApplyPlanningSceneService'...
Loading 'move_group/ClearOctomapService'...
Loading 'move_group/MoveGroupCartesianPathService'...
Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
Loading 'move_group/MoveGroupGetPlanningSceneService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...
[ INFO] [1631332293.928083064]: 

********************************************************
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1631332293.928153588]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1631332293.928187475]: MoveGroup context initialization complete

You can start planning now!

felixmaisonneuve, 2021-09-13:
Hi @yuweiDu,

I see 
>what(): Server protocol error, Error sub type=UNSUPPORTED_METHOD => <srv: 16, fct: 18, msgType: 3>
>description: Non-parsable payload received from device.

I will need 2 things : 
1. Your arm firmware version. (you can find it in the WebApp in *Sytems*->*Upgrade*)
2. The commit you are using. 

I think there might be an incompatibility bitween the 2.

Regards,
Felix