Kinovarobotics/ros_kortex #190: launch kortex_dual_driver.launch incorrectly

jason134017, 2021-09-13:
## Description

When I launch kortex_dual_driver.launch, use command 'roslaunch kortex_driver kortex_dual_driver.launch 
 left_ip_address:=192.168.1.10 left_arm:=gen3_lite left_gripper:=gen3_lite_2f right_ip_address:=192.168.1.100 right_prefix:=right_ right_arm:=gen3 right_dof:=6 right_gripper:=robotiq_2f_85',
I get this error:

RLException: [/home/airobot/test/src/ros_kortex/kortex_driver/launch/kortex_dual_driver.launch] requires the 'prefix' arg to be set
The traceback for the exception was written to the log file.

If I try to set prefix in the launch, it can be start but the arm can't be configured.

## Version

ROS distribution : melodic

Branch and commit you are using : melodic-devel

## Steps to reproduce

1. *Use example command in the launch*
2. *Try to set prefix in the launch*

felixmaisonneuve, 2021-09-13:
Hi @jason134017,

I think this is because your `left_prefix` argument is not set. 

Best,
Felix

jason134017, 2021-09-13:
I got the same problem after adding this parameter to the command . The parameter has a default value in the kortex_dual_driver.launch. The error given in the log seems to require the `prefix` parameter to be set.

felixmaisonneuve, 2021-09-13:
Looking into it, I realised there is no `prefix` argument in kortex_dual_driver.launch. Did you make any modifications to the launch file? 
Also, what commit are you using?

jason134017, 2021-09-14:
I didn't  make any modifications to the launch.  It is not written in kortex_driver/readme.md that the `prefix` needs to be set.
If the `prefix` is not set, it cannot be started, if the `prefix` is set to a non-empty value
It can be started but the arm state can't be initialized.

felixmaisonneuve, 2021-09-14:
Can you give the full output after you run `roslaunch kortex_driver kortex_dual_driver.launch`?

jason134017, 2021-09-15:
When use command 'roslaunch kortex_driver kortex_dual_driver.launch  left_prefix:=left_ left_ip_address:=192.168.1.10 left_arm:=gen3_lite left_gripper:=gen3_lite_2f right_prefix:=right_ right_ip_address:=192.168.1.100  right_arm:=gen3 right_dof:=6 right_gripper:=robotiq_2f_85',the log file is:

```
[roslaunch][INFO] 2021-09-14 21:02:12,436: Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2021-09-14 21:02:12,440: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2021-09-14 21:02:12,440: roslaunch starting with args ['/opt/ros/melodic/bin/roslaunch', 'kortex_driver', 'kortex_dual_driver.launch', 'left_prefix:=left_', 'left_ip_address:=192.168.1.10', 'left_arm:=gen3_lite', 'left_gripper:=gen3_lite_2f', 'right_prefix:=right_', 'right_ip_address:=192.168.1.100', 'right_arm:=gen3', 'right_dof:=6', 'right_gripper:=robotiq_2f_85']
[roslaunch][INFO] 2021-09-14 21:02:12,440: roslaunch env is {'LC_NUMERIC': 'lzh_TW', 'CONDA_PYTHON_EXE': '/home/airobot/anaconda3/bin/python', 'ROS_DISTRO': 'melodic', 'DBUS_STARTER_ADDRESS': 'unix:path=/run/user/1000/bus,guid=162c8e214e053d649d818fa961409bd6', 'QT4_IM_MODULE': 'xim', 'PKG_CONFIG_PATH': '/home/airobot/test/devel/lib/pkgconfig:/opt/ros/melodic/lib/pkgconfig', '_CE_M': '', 'XDG_CURRENT_DESKTOP': 'ubuntu:GNOME', 'LC_PAPER': 'lzh_TW', 'ROSLISP_PACKAGE_DIRECTORIES': '/home/airobot/test/devel/share/common-lisp', 'QT_IM_MODULE': 'ibus', 'LOGNAME': 'airobot', 'USER': 'airobot', 'ROS_PACKAGE_PATH': '/home/airobot/test/src:/opt/ros/melodic/share', 'XDG_VTNR': '2', 'HOME': '/home/airobot', 'PATH': '/opt/ros/melodic/bin:/home/airobot/anaconda3/condabin:/usr/local/cuda-11.1/bin:/home/airobot/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin', 'CMAKE_PREFIX_PATH': '/home/airobot/test/devel:/opt/ros/melodic', 'DISPLAY': ':1', 'SSH_AGENT_PID': '2929', 'LANG': 'en_US.UTF-8', 'CONDA_SHLVL': '0', 'SHELL': '/bin/bash', 'ROS_LOG_FILENAME': '/home/airobot/.ros/log/fa1ae2f4-155b-11ec-9e91-d43b04ff823c/roslaunch-airobot-RobticArm-9881.log', 'XAUTHORITY': '/run/user/1000/gdm/Xauthority', 'JOURNAL_STREAM': '9:43393', 'SESSION_MANAGER': 'local/airobot-RobticArm:@/tmp/.ICE-unix/2834,unix/airobot-RobticArm:/tmp/.ICE-unix/2834', 'LC_MEASUREMENT': 'lzh_TW', 'QT_ACCESSIBILITY': '1', 'LD_LIBRARY_PATH': '/home/airobot/test/devel/lib:/opt/ros/melodic/lib:/usr/local/lib:/home/airobot/Qt5.14.2/5.14.2/gcc_64/lib', 'LC_ADDRESS': 'lzh_TW', 'CLUTTER_IM_MODULE': 'xim', 'TEXTDOMAIN': 'im-config', '_': '/opt/ros/melodic/bin/roslaunch', 'TERM': 'xterm-256color', 'ROS_MASTER_URI': 'http://localhost:11311', 'XDG_RUNTIME_DIR': '/run/user/1000', 'GPG_AGENT_INFO': '/run/user/1000/gnupg/S.gpg-agent:0:1', 'ROS_PYTHON_VERSION': '2', 'INVOCATION_ID': '689266f005954b0bbf6ab2cda28a51e7', 'USERNAME': 'airobot', 'XDG_SESSION_DESKTOP': 'ubuntu', '_CE_CONDA': '', 'ROS_VERSION': '1', 'GTK_IM_MODULE': 'ibus', 'CUDA_HOME': '/usr/local/cuda-11.1', 'PYTHONPATH': '/home/airobot/test/devel/lib/python2.7/dist-packages:/opt/ros/melodic/lib/python2.7/dist-packages', 'SSH_AUTH_SOCK': '/run/user/1000/keyring/ssh', 'SHLVL': '1', 'MANAGERPID': '2764', 'ROS_ROOT': '/opt/ros/melodic/share/ros', 'GDMSESSION': 'ubuntu', 'XMODIFIERS': '@im=ibus', 'LC_TIME': 'lzh_TW', 'TEXTDOMAINDIR': '/usr/share/locale/', 'GNOME_SHELL_SESSION_MODE': 'ubuntu', 'XDG_DATA_DIRS': '/usr/share/ubuntu:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop', 'XDG_CONFIG_DIRS': '/etc/xdg/xdg-ubuntu:/etc/xdg', 'DBUS_STARTER_BUS_TYPE': 'session', 'LESSOPEN': '| /usr/bin/lesspipe %s', 'XDG_SESSION_ID': '5', 'DBUS_SESSION_BUS_ADDRESS': 'unix:path=/run/user/1000/bus,guid=162c8e214e053d649d818fa961409bd6', 'WINDOWPATH': '2', 'GNOME_TERMINAL_SERVICE': ':1.58', 'LC_IDENTIFICATION': 'lzh_TW', 'DESKTOP_SESSION': 'ubuntu', 'IM_CONFIG_PHASE': '2', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'GTK_MODULES': 'gail:atk-bridge', 'VTE_VERSION': '5202', 'XDG_SESSION_TYPE': 'x11', 'XDG_SEAT': 'seat0', 'LC_TELEPHONE': 'lzh_TW', 'LC_MONETARY': 'lzh_TW', 'PWD': '/home/airobot/test', 'ROS_ETC_DIR': '/opt/ros/melodic/etc/ros', 'COLORTERM': 'truecolor', 'LC_NAME': 'lzh_TW', 'XDG_MENU_PREFIX': 'gnome-', 'GNOME_DESKTOP_SESSION_ID': 'this-is-deprecated', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl= [... cut ...]
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/parent.py", line 144, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/roslaunch/config.py", line 463, in load_config_default
    raise RLException(e)
RLException: [/home/airobot/test/src/ros_kortex/kortex_driver/launch/kortex_dual_driver.launch] requires the 'prefix' arg to be set

[rospy.core][INFO] 2021-09-14 21:02:13,028: signal_shutdown [atexit]
```

felixmaisonneuve, 2021-09-15:
Hi @jason134017,

I think I found the source of the issue. `kortex_dual_driver.launch` wants to load `joint_limits.yaml`, which require a `prefix` arg to be set. 

A workaround you could do, since you are using  two different arms, is to manually edit those `joint_limits.yaml` with the correct prefix. You will need to edit 4 files : 
`kortex_description/arms/gen3/6dof/config/joint_limits.yaml` : change `$(arg prefix)` to `right_` for the joint_names
`kortex_description/grippers/robotiq_2f_85/config/joint_limits.yaml` : change `$(arg prefix)` to `right_`
`kortex_description/arms/gen3_lite/6dof/config/joint_limits.yaml` : change `$(arg prefix)` to `left_` for the joint_names
`kortex_description/grippers/gen3_lite_2f/config/joint_limits.yaml` : change `$(arg prefix)` to `left_`

This is not an ideal fix and at some point it will probably be changed to something more generic, but it should be enough to get you going. 

I think you might face other minor issues since  `kortex_dual_driver.launch` seems to be outdated a bit, but it shouldn't to much of a problem.

Let me know how it goes,
Felix

felixmaisonneuve, 2021-10-06:
Hi @jason134017,

I assume this has fixed your problem, so I will close this issue. 
If you still have a problem, you can re-open it at any time and I will help you

Regards,
Felix

fi6, 2022-01-19:
Hi Felix,

Is there any way to avoid modifying joint_limits files? I'd like to leave these files from official repo unchanged to avoid any other potential problems in the future(which may need me to change them back)

Best,
Yida

felixmaisonneuve, 2022-01-19:
Hi @fi6,

I did some more testing on this. The workaround did work, but, like you said, it involved changing the joint_limit.yaml file. 
Since we load joint_limits.yaml with `subst_value="true"`, it means that joint_limits.yaml can access the arguments defined in kortex_dual_driver.launch. Because joint_limits.yaml is using a `prefix` argument, it has to be defined in the launch file (which is not the case).

So I created a launch file that only load this yaml file and I included it in kortex_dual_driver.launch. That way, I can specify a different `prefix` argument for both left and right arm. 
You can look at the modifications I did in the new branch I created : `bugfix/dual-driver-not-launching` (based on `noetic-devel`)

With this, I was able to launch both arms without changing the yaml file.

Feel free to try it and give your feedback on it. 

I might merge this in the futur if it seems to fix the issue without breaking anything

Best,
Felix

fi6, 2022-01-20:
Hi @felixmaisonneuve ,

When comparing the two branches I can't find the changes of `kortex_dual_driver.launch`(looks like some vision related change?). Link to the comparison:
https://github.com/Kinovarobotics/ros_kortex/compare/bugfix/dual-driver-not-launching...noetic-devel

Also, I'm having trouble launching moveit via kortex_dual_driver.launch, could you please confirm if you can launch it correctly?

Best,
Yida

felixmaisonneuve, 2022-01-20:
Hahaha, you are right. I don't know what happened. 
It is fixed now and you can see the actual modifications I did.

I am able to launch Rviz correctly, but not Moveit, since we are not include any move_group in the launch file.
If you want to use MoveIt you will have to create your own moveit config for your specific setup

fi6, 2022-01-21:
The fix works, could you please merge it into master branches? Also, do you mean that I need to use MoveIt Setup Assistant to generate the config files myself? I'll try it later.

felixmaisonneuve, 2022-01-21:
You could probably copy one of the configs (e.g. ros_kortex/kortex_move_it_config/gen3_move_it_config/) and rename it to something like ros_kortex/kortex_move_it_config/dual_arm_move_it_config. This would be the equivalent of using the setup assistant. 
You would then need to adapt the launch files in that config in order for moveit to recognize the multiple arms and move them.

In the kortex_dual_driver.launch, it is missing the part where we launch moveit : 
https://github.com/Kinovarobotics/ros_kortex/blob/fa072f13b500b399e109bb146891c3e9ebb830cc/kortex_driver/launch/kortex_driver.launch#L82-L97

But if you want to add that part, you need to have a move_group.launch for your specific dual arm setup. 

So anyway, that is where I would start if you want to use moveit. This would require a bit of work and I am not going to do it, but I can support you if you decide to try it. Also this would make an awesome addition to our current stack if you want to share your work and open a PR once it is working.

I will close this issue, since the original problem has been fixed (I will merge it soon), but feel free to open a new issue if you are having trouble to implement a dual arm moveit config.

Regards,
Felix

closed 2022-01-21 by felixmaisonneuve