Kinovarobotics/ros_kortex #191: Kortex API movement does not work in Gazebo. Noetic-ROS

alexlt97, 2021-09-14:
## Description

Good morning, I was trying to use the gazebo examples combined with the examples of movement. My idea is to learn how to use KINOVA API so I can create my own nodes but it looks that in gazebo the robot (I tried with both models) does not respond 
to the movement commands, however, it does to open the gripper command`self.example_send_gripper_command(0.0)` and the initial movement that moves the robot to HOME from the node `home_robot.py` called in `kortex_gazebo spawn_kortex_robot.launch`. Do I need to change something inside the .launch files?
## Version

ROS distribution : Noetic

Branch and commit you are using : noetic-devel last commit

## Steps to reproduce

1. roslaunch kortex_gazebo spawn_kortex_robot.launch
2. roslaunch kortex_examples full_arm_movement_python.launch

## Expected behavior

The robot follows the movement examples from `kortex_examples full_arm_movement_python.launch` in simulation as it does in the real robot.

## Any other information
I have no problems using the API in the real robot.

felixmaisonneuve, 2021-09-14:
Hi @alexlt97,

The current `full_arm_movement_python.launch` does not work when working with a Gazebo simulation. 

The API call used to move the arm has changed with the [release 2.3](https://github.com/Kinovarobotics/ros_kortex/releases/tag/v2.3.0). It was previously using `PlayCartesianTrajectory` and `PlayJointTrajectory` (now deprecated). The API is now using `ExecuteWaypointTrajectory`. 

You can see how our simulation stack works in [this readme](https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel/kortex_examples#the-ways-to-use-a-kortex-arm-with-ros), but basically each call of the API has to be implemented manually to simulated the expected behaviour in Gazebo. The ones that are not implemented will return a default response and a warning. 

What happens here is that the examples has been updated to use the current `ExecuteWaypointTrajectory`, but this API call has not been implemented in our simulation stack, so it is not working in Gazebo. In Gazebo, only the previous `PlayCartesianTrajectory` and `PlayJointTrajectory` have been implemented. 

What you could do is use an old version of the example source file (one that is using `PlayCartesianTrajectory` and `PlayJointTrajectory`). You would have to look to any commit before `f0218097604a902cf8bfc65db64ef3aa5eefb455` (which is when the 2.3 release was merged on June 4). You could also look at PR #179, where (some of) the deprecated calls were updated. You would have to revert these changes to use the old calls again. 

If you take an old `example_full_arm_movement.py` file, it is possible some calls will be incompatible, but it shouldn't be too much trouble to make it work. Also, you will probably be able to use it on a real robot, but you will get some warnings when using deprecated API calls.

Try using an old example and let me know how it goes,
Felix

alexlt97, 2021-09-16:
Ok as you said it works with the previous functions,

Are you planning to update the simulator to include the new features of the API?

closed 2021-09-16 by alexlt97