Kinovarobotics/ros_kortex #193: Can I use FollowJointTrajectory Action to override a motion command?

yukizakisoh, 2021-09-17:
## Description

I'm trying to implement "Jogging" using ROS, but I couldn't get it to work.

## Version

ROS distribution : melodic

Branch and commit you are using : melodic-devel

## Any other information

I'm trying to implement "Jogging" using ROS, but I couldn't get it to work.
For example, I would like to use this package (https://github.com/tork-a/jog_control) to control a robot with a joystick. This package seems to be proven in many other robots. 
In this package, we are using FollowJointTrajectory Action topic to send motion commands.
Joystick operation is a command that changes from moment to moment. Therefore, we are sending movement commands while overwriting multiple FlowJointTrajectoryAction goal.

I made the config files of jog_control for Gen3.
When I tried this package with the Gazebo simulator, I confirmed that it worked correctly.
This means that I was able to control the robot on the simulator according to the joystick operation as I wanted.

However, the real robot did not move as we expected at all.
To be specific, I kept getting the warning "There is already an active goal. It is being cancelled."
Below is an example of FlowJointTrajectoryAction and the log of its execution.

I found out that the API for Gen3 real robots prohibits using FollowJointTrajectoryAction to override motion commands.
https://github.com/Kinovarobotics/ros_kortex/blob/noetic-devel/kortex_driver/src/non-generated/driver/joint_trajectory_action_server.cpp#L81-L86

I have two questions as follows.
* Is there any way to override move commands using FollowJointTrajectoryAction?
* Is there any good way to achieve time-to-time motion commands with a joystick using ROS?

<details>
<summary>Example of a published FlowJointTrajectoryAction</summary>

rostopic echo /my_gen3/gen3_joint_trajectory_controller/follow_joint_trajectory/goal
header: 
  seq: 0
  stamp: 
    secs: 1631863663
    nsecs: 282770824
  frame_id: ''
goal_id: 
  stamp: 
    secs: 1631863663
    nsecs: 282771229
  id: "/my_gen3/jog_frame_node-1-1631863663.282771229"
goal: 
  trajectory: 
    header: 
      seq: 0
      stamp: 
        secs: 1631863663
        nsecs: 282761965
      frame_id: "base_link"
    joint_names: 
      - joint_1
      - joint_2
      - joint_3
      - joint_4
      - joint_5
      - joint_6
    points: 
      - 
        positions: [-1.6572501005146067, 1.1549577598796679, 1.3144496985965053, 2.0144126657383117, 0.28768419055041156, -0.821965261546824]
        velocities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        accelerations: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
        effort: []
        time_from_start: 
          secs: 0
[... 1215 lines cut ...]
[ERROR] [1631863679.881760631]: Joint Trajectory failed validation in the arm.
[ERROR] [1631863679.881796043]: Error 1 : Invalid Acceleration - Provided Waypoints Lead to Go Over Maximum Acceleration on Actuator (0) and Segment (0).
[ INFO] [1631863679.937701257]: New goal received.
[ WARN] [1631863679.937745006]: There is already an active goal. It is being cancelled.
[ INFO] [1631863679.937762570]: Calling Stop on the robot.
[ INFO] [1631863679.954568968]: Joint Trajectory Goal is accepted.
[ERROR] [1631863680.056427105]: Joint Trajectory failed validation in the arm.
[ERROR] [1631863680.056475765]: Error 1 : Invalid Acceleration - Provided Waypoints Lead to Go Over Maximum Acceleration on Actuator (0) and Segment (0).
[ERROR] [1631863680.056491169]: Error 2 : Invalid Speed - Provided Waypoints Lead to go Over Maximum Speed on Actuator (3) and Segment (0).
[ INFO] [1631863680.110505878]: New goal received.
[ WARN] [1631863680.110547141]: There is already an active goal. It is being cancelled.
[ INFO] [1631863680.110566574]: Calling Stop on the robot.
[ INFO] [1631863680.133234538]: Joint Trajectory Goal is accepted.
[ERROR] [1631863680.230677126]: Joint Trajectory failed validation in the arm.
[ERROR] [1631863680.230717175]: Error 1 : Invalid Acceleration - Provided Waypoints Lead to Go Over Maximum Acceleration on Actuator (0) and Segment (0).
[ERROR] [1631863680.230728618]: Error 2 : Invalid Speed - Provided Waypoints Lead to go Over Maximum Speed on Actuator (3) and Segment (0).
[ INFO] [1631863680.291298553]: New goal received.
[ WARN] [1631863680.291374865]: There is already an active goal. It is being cancelled.
[ INFO] [1631863680.291398851]: Calling Stop on the robot.
[ INFO] [1631863680.304537457]: Joint Trajectory Goal is accepted.
[ERROR] [1631863680.405305194]: Joint Trajectory failed validation in the arm.
[ERROR] [1631863680.405340383]: Error 1 : Invalid Acceleration - Provided Waypoints Lead to Go Over Maximum Acceleration on Actuator (0) and Segment (0).
[ERROR] [1631863680.405351095]: Error 2 : Invalid Speed - Provided Waypoints Lead to go Over Maximum Speed on Actuator (3) and Segment (0).

</details>

felixmaisonneuve, 2021-09-17:
Hi @yukizakisoh,

I think what you are looking for is the `sendTwistCommand`/`sendTwistJoystickCommand` (found in the [base](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/summary_pages/Base.md))

`sendTwistCommand` will take a [TwistCommand](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/messages/Base/TwistCommand.md#) with absolute values (in m/s)

`sendTwistJoystickCommand` will take a [TwistCommand](https://github.com/Kinovarobotics/kortex/blob/master/api_cpp/doc/markdown/messages/Base/TwistCommand.md#) with values from -1.0 to 1.0 that represent a percentage of the max speed. 

This is the commands used by the joystick to move the arm and everything is computed by the base. These commands will continue indefinitely, so you will need to send a `TwistCommand` with zeros or a `stop` command to stop the movement.

I think this is what you are trying to implement, and I don't think you will ever be able to achieve this behaviour using `FlowJointTrajectoryAction`. Let me know if I'm not understanding correctly what you want to do and `sendTwistCommand` is not what you need.

Regards,
Felix

yukizakisoh, 2021-09-21:
Thank you for your reply.
It's a pity that I can't use `FollowJointTrajectoryAction`, but `sendTwistCommand`/`sendTwistJoystickCommand` seems to suit my purpose.
I'll try this first and report back later.

yukizakisoh, 2021-09-24:
Using `sendTwistCommand`,  I was able to achieve the "jogging" motion with the joystick.
I was able to fulfill the functionality that I initially wanted to achieve.
Thank you very much.
I will close this issue.

closed 2021-09-24 by yukizakisoh