Kinovarobotics/ros_kortex #197: Example moveit not works

GitNzino, 2021-09-28:
## Description

I have tried to launch `rosrun kortex_examples example_move_it_trajectories.py` but there was an unexpected error.

[image]

What should I do?

## Version

ROS distribution : Melodic

Branch and commit you are using : [Actual Melodic Branch](https://github.com/Kinovarobotics/ros_kortex/tree/melodic-devel)

felixmaisonneuve, 2021-09-28:
Hi @GitNzino,

Is your node running? 
Did you do `roslaunch kortex_driver kortex_driver.launch ...` in another terminal?

Best,
Felix

GitNzino, 2021-09-28:
Yes, also in RVIZ moveit works properly

felixmaisonneuve, 2021-09-28:
I think there is a problem with the namespace when using `rosrun`. I would recommend to use our launch file instead of running directly with `rosrun`. Also, if you have changed your  `robot_name` when launching your arm driver, you will need to pass this param too.
```
roslaunch kortex_examples moveit_example.launch robot_name:=<your_robot_name>
```

Regards,
Felix

GitNzino, 2021-09-29:
Yes, sorry!
You have completely reason!
Was my fault!

Tank you and have a nice day,
Lorenzo

GitNzino, 2021-10-01:
Sorry for reopening this Issue, but i have a doubt.

I need to launch a script similar to this via python, so not passing from `roslanch`. Inasmuch as in the last comment is really clear that the argument `gen3_lite` ot `gen_3` passed as argument of the launch command is important then I thought to change all the `rospy.get_namespace()` in `\my_gen3_lite` (I work with a gen3 lite). Following the complete exchange:

### Original
```python
class MoveGroupPythonIntefaceTutorial(object):

	def __init__(self):

		# Initialize the node
		super(MoveGroupPythonIntefaceTutorial, self).__init__()
		moveit_commander.roscpp_initialize(sys.argv)
		rospy.init_node('constrain_path', anonymous=True)

		try:
			self.is_gripper_present = rospy.get_param(rospy.get_namespace() + "is_gripper_present", False)
			if self.is_gripper_present:
				gripper_joint_names = rospy.get_param(rospy.get_namespace() + "gripper_joint_names", [])
				self.gripper_joint_name = gripper_joint_names[0]
			else:
				gripper_joint_name = ""
		  	self.degrees_of_freedom = rospy.get_param(rospy.get_namespace() + "degrees_of_freedom", 6)

			# Create the MoveItInterface necessary objects
			arm_group_name = "arm"
			self.robot = moveit_commander.RobotCommander("robot_description")
			self.scene = moveit_commander.PlanningSceneInterface(ns=rospy.get_namespace())
			self.arm_group = moveit_commander.MoveGroupCommander(arm_group_name, ns=rospy.get_namespace())
			self.arm_group.set_max_velocity_scaling_factor(1)
			self.arm_group.set_max_acceleration_scaling_factor(1)
			self.arm_group.set_num_planning_attempts(50)
			self.display_trajectory_publisher = rospy.Publisher(rospy.get_namespace() + 'move_group/display_planned_path',
								                    moveit_msgs.msg.DisplayTrajectory,
								                    queue_size=50)

			if self.is_gripper_present:
				gripper_group_name = "gripper"
				self.gripper_group = moveit_commander.MoveGroupCommander(gripper_group_name, ns=rospy.get_namespace())

			rospy.loginfo("Initializing node in namespace " + rospy.get_namespace())
		except Exception as e:
			print (e)
			self.is_init_success = False
		else:
			self.is_init_success = True
```
### Modified
```
class MoveGroupPythonIntefaceTutorial(object):

    def __init__(self):
        # Initialize the node
        super(MoveGroupPythonIntefaceTutorial, self).__init__()
        moveit_commander.roscpp_initialize(sys.argv)
        rospy.init_node('robot_camera_interaction') #modifica nome nodo
        """
        ATTENZIONE, per fare funzionare il codice senza lanciarlo da nodo
        bisogna sostituire rospy.get_namespace() con my_gen3_lite
        """
        try:
            my_gen3_lite = "/my_gen3_lite/"
            self.is_gripper_present = rospy.get_param("/my_gen3_lite/" + "is_gripper_present", False)
            if self.is_gripper_present:
                gripper_joint_names = rospy.get_param("/my_gen3_lite/" + "gripper_joint_names", [])
                self.gripper_joint_name = gripper_joint_names[0]
            else:
                gripper_joint_name = ""
            self.degrees_of_freedom = rospy.get_param("/my_gen3_lite/" + "degrees_of_freedom", 6)

            # Create the MoveItInterface necessary objects
            arm_group_name = "arm"
            self.robot = moveit_commander.RobotCommander("robot_description")
            self.scene = moveit_commander.PlanningSceneInterface(ns="/my_gen3_lite/")
            self.arm_group = moveit_commander.MoveGroupCommander(arm_group_name, ns="/my_gen3_lite/")
            self.arm_group.set_max_velocity_scaling_factor(1)
            self.arm_group.set_max_acceleration_scaling_factor(1)
            self.arm_group.set_num_planning_attempts(20)
            self.display_trajectory_publisher = rospy.Publisher("/my_gen3_lite/" + 'move_group/display_planned_path',
                                                    moveit_msgs.msg.DisplayTrajectory,
                                                    queue_size=50)

            if self.is_gripper_present:
                gripper_group_name = "gripper"
                self.gripper_group = moveit_commander.MoveGroupCommander(gripper_group_name, ns="/my_gen3_lite/")

            rospy.loginfo("Initializing node in namespace " + "/my_gen3_lite/")
        except Exception as e:
            print (e)
            self.is_init_success = False
        else:
            self.is_init_success = True
```

But I continue to obtain the precedent error. Can you suggest anything to solve?

Thank you in advance,
Lorenzo

felixmaisonneuve, 2021-10-01:
Hi @GitNzino,

I am quickly looking at it and I see this line 
```python
self.robot = moveit_commander.RobotCommander("robot_description")
```
 
I think you might want to specify the namespace for that command too
```python
self.robot = moveit_commander.RobotCommander(robot_description = "robot_description", ns="/my_gen3_lite/")
```
or just 
```python
self.robot = moveit_commander.RobotCommander(ns="/my_gen3_lite/")
```
because I think `robot_description = "robot_description"` is the default value. 

I think it might be why moveit does not find the robot_description

Best,
Felix

felixmaisonneuve, 2021-10-05:
I looked into it a bit more. 

You could use `rosrun` to launch your node, but you need to pass the namespace parameter : 
```
rosrun kortex_example example_move_it_trajectories.py __ns:=my_gen3_lite
```

You could do the same while using python : 
```
python path/to/example_move_it_trajectories.py __ns:=my_gen3_lite
```

If you can't specify arguments when launching you command, you could also define your namespace while initializing your moveit_commander : 
```python
    args = sys.argv
    args.append("__ns:=my_gen3")
    moveit_commander.roscpp_initialize(args)
```

You can revert your file to what you had previously and any of these solution should work.

Hope this helps,
Felix

GitNzino, 2021-10-06:
Thank you so much! Now I can work!
Have a nice day,
Lorenzo

closed 2021-10-06 by GitNzino