Kinovarobotics/ros_kortex #202: ABORTED:Solution found but controller failed during execution

JingKangSun, 2021-10-13:
Hi @felixmaisonneuve ,

I'm sorry to bother you again. #182

After I update my firmware version to 2.3 and software to 2.3 release , the question of 'Solution found but controller failed during execution'  seems disappeared. However, after a long period of testing , I found that this problem still exists.

I use a list which defines the joint angle of the seven joints to adjust the arm pose, but sometimes this operation will complete, sometimes it will fail. When it failed, I will either restart the arm or try a few more times untill  the action is complete.

I guess that  this problem is caused by the motor temperature being too high or the joint speed limitations. Do you have any suggestion?

Regards,
JingKang

PoplarCrystal, 2021-10-14:
Hi, @JingKangSun 
I think I meet the same problem. I think it is due to the IK solution of KDL in moveit. 
I suggest you try the IK_fast or Trac_ik
by the way, can we chat with wechat, PoplarCrystal.

JingKangSun, 2021-11-03:
Hi @felixmaisonneuve ,

I'm sorry for taking so long to reply to you, because this problem does not always appear.

Here is the full log of this problems.

[image]

How can I avoid this kind of problem? Or when this problem occurs, what can I do?

Regards,
JingKang

felixmaisonneuve, 2021-11-03:
Hi @JingKangSun,

Since this problem does not always appear, it is possible that one of the joints motor gets too hot (like you said) if you use the arm intensively, but I feel it would trigger a fault and you wouldn't be able to use the arm if you don't do a `ClearFault()`.

You could use the WebApp to monitor what is going on. In *Monitoring*->*Detailed* you can see the current temperature and position of each joint. This could give an indication on what causes this issue (if a fault pops-up, temperature is too hot, a joint is "out of bounds"...). 

Is this something you could do? If it fails you could send a screenshot of the webapp page. 

Best,
Felix

felixmaisonneuve, 2022-01-04:
Hi @JingKangSun,

Do you have any update on this? Were you able to reproduce the error? 
Did you try using IK_fast or Trac_ik?

I might close this issue in a couple of weeks if I don't get any answer from you.

Best,
Felix

PoplarCrystal, 2022-01-04:
你好，我已收到你的来信。若有重要事情，请短信告知！

felixmaisonneuve, 2022-04-12:
Hi @JingKangSun,

I am closing this issue. If you have more questions, feel free to comment/re-open this issue or open a new one.

Best,
Felix

closed 2022-04-12 by felixmaisonneuve