Kinovarobotics/ros_kortex #206: "Joint Trajectory failed validation in the arm" when sending commands from Ros2

Thijsons, 2021-11-08:
Hello everyone,

The last few days I've been working on bridging our Ros2 path planning pipeline with the Kortex driver using the Ros1 Bridge. The idea being to control a gen3_lite from Ros2. We've been giving commands to the robot by publishing the desired path on the `my_gen3_lite/gen3_lite_joint_trajectory_controller/follow_joint_trajectory/goal` topic in Ros1. When running the Gazebo simulator this works perfectly, however when trying to connect to the real robot errors are returned saying parts of the trajectory are breaching the velocity and acceleration limits of the arm. Of course we made sure to pass the `joint_limits.yaml` to the move_group node in Ros2 to make sure the time-parameterization would stay within the limits of the arm.

As far as I can tell the error is thrown from the `joint_trajectory_action_server` in the `JointTrajectoryActionServer::goal_received_callback()` function by using the `ValidateWaypointList()` function of the BaseClient. I can't seem to find the implementation of the BaseClient anywhere?

I noticed the trajectories planned in Ros2 have all timestamps neatly separated by 0.1 second, Ros 1 on the other hand seems to use a less neat number. Would this make a difference to the trajectory validation in the Kortex driver?

I realize this issue might be a bit outside the scope of this repository; Any advice would be appreciated however.

extra information:
Ros2 distribution: Foxy
Ros1 distribution: Noetic

felixmaisonneuve, 2021-11-08:
Hi @Thijsons,

Do you have a complete output of the error given by `ValidateWaypointList()` (or the error output provided by ROS)?  

Behaviour between Gazebo and a real arm can be different since Gazebo is using our simulator (instead of translating calls to the Kortex API), which re-implement some "core" services.

You say timestamps are not the same. This might cause an error if trajectory points do not match these timestamps (e.g. if the point is supposed to be reached at 1.24s but timestamp is at 1.19s). Looking at the simulation implementation of [ExecuteReachJointAngles](https://github.com/Kinovarobotics/ros_kortex/blob/037233668ee5610d62ece15159fe767259922803/kortex_driver/src/non-generated/driver/kortex_arm_simulation.cpp#L773) (which is called when publishing to `follow_joint_trajectory` in simulation), each trajectory point is manually separeted by a timestep of `JOINT_TRAJECTORY_TIMESTEP_SECONDS` (0.01s). 
If, when using a real arm, these timesteps are not the same, but the trajectory points are, it is normal the trajectory is not valid. 

Do you have more informations about this? What are these timestamps exactly? 

Regards,
Felix

felixmaisonneuve, 2021-11-08:
Do you have the list of every trajectory point and the corresponding error message (e.g. if the driver returns with point 13, it would be nice to have the values for the point 12, 13 and 14)? We might be able to see what values are problematics and figure out why they create an error.

Maybe the velocity and accelerations values are incorrect, but the simulation doesn't validate that velocity and acceleration match the variation of position of the joint. 
Positions might be reachable (based on the arm speed/acceleration limits) and the simulation simply accept the trajectory, but the real arm driver will verify that the velocity and acceleration provided match the variation from one position to the other.

Just to make sure I understand correctly, you are trying to use a trajectory created by MoveIt2 that went throught your ROS1 bridge. Using moveit1 works fine (both in simulation and with a real arm), but the trajectory provided by moveit2 does not. It works only for the simulation environment, but using moveit2 + ROS1 bridge + real arm creates errors. Is that correct? 

Also, is it possible to see the trajectory list before and after passing throught the ROS1 bridge. It would be a good idea to make sure the bridge does not change the timestamps. I suppose the format of the trajectory goal changes between ROS1 and ROS2 and it is possible your bridge is changing the timestamps. If ROS2 does not use timestamps, maybe your bridge is simply assuming there is a point every 0.1s (which would explain what we are seeing here) and that might not be true.

Best,
Felix

Thijsons, 2021-11-09:
> Just to make sure I understand correctly, you are trying to use a trajectory created by MoveIt2 that went throught your ROS1 bridge. Using moveit1 works fine (both in simulation and with a real [...]

Yes, that is indeed correct.

I just ran code using the following joint limits in Moveit2:
[image]

Setting the scaling factor in Rviz to 0.6 for both acceleration and velocity gives me the following error when moving from the 'vertical' to the 'home' position:
[image]

Taking a look at the `/my_gen3_lite/gen3_lite_joint_trajectory_controller/follow_joint_trajectory/goal` topic gives the following: (the highlighted part is segment 57, assuming counting starts at 1)

[image]

As far as I can tell none of these points go over the limits in the joint_limits.yaml? I verified the joint_limits with both the limits from the original joint_limits.yaml in the official driver and the joint limits specified in the manual of the gen3_lite; So they should be suitable for the gen3_lite.

Reducing the acceleration and velocity scaling in Moveit2 to 0.5 does allow me to plan and move from 'vertical' to 'home', however the movement becomes quite slow, and from experience I know other trajectories might not work with these scaling settings. 

with regards to comparing the values before and after being bridged, I have found no difference between the two. Here is a screenshot of the output of the bridged topic (there might be some difference in the values here compared to the previous screenshots a I took these values from a different run. The planning request and the error where exactly the same though). Unfortunately comparing the two is a bit difficult due to the slight different format in moveit1 vs 2. (Ros2 on the left, Ros1 on the right)

[image]

felixmaisonneuve, 2021-11-10:
Hi @Thijsons,

Like you said there isn't any point that looks weird. Also, points values are not changed when going throught the ROS1 bridge.

The next thing I would do is modify the error output to include more details on the error. 
Change the trajectory error report : 
https://github.com/Kinovarobotics/ros_kortex/blob/037233668ee5610d62ece15159fe767259922803/kortex_driver/src/non-generated/driver/joint_trajectory_action_server.cpp#L165-L169

to something like this 
```c++
 ROS_ERROR("Joint Trajectory failed validation in the arm.");
// Go through report and print errors
for (unsigned int i = 0; i < report.trajectory_error_report().trajectory_error_elements_size(); i++)
{
	ROS_ERROR("Error %i : Type : %i", i+1, report.trajectory_error_report().trajectory_error_elements(i).error_type());
	ROS_ERROR("Error %i : Error Value : %i", i+1, report.trajectory_error_report().trajectory_error_elements(i).error_value());
	ROS_ERROR("Error %i : Min Value : %i", i+1, report.trajectory_error_report().trajectory_error_elements(i).min_value());
	ROS_ERROR("Error %i : Max Value : %i", i+1, report.trajectory_error_report().trajectory_error_elements(i).max_value());
	ROS_ERROR("Error %i : Message : %s", i+1, report.trajectory_error_report().trajectory_error_elements(i).message().c_str());
}
```

For this change to take effect, you will need to re-compile using `catkin_make`.

Saving the trajectory (the output of the goal topic) in a file will still be usefull to help us identify which goal is problematic and why.

Best,
Felix

Thijsons, 2021-11-11:
Hi @felixmaisonneuve,
thank you for the tip. After doing the same experiment as last time (vel. and accel. at 0.6, moving from vertical to retract), I get the following results:
[image]

Is there any information on the meaning behind these error values? The comments in the code say: 
[image]
This seems to suggest that the 'error_value' is the value that breaches the maximum limits of the joints. Does this mean that the acceleration goes up to -437313712 rad/s^2 according to the Kortex driver? If so, it doesn't seem like the kortex driver uses the velocities and accelerations provided by the trajectory, instead it might compute them on its own? (based on the positions alone values maybe?).

Here is an echo of the `/my_gen3_lite/gen3_lite_joint_trajectory_controller/follow_joint_trajectory/goal` topic, the highlighted part is the 97th point (if we start counting at 1):

[image]

Interestingly, the acceleration error is at the second to last waypoint, the only place where the difference between the current time stamp and the next one is not 0.1 seconds.

I also ran the same test again, but this time with the velocity and speed scaling set to the max (1.0). this resulted in both a acceleration error **and** a velocity error. The acceleration error was again in the last part of the trajectory; However, the velocity error wasn't.
The error messages:
[image]

The points of the acceleration error:
[image]

The points of the velocity error:
[image]

felixmaisonneuve, 2021-11-11:
`error_value`, `min_value` and `max_value` are floats, so you'll need to change  
```c++
ROS_ERROR("Error %i : Error Value : %i", i+1, report.trajectory_error_report().trajectory_error_elements(i).error_value());
ROS_ERROR("Error %i : Min Value : %i", i+1, report.trajectory_error_report().trajectory_error_elements(i).min_value());
ROS_ERROR("Error %i : Max Value : %i", i+1, report.trajectory_error_report().trajectory_error_elements(i).max_value());
```
to (notice the second `%i` changing to `%f`)
```c++
ROS_ERROR("Error %i : Error Value : %f", i+1, report.trajectory_error_report().trajectory_error_elements(i).error_value());
ROS_ERROR("Error %i : Min Value : %f", i+1, report.trajectory_error_report().trajectory_error_elements(i).min_value());
ROS_ERROR("Error %i : Max Value : %f", i+1, report.trajectory_error_report().trajectory_error_elements(i).max_value());
```
`error_value` is supposed to match the joint velocity/acceleration from the goal that creates an error. With that change, we'll have something more reasonnable like `-0.239999` instead of that `-437313712` nonsense. I copied too fast and forgot about that, my mistake. 

Now, these values are rounded to the 5th decimal and, if they appear to be the same (between `error_value`, `min_value` and `max_value`), it is possible that smaller decimals causes the error. I think it might be the case because, even if they are displayed in integers, the values are identical. If it is the case, I would suspect a float rounding to be the cause of error (because floats values are not exact). The best way to test that would be to change the velocity/acceleration limits just a little. (e.g. `0.5` to `0.499`). 

However, all your velocity limits are at 0.5 rad/s and, looking at the Gen3_lite User Guide, they could go up to 1.0 rad/s (and 1.57 rad/s for joint 6). That makes me think that the rounding error is probably between 2 goal points.

For example, an actuator position is rounded down a bit in goal 5 than rounded up a bit in goal 6. This would require a highier velocity, but lets say velocity is also rounded down. I think velocity could end up outside the range of required speed to get from goal 5 to goal 6. This would create the error not because the requested velocity is outside of the arm capabilities, but because the requested velocity is not enough to get form one position to the other. In other words, requested velocity would be too far from the computed velocity needed to get from 5 to 6.

You mentionned another very interisting point, the acceleration is failing in the second to last point : the only point where the timestep is different from 0.1. Maybe this last point is calculated as if there was a 0.1s timestep but the timestamp is not updated correctly. 

I think it is worth trying one more time after changing `%i` to `%f` and see what the values are. I am not sure what the next move will be if it looks like the error is from rounding errors and if the `error_value` doesn't seem to be over the arm's capability...

Regards,
Felix

Thijsons, 2021-11-11:
Thank you for your answer, and good catch on the prints! That does indeed explain the weird values.

Doing the test again (speed and velocity at 0.6) with proper printing results in the following error message and points
[image]
[image]

The error values now make more sense, obviously the acceleration is a bit higher than what is allowed according to the driver. I am still a bit confused about the exact values of the numbers however; The accelerations given by the trajectory don't go over 0.24 rad/s^2 (13.75 deg/s^2), but the error message indicates an acceleration of 20.97(I assume the values are in deg/s^2?). 

The indicated maximum value is also lower than I would expect; 20.63deg (0.36rad) instead of 22.91deg (0.4rad), the latter of which is the lowest acceleration limit in the 'joint_limits.yaml'. These differences seem higher than I would expect from a rounding error. I made sure to validate the values in the joint_limits.yaml with those in the Gen3_Lite user manual just to make sure, and they match.

In order to get some extra information on what's going on I also re-did the test with the speed and velocity scaler set to 1.0:
[image]
Trajectory points with the velocity error:
[image]

Trajectory points with the acceleration error:
[image]

The maximum value for the velocity does match the expected joint limits very precisely: 0.5 rad is about 28.65 deg. The velocity indicated is also slightly above the maximum of 0.5rad/s, seems to me like this is probably a rounding error that happens somewhere along the way. (maybe during bridging?). This hypothesis is confirmed when running the test again with a scaling of 0.99 instead of 1.0, as this removes the velocity error :).

The value of the acceleration still puzzles me a bit though, as the value in the error is much higher than anything in the path and the maximum limit remains lower then the official limits? 
This got me thinking about the following thing you mentioned:
>You mentionned another very interisting point, the acceleration is failing in the second to last point : the only point where the timestep is different from 0.1. Maybe this last point is calculated as if there was a 0.1s timestep but the timestamp is not updated correctly.

Maybe there is some weird scaling going on there, although if the kortex driver assumes a larger time step than it receives this would result in lower acceleration values as apposed to higher values as far as I understand it. It also wouldn't affect the limits.

best regards,
Thijsons

felixmaisonneuve, 2021-11-11:
Even if I think a 0.11 degree (0.002 rad) is a pretty big difference for a rounding error, it sure looks like it could be what is causing the velocity errors.

Since goals doesn't seem to change inside the bridge, I would assume it is Moveit2 that is rounding values a bit to much. If it is the case, I don't really know what you could do to increase the arm speed. Using a scaling of 0.99 is close enough, but it still bothers me that you cannot use a scaling of 1. 

About the acceleration errors, I was thinking the opposite. I don't think the driver assume a larger timestamp, I think Moveit is calculating position, velocity and acceleration as if the timestep is 0.1s, but it doesn't actually update the timestamp to reflect that. 

I don't know if you are able to do that, but I would try to round up the timestamp of the last point in the goal (e.g. change `time_to_start` from 308878364ns to 400000000ns). Maybe doing this inside your ROS1 bridge or before sending the trajectory to the driver. I don't know how everything is implemented, do you think this could be done?

Best,
Felix

Thijsons, 2021-11-12:
Thank you again for the answer.

In my previous reply I said:
> The indicated maximum value is also lower than I would expect; 20.63deg (0.36rad) instead of 22.91deg (0.4rad), the latter of which is the lowest acceleration limit in the 'joint_limits.yaml'. These [...]

I discovered that this is caused by a Boolean variable that can set when launching the driver called `use_hard_limits`, setting this to true seems to set the joint limits to the values indicated in the manual

In your reply you suggested the following:
> I don't know if you are able to do that, but I would try to round up the timestamp of the last point in the goal (e.g. change time_to_start from 308878364ns to 400000000ns). Maybe doing this inside [...]

The good news: doing this did indeed fix the error at the last point of the trajectory; This allows me to execute the same test as in the previous post with the acceleration and speed scale at 0.99 (moving from 'vertical' to 'retract') without errors.

The bad news is that for other trajectories the errors still persist. For example, I managed to get the following error when moving the arm from one random place to another:
[image]

Therefore I don't believe the problem was actually caused by the timestamp; More likely is that lengthening the timestamp solved the problem by significantly reducing the necessary acceleration requirements, in the same way as decreasing the speed and acceleration scaling in Moveit would.

I spend some time looking at the source code around the error and it does indeed seem like the driver only cares about the positions in the trajectory, see [here](https://github.com/Kinovarobotics/ros_kortex/blob/72fc06e52b0524dd36155c7ce9be40650ca8b7d6/kortex_driver/src/non-generated/driver/joint_trajectory_action_server.cpp#L142). This explains the different acceleration values in the driver vs the original trajectory, but it doesn't give any information on how to fix the problem. The actual Kortex Api implementation that seems to determine the necessary accelerations and velocity to follow the trajectory is not open source as far as I can tell, so no help there.

What remains interesting to me is that the official drivers seem to have none of these problems, so there must be something different about the trajectories they generate vs the ones generated by my Moveit2 setup. 

Regards,
Gijs

felixmaisonneuve, 2021-11-12:
Like you said, Kortex driver calculates its own velocity and acceleration based on the positions. Since the driver is working with a trajectory from MoveIt1, I suppose the trajectory given by MoveIt2 contains errors (which). Either timesteps are to short or positions are two far appart between 2 points. Maybe Moveit2 assumes a perfect acceleration, but the driver takes jerk (acceleration of acceleration) into consideration and this is why it is failling, but why would velocity be problematic too and wy would MoveIt1 work? 

On the last screenshot you sent there is a 15 degrees (0.3 rad) error margin, which is way too much for a rounding error. It makes no sense. To me, the trajectory from Moveit2 is simply not valid, but I don't understand why this happens. 

Do you think you could compare the same trajectory (e.g. moving from 'vertical' to 'retract') from Moveit2 and from Moveit1. For Moveit2 it could be after going throught ROS1 bridge (since it doesn't affect values), so it will be the same format as Moveit1. Even if the trajectory succeeds, it could be interesting to see what the differences are between the two. Otherwise,I am not sure what to do. 

Of course, you could add a arbitrairy delay to each point's duration, but that would result in a slower trajectory, which is not optimal. Something like this 
```c++
// Fill the waypoint
for (auto position : traj_point.positions)
{
        angular_waypoint->add_angles(KortexMathUtil::toDeg(position));
}
angular_waypoint->set_duration(waypoint_duration + 0.05); // arbitrairy 0.05s between each trajectory positions
```

alexlt97, 2021-11-16:
Good afternoon,

I followed the conversation as @Thijsons and I worked together on this issue. I replicated the same trajectory both in Moveit and in Moveit2, including the case of scale 1.0 (that fails) and a reachable trajectory applying a scale of 0.9. The trajectory goes from the vertical position [0,0,0,0,0,0] to the retracted position.

In MoveIt2 when the acceleration scale is 1.0, the computed trajectory is rejected by the Kinova driver, displaying the following error:

[image]

The error is found in the actuator[2], meaning an over acceleration in Joint3 in the first point. Using a scale of 0.9 computed a trajectory with no errors. I did an analysis based on the time to start times and the positions created for the trajectory as it is what Kinova driver uses. Both trajectories from Moveit2 generates the same trajectory besides the number of points/time of the trajectory. The trajectory follows a Piece-wise polynomial function so I assumed that Kinova driver would use this trajectory to derive/compute velocities and accelerations. The results I obtained for both trajectories are:

[image]

[image]

If we focus on Joint 3 (grey), the behaviour is pretty similar differing in the peak velocity and so we can expect the same for the acceleration. However, the error appears in the first position of the trajectory. I inverse the order of the trajectory: retract position to vertical. It outcomes the same data but the error for an acceleration scale of 1.0 is found in the last point of the trajectory, the same point for vertical-retract trajectory. 

[image]

[image]

All these results are now compared with the same trajectory but computed by Moveit1. Its main difference lies in the variable "time to start" which is not constant as it is in Moveit2.

[image]

Overall, we find the difference between Moveit1 and 2 hard to understand. Do you perhaps have any insights into the assumptions the API makes on the input path? What kind of information could we provide to help understand this problem better?

felixmaisonneuve, 2021-11-16:
Thank you @alexlt97 for these graphs, this is very appreciated. I see you already have opened an issue about time parametrization in MoveIt2. This is surely the issue here. 

I just learned there was some modifications in the [2.3 release](https://github.com/Kinovarobotics/ros_kortex/releases/tag/v2.3.0) about the time parametrization used by moveit. 

> the MoveIt configuration for all arms has been adapted to use the `AddIterativeSplineParameterization` [Time Parametrization algorithm](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/planning_request_adapter_plugins/src/add_iterative_spline_parameterization.cpp)

You'll find this is in `kortex_move_it_config/your_robot_move_it_config/launch/ompl_planning_pipeline.launch.xml` : 
```
<!-- The request adapters (plugins) used when planning with OMPL.
       ORDER MATTERS -->
  <arg name="planning_adapters" value="default_planner_request_adapters/AddIterativeSplineParameterization
				       default_planner_request_adapters/FixWorkspaceBounds
				       default_planner_request_adapters/FixStartStateBounds
				       default_planner_request_adapters/FixStartStateCollision
				       default_planner_request_adapters/FixStartStatePathConstraints" />
```

This change was made especially to address the kind of issue you are having ("invalid velocity" or "invalid acceleration"). 

Since this a MoveIt2 problem, I cannot help you that much to fix it. 

I will close this issue since you already have figure out the issue comes from the time parametrization in MoveIt2 and you already have opened an issue about it.

If you need any other information, leave a comment and I will answer. You can also reopen this issue if you feel there is still something I can do.

Regards,
Felix

closed 2021-11-16 by felixmaisonneuve