Kinovarobotics/ros_kortex #207: How to get the cartesian position and rotation matrix corresponding to the robot base frame?

HeegerGao, 2021-12-29:
Hi, I'd like to know how to read the cartesian position and rotation matrix corresponding to the robot base frame. Is there any ros topic responsible for this, or can we calculate them from some other information? Thank you very much!

felixmaisonneuve, 2022-01-03:
Hi @HeegerGao,

I am not sure what you are looking for. In what context do you plan to use such information?

In the User Guide, you will find many information the arm's reference frame and rotation matrix in the *Guidance for advanced users*->*Reference frames and transformations* section. Is that what you are looking for?

Best,
Felix

HeegerGao, 2022-01-03:
Hi @felixmaisonneuve,

Thanks for your reply! I am sorry that I didn't make the question clear.

I want to know the Cartesian position and Euler angles of the **camera** (the one on the arm) at any time. For now, I can only get positions of each joint of the arm. Since the camera is not at the same position as joint 6 (end effector), I don't know how to get the cartesian position of the camera.

For example, if I want to get the position and Euler angles of the end effector, I will use `rostopic echo /my_gen3/base_feedback`and read `tool_pose_x`, `tool_pose_y`, `tool_pose_z`, `tool_pose_theta_x`, `tool_pose_theta_y`, `tool_pose_theta_z` to get answers. Is there any way to get the same information about the camera? Thank you very much!

felixmaisonneuve, 2022-01-03:
Ok, I understand. 

I could refer the issue #46 which led to PR #82.
This added a joint for the Camera depth frame and the camera color frame with the offset from the User guide (*Vision module sensors reference frames* section). With this, I am able to access the position and the orientation of the cameras in RVIZ
[image]

In your code, you should be able to get these values using tf2 (using those links). 

You cannot get the cameras origin directly from the API. You would need to compute the position of the cameras based on the `tool_pose` and the offset from the User guide.

I will mention that this fix was only added for 7dof arms. I will update the xacro file for the 6dof.

Regards,
Felix

HeegerGao, 2022-01-10:
Thank you very much. It solved my question.

closed 2022-01-10 by HeegerGao