Kinovarobotics/ros_kortex #210: Unable to connect to conan-center

Qingyuan-Jiang, 2022-01-05:
## Description

Hello! I would like to ask for some help on installing the ros_kortex package. I followed the instruction [here](https://github.com/Kinovarobotics/ros_kortex/tree/kinetic-devel) and set up the Conan profile. However, when I tried to compile the catkin workspace. I received the following error message. 

`-- +++ processing catkin package: 'kortex_driver'
-- ==> add_subdirectory(ros_kortex/kortex_driver)
CONAN_TARGET_PLATFORM is x86
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
-- Generating .msg files for action kortex_driver/FollowCartesianTrajectory /home/jiang/catkin_ws/src/ros_kortex/kortex_driver/action/non_generated/FollowCartesianTrajectory.action
-- kortex_driver: 432 messages, 259 services
-- Conan: checking conan executable in path
-- Conan: Found program /usr/local/bin/conan
-- Conan: Version found Conan version 1.23.0

-- Conan: Adding kinova_public remote repository (https://artifactory.kinovaapps.com/artifactory/api/conan/conan-public)
-- Conan: Automatic detection of conan settings from cmake
-- Conan: Settings= -s;build_type=Release;-s;compiler=gcc;-s;compiler.version=5;-s;compiler.libcxx=libstdc++11;-s;kortex_api_cpp:compiler=gcc;-s;kortex_api_cpp:compiler.version=5;-s;kortex_api_cpp:compiler.libcxx=libstdc++11
-- Conan executing: conan install /home/jiang/catkin_ws/src/ros_kortex/kortex_driver/conanfile.py -s build_type=Release -s compiler=gcc -s compiler.version=5 -s compiler.libcxx=libstdc++11 -s kortex_api_cpp:compiler=gcc -s kortex_api_cpp:compiler.version=5 -s kortex_api_cpp:compiler.libcxx=libstdc++11 -g=cmake --update
Configuration:
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=gcc
compiler.libcxx=libstdc++11
compiler.version=5
os=Linux
os_build=Linux
kortex_api_cpp:compiler=gcc
kortex_api_cpp:compiler.version=5
kortex_api_cpp:compiler.libcxx=libstdc++11
[options]
[build_requires]
[env]

kortex_api_cpp/2.3.0-r.34@kortex/stable: Not found in local cache, looking in remotes...
kortex_api_cpp/2.3.0-r.34@kortex/stable: Trying with 'conan-center'...
ERROR: Failed requirement 'kortex_api_cpp/2.3.0-r.34@kortex/stable' from 'conanfile.py'
ERROR: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:645)

Unable to connect to conan-center=https://conan.bintray.com
1. Make sure the remote is reachable or,
2. Disable it by using conan remote disable,
Then try again.
CMake Error at ros_kortex/kortex_driver/cmake/conan.cmake:399 (message):
  Conan install failed='1'
Call Stack (most recent call first):
  ros_kortex/kortex_driver/cmake/conan.cmake:485 (conan_cmake_install)
  ros_kortex/kortex_driver/CMakeLists.txt:72 (conan_cmake_run)

-- Configuring incomplete, errors occurred!
See also "/home/jiang/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/jiang/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:318: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
`

ROS distribution : Ubuntu 16.04 + Kinetic

Branch and commit you are using : kinetic-devel

Is there a way to solve this problem? Or are there any pre-requisite that I might missed? Thank you so much for your help!

Qingyuan-Jiang, 2022-01-06:
Hi Felix!
Thank you so much for your reply!
I believe I should have tried this solution, which gives me the following message.

`Trying to clone repo: https://github.com/conan-io/conanclientcert.git
Repo cloned!
Copying file LICENSE to /home/jiang/.conan/.
Copying file cacert.pem to /home/jiang/.conan/.
`

However, when I get back and try to compile the workspace with catkin_make, it still reports the same error.
Are there any suggestions to this situation? Thanks much!

felixmaisonneuve, 2022-01-07:
I see you are using Conan version 1.25, You could maybe try to update conan. `pip3 install conan --upgrade` or `python3 -m pip install conan --upgrade`.

Qingyuan-Jiang, 2022-01-12:
Hello! I've updated my Conan, but the same error occurs though the message changes a little.

My current Conan information is as below.
``Name: conan
Version: 1.44.0
Summary: Conan C/C++ package manager
Home-page: https://conan.io
Author: JFrog LTD
Author-email: luism@jfrog.com
License: MIT
Location: /home/jiang/.local/lib/python3.5/site-packages
Requires: patch-ng, tqdm, six, distro, node-semver, PyJWT, requests, PyYAML, fasteners, python-dateutil, Jinja2, colorama, pluginbase, urllib3, pygments, bottle
Required-by: 
``

The updated error message is 
``
-- +++ processing catkin package: 'kortex_driver'
-- ==> add_subdirectory(ros_kortex/kortex_driver)
CONAN_TARGET_PLATFORM is x86
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
-- Generating .msg files for action kortex_driver/FollowCartesianTrajectory /home/jiang/catkin_ws/src/ros_kortex/kortex_driver/action/non_generated/FollowCartesianTrajectory.action
-- kortex_driver: 432 messages, 259 services
-- Conan: checking conan executable in path
-- Conan: Found program /home/jiang/.local/bin/conan
-- Conan: Version found Conan version 1.44.0

-- Conan: Adding kinova_public remote repository (https://artifactory.kinovaapps.com/artifactory/api/conan/conan-public)
-- Conan: Automatic detection of conan settings from cmake
-- Conan: Settings= -s;build_type=Release;-s;compiler=gcc;-s;compiler.version=5;-s;compiler.libcxx=libstdc++11;-s;kortex_api_cpp:compiler=gcc;-s;kortex_api_cpp:compiler.version=5;-s;kortex_api_cpp:compiler.libcxx=libstdc++11
-- Conan executing: conan install /home/jiang/catkin_ws/src/ros_kortex/kortex_driver/conanfile.py -s build_type=Release -s compiler=gcc -s compiler.version=5 -s compiler.libcxx=libstdc++11 -s kortex_api_cpp:compiler=gcc -s kortex_api_cpp:compiler.version=5 -s kortex_api_cpp:compiler.libcxx=libstdc++11 -g=cmake --update
Configuration:
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=gcc
compiler.libcxx=libstdc++11
compiler.version=5
os=Linux
os_build=Linux
kortex_api_cpp:compiler=gcc
kortex_api_cpp:compiler.version=5
kortex_api_cpp:compiler.libcxx=libstdc++11
[options]
[build_requires]
[env]

kortex_api_cpp/2.3.0-r.34@kortex/stable: Not found in local cache, looking in remotes...
kortex_api_cpp/2.3.0-r.34@kortex/stable: Trying with 'conan-center'...
kortex_api_cpp/2.3.0-r.34@kortex/stable: WARN: Remote https://conan.bintray.com is deprecated and will be shut down soon.
kortex_api_cpp/2.3.0-r.34@kortex/stable: WARN: Please use the new 'conancenter' default remote.
kortex_api_cpp/2.3.0-r.34@kortex/stable: WARN: Add it to your remotes with: conan remote add -i 0 conancenter https://center.conan.io
ERROR: Failed requirement 'kortex_api_cpp/2.3.0-r.34@kortex/stable' from 'conanfile.py'
ERROR: HTTPSConnectionPool(host='conan.bintray.com', port=443): Max retries exceeded with url: /v1/ping (Caused by SSLError(SSLError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:645)'),))

Unable to connect to conan-center=https://conan.bintray.com
1. Make sure the remote is reachable or,
2. Disable it by using conan remote disable,
Then try again.
CMake Error at ros_kortex/kortex_driver/cmake/conan.cmake:399 (message):
  Conan install failed='1'
Call Stack (most recent call first):
  ros_kortex/kortex_driver/cmake/conan.cmake:485 (conan_cmake_install)
  ros_kortex/kortex_driver/CMakeLists.txt:72 (conan_cmake_run)
``

This error seems similar to the issue #198, but although I tried 
``
conan config install https://github.com/conan-io/conanclientcert.git
``
It still gives the same error message.

Any ideas on this situation? Thank you so much!

felixmaisonneuve, 2022-01-12:
Try these two command : 
```
conan remote remove conan-center
conan remote add conancenter https://center.conan.io --insert 1
```

It will change the public repository of conan

Qingyuan-Jiang, 2022-01-17:
This solves my problem! Thank you so much for your help!

closed 2022-01-17 by felixmaisonneuve

sankethm22, 2023-06-16:
Hi, I know this issue is closed, but I am running into the same error and none of the solution from #198 or this thread solved my problem. 

The error message I am getting:

```
-- +++ processing catkin package: 'kortex_driver'
-- ==> add_subdirectory(ros_kortex/kortex_driver)
CONAN_TARGET_PLATFORM is x86
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
-- Generating .msg files for action kortex_driver/FollowCartesianTrajectory /local/home/mars/Documents/ros/catkin_workspace/src/ros_kortex/kortex_driver/action/non_generated/FollowCartesianTrajectory.action
-- kortex_driver: 432 messages, 259 services
-- Conan: checking conan executable in path
-- Conan: Found program /usr/local/bin/conan
-- Conan: Version found Conan version 1.59.0

-- Conan: Adding kinova_public remote repository (https://artifactory.kinovaapps.com/artifactory/api/conan/conan-public)
-- Conan: Automatic detection of conan settings from cmake
-- Conan: Settings= -s;build_type=Release;-s;compiler=gcc;-s;compiler.version=7;-s;compiler.libcxx=libstdc++11;-s;kortex_api_cpp:compiler=gcc;-s;kortex_api_cpp:compiler.version=5;-s;kortex_api_cpp:compiler.libcxx=libstdc++11;-s;kortex_api_cpp:build_type=Release
-- Conan executing: conan install /local/home/mars/Documents/ros/catkin_workspace/src/ros_kortex/kortex_driver/conanfile.py -s build_type=Release -s compiler=gcc -s compiler.version=7 -s compiler.libcxx=libstdc++11 -s kortex_api_cpp:compiler=gcc -s kortex_api_cpp:compiler.version=5 -s kortex_api_cpp:compiler.libcxx=libstdc++11 -s kortex_api_cpp:build_type=Release -g=cmake --update
Configuration:
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=gcc
compiler.libcxx=libstdc++11
compiler.version=7
os=Linux
os_build=Linux
kortex_api_cpp:compiler=gcc
kortex_api_cpp:compiler.version=5
kortex_api_cpp:compiler.libcxx=libstdc++11
kortex_api_cpp:build_type=Release
[options]
[build_requires]
[env]

kortex_api_cpp/2.3.0-r.34@kortex/stable: Not found in local cache, looking in remotes...
kortex_api_cpp/2.3.0-r.34@kortex/stable: Trying with 'conan-center'...
kortex_api_cpp/2.3.0-r.34@kortex/stable: Trying with 'kinova_public'...
ERROR: Failed requirement 'kortex_api_cpp/2.3.0-r.34@kortex/stable' from 'conanfile.py'
ERROR: HTTPSConnectionPool(host='artifactory.kinovaapps.com', port=443): Max retries exceeded with url: /artifactory/api/conan/conan-public/v1/ping (Caused by SSLError(SSLError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:852)'),))

Unable to connect to kinova_public=https://artifactory.kinovaapps.com/artifactory/api/conan/conan-public
1. Make sure the remote is reachable or,
2. Disable it by using conan remote disable,
Then try again.
CMake Error at ros_kortex/kortex_driver/cmake/conan.cmake:399 (message):
  Conan install failed='1'
Call Stack (most recent call first):
  ros_kortex/kortex_driver/cmake/conan.cmake:485 (conan_cmake_install)
  ros_kortex/kortex_driver/CMakeLists.txt:86 (conan_cmake_run)
```

I see that I get a diiferent ssl error code compared to the OP, which is - _ssl.c:852.

Any ideas what might be wrong?

felixmaisonneuve, 2023-06-26:
Hi @sankethm22 

Can you confirm you have the correct conan certificate
```
md5sum ~/.conan/cacert.pem
```

The md5 checksum should be `54f488cc2ba6483793a42b6852189b39`

Best,
Felix

sankethm22, 2023-06-28:
@felixmaisonneuve 

md5 checksum is fine. Have verified that multiple times at this point.

The issue I think is that I am behind a corporate proxy. So it keeps blocking any ssl verification. 

Although the main issue with the proxy hasn't been resolved. I managed to change the CMakeLists.txt in kortex_driver folder and removed the line where the kinova_public repository was being added. However, I added kinova_public directly in the terminal with ssl verification set to False. I did this since I trusted the repository and wouldn't recommend this for everyone.