Kinovarobotics/ros_kortex #212: Inconsistency between mass values of Robotiq 2F-85 gripper provided in Gen3 User Guide and URDF.

RKJenamani, 2022-01-12:
I have been trying to implement a gravity compensation controller for my Gen3 6DoF fitted with a Robotiq 2F-85 gripper. However, it is not working well due to what I believe are inconsistencies in the gripper URDF. 

In the Gen3 User Guide provided on [Kinova's website](https://www.kinovarobotics.com/product/gen3-robots), the mass of the Robotiq 2F-85 gripper is specified to be 0.925 kg (Page 69, Section "Robotiq 2F-85 Gripper tool configuration"). However, the sum of masses of all the links in the [URDF](https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel/kortex_description/grippers/robotiq_2f_85/urdf) provided for this gripper model in this repository comes out to be around 0.36 kg. While this issue would better suited for the source repository: [https://github.com/ros-industrial/robotiq](https://github.com/ros-industrial/robotiq), the source repository is unmaintained. :( 

Any help/clarification with this would be much appreciated. Thank you!

felixmaisonneuve, 2022-01-13:
Hi @RKJenamani,

I have looked a bit at this and I will try to summarize my thought on it. 

0.925 kg is the correct weight for the Robotiq 2F-85 gripper. 

The sum of all the links do end up at 0.36 kg. However, this weight does not include the pads mass, which is not specified in the xacro file.
Also, I don't know if the robotiq_arg2f_base_link mass includes all the hardware, or if it only includes the shell. In which case, the actual mass for a real gripper would be heavier, since it has a motor in it. I think ~220g is a bit light for such a "big" piece if it includes a motor and sensors (and other hardware) in order to make the fingers move, but maybe I am mistaken. 
Those 2 factors might explain why we end up with a much lighter mass. 

All the xacro are coming directly from the Robotiq github repository you have linked, and, if they are not accurate, it is hard for us the validate the data. 
If you compare the files, you will see that all the mass and inertial parameters for the 85 gripper in the xacro files are copied directly from the 140, so I am not even sure the mass is correct for the fingers and knuckles links on the 85. Looking at the original robotiq repo, `<inertial>` tags are commented for the 85 gripper, so I don't know if they actually validated the data for this gripper. 

I know you have been told to increase each component to reach the correct 0.925 kg. I would also suggest to add a `<inertial>` tag (with a mass) for the pads : https://github.com/Kinovarobotics/ros_kortex/blob/fa072f13b500b399e109bb146891c3e9ebb830cc/kortex_description/grippers/robotiq_2f_85/urdf/robotiq_arg2f_85_model_macro.xacro#L107-L128

I suppose you could also check if you want to increase the base_link mass more instead of the finger and knuckles. 

Do you need to be very precise with the gripper's mass? Why do you need this?

Regards,
Felix

RKJenamani, 2022-01-18:
Hi @felixmaisonneuve,

Thank you very much for your help! I

closed 2022-01-18 by RKJenamani

RKJenamani, 2022-01-18:
I would be happy to create a PR updating the gripper URDF on this repository with the one that I am currently using. However, I am still waiting on its validation by getting the gravity compensation controller working perfectly (currently facing some problems with torque control).

felixmaisonneuve, 2022-01-18:
Sure, if you open a PR with this new value, I will make sure to review it.

Best,
Felix