Kinovarobotics/ros_kortex #216: resource not found: robotiq_2f_85_gripper_visualisation

jabbershort, 2022-02-11:
## Description

When launching the Gazebo simulation, including the Robotiq 2F 85 gripper, an error is hit not finding the resource.

## Version

ROS distribution : 
Noetic

Branch and commit you are using : 

noetic-devel: 5b2c03a

## Steps to reproduce

roslaunch kortex_gazebo spawn_kortex_robot.launch gripper:=robotiq_2f_85

## Code example (if necessary)

```
root@2e0e0d8f79c9:/# roslaunch kortex_gazebo spawn_kortex_robot.launch gripper:=robotiq_2f_85
... logging to /root/.ros/log/1400d87c-8b27-11ec-9185-0242ac110002/roslaunch-2e0e0d8f79c9-544.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
resource not found: robotiq_2f_85_gripper_visualization
ROS path [0]=/opt/ros/noetic/share/ros
ROS path [1]=/catkin_workspace/src
ROS path [2]=/opt/ros/noetic/share
when instantiating macro: load_gripper (/catkin_workspace/src/kortex_description/grippers/robotiq_2f_85/urdf/robotiq_2f_85_macro.xacro)
instantiated from: load_robot (/catkin_workspace/src/kortex_description/robots/kortex_robot.xacro)
in file: /catkin_workspace/src/kortex_description/grippers/robotiq_2f_85/urdf/robotiq_arg2f_85_model_macro.xacro
included from: /catkin_workspace/src/kortex_description/robots/gen3_robotiq_2f_85.xacro
RLException: Invalid <param> tag: Cannot load command parameter [robot_description]: command [['/opt/ros/noetic/lib/xacro/xacro', '--inorder', '/catkin_workspace/src/kortex_description/robots/gen3_robotiq_2f_85.xacro', 'dof:=7', 'vision:=true', 'sim:=true', 'prefix:=']] returned with code [2].

Param xml is <param name="robot_description" command="$(find xacro)/xacro --inorder $(find kortex_description)/robots/$(arg arm)_$(arg gripper).xacro dof:=$(arg dof) vision:=$(arg vision) sim:=true prefix:=$(arg prefix)" unless="$(eval not arg('gripper'))"/>
The traceback for the exception was written to the log file
```

## Expected behavior

Launch file would instantiate the robot and and send it home.

felixmaisonneuve, 2022-02-11:
Hi @jabbershort,

I tested it and it worked for me.
Did you do any changes to the launch files?
The only place I found `robotiq_2f_85_gripper_visualization` is here and it is commented :
https://github.com/Kinovarobotics/ros_kortex/blob/5b2c03a5b3df696e3dcdb0cb18c6a4e474a0a5ea/kortex_description/grippers/robotiq_2f_85/urdf/robotiq_arg2f_85_model_macro.xacro#L1-L5

Best,
Felix

jabbershort, 2022-02-11:
Hi Felix,

Thanks for the quick response. I did a completely clean install (albeit with Melodic as that's what I've got on my test machine). It still didn't work, so out of desperation, I deleted those two and now it works fine, even though they are commented out. I tried the 140 robotiq and that was unaffected. 

Seems strange that its only affecting me and not you, but thanks for pointing me in the right direction.

closed 2022-02-11 by jabbershort