Kinovarobotics/ros_kortex #217: How to control robotiq_2f_140 gripper with the "my_gen3" robot.

2000222, 2022-02-21:
## Description

Firstly I changed the code in "spawn_kortex_robot.launch" as follows to add the robotiq_2f_140 with gen3 robot
```
<arg name="gripper" default="robotiq_2f_140" if="$(eval arg('arm') == 'gen3')"/>
```
but while running "spawn_kortex_robot.launch", I get the following issues:
```
[INFO] [1645444112.307428, 0.612000]: Loading controller: robotiq_2f_140_gripper_controller
[INFO] [1645444112.401840, 0.656000]: Loading controller: joint_3_position_controller
[ERROR] [1645444112.407502590, 0.663000000]: Could not load controller 'robotiq_2f_140_gripper_controller' because controller type 'position_controllers/GripperActionController' does not exist.
[ERROR] [1645444112.408009342, 0.663000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ INFO] [1645444112.453165108, 0.691000000]: Added FollowJointTrajectory controller for gen3_joint_trajectory_controller
[INFO] [1645444112.523859, 0.732000]: Loading controller: joint_4_position_controller
[INFO] [1645444112.580440, 0.759000]: Loading controller: joint_5_position_controller
[INFO] [1645444112.635234, 0.779000]: Loading controller: joint_6_position_controller
[INFO] [1645444112.694456, 0.798000]: Loading controller: joint_7_position_controller
[INFO] [1645444112.749126, 0.813000]: Controller Spawner: Loaded controllers: joint_1_position_controller, joint_2_position_controller, joint_3_position_controller, joint_4_position_controller, joint_5_position_controller, joint_6_position_controller, joint_7_position_controller
[ERROR] [1645444113.409066, 1.233000]: Failed to load robotiq_2f_140_gripper_controller
[INFO] [1645444113.410654, 1.233000]: Controller Spawner: Loaded controllers: gen3_joint_trajectory_controller, joint_state_controller
[INFO] [1645444113.415448, 1.236000]: Started controllers: gen3_joint_trajectory_controller, joint_state_controller
[ WARN] [1645444126.236219712, 5.722000000]: Waiting for robotiq_2f_140_gripper_controller/gripper_cmd to come up
[ WARN] [1645444183.453899868, 11.722000000]: Waiting for robotiq_2f_140_gripper_controller/gripper_cmd to come up
[ERROR] [1645444247.855769719, 17.722000000]: Action client not connected: robotiq_2f_140_gripper_controller/gripper_cmd
```
so I'm not sure how can we control the gripper together with the whole gen3 arm? Hope to get more solutions, thank you. 

## Version

ROS distribution :  ros-noetic

2000222, 2022-02-23:
I've tried the simulation without a gripper, and it works well. 
The whole output is shown as follows: 
```
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://sarah-Inspiron-5567:38855/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /my_gen3/gazebo_ros_control/pid_gains/finger_joint/d: 0.01
 * /my_gen3/gazebo_ros_control/pid_gains/finger_joint/i: 0.0
 * /my_gen3/gazebo_ros_control/pid_gains/finger_joint/p: 10.0
 * /my_gen3/gazebo_ros_control/pid_gains/left_inner_finger_joint/d: 0.001
 * /my_gen3/gazebo_ros_control/pid_gains/left_inner_finger_joint/i: 0.0
 * /my_gen3/gazebo_ros_control/pid_gains/left_inner_finger_joint/p: 0.5
 * /my_gen3/gazebo_ros_control/pid_gains/left_inner_knuckle_joint/d: 0.01
 * /my_gen3/gazebo_ros_control/pid_gains/left_inner_knuckle_joint/i: 0.0
 * /my_gen3/gazebo_ros_control/pid_gains/left_inner_knuckle_joint/p: 10.0
 * /my_gen3/gazebo_ros_control/pid_gains/right_inner_finger_joint/d: 0.001
 * /my_gen3/gazebo_ros_control/pid_gains/right_inner_finger_joint/i: 0.0
 * /my_gen3/gazebo_ros_control/pid_gains/right_inner_finger_joint/p: 0.5
 * /my_gen3/gazebo_ros_control/pid_gains/right_inner_knuckle_joint/d: 0.01
 * /my_gen3/gazebo_ros_control/pid_gains/right_inner_knuckle_joint/i: 0.0
 * /my_gen3/gazebo_ros_control/pid_gains/right_inner_knuckle_joint/p: 10.0
 * /my_gen3/gazebo_ros_control/pid_gains/right_outer_knuckle_joint/d: 0.0
 * /my_gen3/gazebo_ros_control/pid_gains/right_outer_knuckle_joint/i: 0.0
 * /my_gen3/gazebo_ros_control/pid_gains/right_outer_knuckle_joint/p: 10.0
 * /my_gen3/gen3_joint_trajectory_controller/action_monitor_rate: 25
 * /my_gen3/gen3_joint_trajectory_controller/constraints/goal_time: 1.0
 * /my_gen3/gen3_joint_trajectory_controller/constraints/stopped_velocity_tolerance: 0.5
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_1/d: 2.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_1/i: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_1/i_clamp_max: 100.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_1/i_clamp_min: -100.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_1/p: 3000.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_2/d: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_2/i: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_2/i_clamp_max: 5.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_2/i_clamp_min: -5.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_2/p: 50000.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_3/d: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_3/i: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_3/i_clamp_max: 1.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_3/i_clamp_min: -1.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_3/p: 3000.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_4/d: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_4/i: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_4/i_clamp_max: 1.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_4/i_clamp_min: -1.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_4/p: 50000.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_5/d: 0.2
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_5/i: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_5/i_clamp_max: 1.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_5/i_clamp_min: -1.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_5/p: 750.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_6/d: 1.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_6/i: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_6/i_clamp_max: 1.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_6/i_clamp_min: -1.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_6/p: 5000.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_7/d: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_7/i: 0.0
 * /my_gen3/gen3_joint_trajectory_controller/gains/joint_7/i_clamp_max: 0.1
[... 403 lines cut ...]

********************************************************
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1645617776.794610318, 17.347000000]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1645617776.795723204, 17.347000000]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1645617777.547651870, 17.481000000]: Ready to take commands for planning group arm.
[ INFO] [1645617777.813737799, 17.531000000]: Ready to take commands for planning group gripper.
```
And at this case it's possible to conduct motion planning for Gen 3 robot , but for the added robotiq_2f_140 gripper, I have no ideas.

2000222, 2022-02-23:
I've solved the problem to successfully control the gripper, thank you.

ZakariaChekakta, 2022-03-21:
@2000222 
How did you manage to control the gripper?