Kinovarobotics/ros_kortex #218: catkin installation broken: missing launch files

alexswerner, 2022-02-26:
## Description
After installation to a separate install prefix (`catkin config --install-space PATH`) with catkin tools, launching the gazebo demo with the robot does not work. Specifically `roslaunch kortex_gazebo spawn_kortex_robot.launch` fails because `kortex_move_it_config/gen3_move_it_config/launch/move_group.launch` can not be loaded.

## Version
ROS distribution : noetic

Branch and commit you are using : noetic-devel 5b2c03a5b3df696e3dcdb0cb18c6a4e474a0a5ea

## Steps to reproduce
- follow install instructions, substituting catkin tools for catkin_make. Hence for workspace initialization run
```
catkin init
catkin config --install --install-space /opt/kinova
```
- build the workspace with
```
catkin build
```
- Try executing `spawn_robot.launch` in `kortex_gazebo`
```
source /opt/kinova/setup.sh
roslaunch kortex_gazebo spawn_kortex_robot.launch
```
This gives the following output
```
... logging to /home/awerner/.ros/log/1eebf08a-96a0-11ec-990b-c0b6f9fc5a17/roslaunch-robotlab-lt1-964.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
RLException: while processing /opt/kinova/share/kortex_description/../kortex_move_it_config/gen3_move_it_config/launch/move_group.launch:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: '/opt/kinova/share/kortex_description/../kortex_move_it_config/gen3_move_it_config/launch/move_group.launch'
The traceback for the exception was written to the log file
```

## Expected behavior
Launch files are correctly installed.

felixmaisonneuve, 2022-02-28:
Hi @alexswerner,

Your `roslaunch` command is not working because you are sourcing your install folder. You should source your devel folder instead 
```
cd your_catkin_ws
source devel/setup.sh
roslaunch kortex_gazebo spawn_kortex_robot.launch
```

I have done some testing and I was having issues to make `source devel/setup.sh` work. But it worked fine when I started a new workspace if I didn't run `catkin config --install`. 
I created a new workspace, install our ros stack and ran `catkin build` and evertything worked. 

I am not sure what `catkin config --install` does exactly, but this seems to be the issue (as well as sourcing the install folder instead of the devel folder).

Try this out and let me know if you still have problems

Best,
Felix

felixmaisonneuve, 2022-04-12:
Hi @alexswerner,

Do you still have problem with your catkin workspace?
Can we close this issue?

I will wait a couple of weeks and close this issue if I don't hear from you by then.

Best,
Felix

closed 2022-06-08 by felixmaisonneuve