Kinovarobotics/ros_kortex #219: joint velocity topic in gazebo simulation

th1991-01, 2022-03-07:
## Description

I try pubulishing joint velocity topic in kortex gazebo simulation. Although `ros_kortex/kortex_driver/readme.md` says `Publishing joint velocities (radians/second) ...`, it seems degree/second in gazebo simulation.

## Version

ROS distribution : noetic

Branch and commit you are using : noetic-devel 5b2c03a

## Steps to reproduce

1. launch gazebo simulator
```
$ roslaunch kortex_gazebo spawn_kortex_robot.launch
```

2. send joint velocity topic
```
$ rostopic pub /my_gen3/in/joint_velocity kortex_driver/Base_JointSpeeds "{joint_speeds : [{joint_identifier: 0, value: 10},{joint_identifier: 1, value: 0.0},{joint_identifier: 2, value: 0.0},{joint_identifier: 3, value: 0.0},{joint_identifier: 4, value: 0.0},{joint_identifier: 5, value: 0.0},{joint_identifier: 6, value: 0.0}], duration: 1}"
```
It seemd to take 36 seconds for joint_0 to make one revolution, implying that `value: 10` means 10 (degree/second). 

## Expected behavior

The description in `ros_kortex/kortex_driver/readme.md` is consistent with gazebo simulation.

felixmaisonneuve, 2022-03-07:
Hi @th1991-01,

I tested and you are right. When running a gazebo simulation and publishing `in/joint_velocity`, it is degrees/second. 
I create the branch bugfix/simulation-velocity-topic-is-not-in-rad based on noetic-devel that fixes this issue. 
I simply took the logic from kortex_subscriber..cpp (that is called when publishing on `in/joint_velocity` with a real arm) that was missing from kortex_arm_simulation.cpp (that is called in gazebo simulation).

https://github.com/Kinovarobotics/ros_kortex/blob/5b2c03a5b3df696e3dcdb0cb18c6a4e474a0a5ea/kortex_driver/src/non-generated/driver/kortex_subscribers.cpp#L36-L40

You are free to test the branch, I experimented a bit with it and it seems to be working alright now. I will merge this change in the next days unless you notice something wrong with it.

Regards,
Felix

th1991-01, 2022-03-08:
The branch works as expected. Thank you so much!

PR #220 (2022-03-09): change velocity to rad in topic in/joint_velocity when in gazebo simulation
Resolves #219 

Input for topic `in/joint_velocity` is in rad/s when using a real arm, as specified in the readme file. 
However, when using a Gazebo simulation, this topic's input is in degrees/s.

This PR adds the conversion so that `in/joint_velocity` now takes values in rad/s to make everything more consistent
closed 2022-03-09 by felixmaisonneuve