Kinovarobotics/ros_kortex #221: Gazebo simulation error

AaronLiu666, 2022-03-09:
## Description

No changes on the code. The example_full_arm_movement.py seems not working with simulation in gazebo but works perfectly on a real arm( gen3 with robotiq 85). I dont know why and this is what it gives me:

[ WARN] [1646814359.447489155, 108.788000000]: The simulation handler for base/clear_faults is not implemented, so the service calls will return the default response.
[ WARN] [1646814361.953915393, 111.292000000]: The simulation handler for base/activate_publishing_of_action_topic is not implemented, so the service calls will return the default response.
[ WARN] [1646814363.555786747, 112.893000000]: The simulation handler for control_config/set_cartesian_reference_frame is not implemented, so the service calls will return the default response.
[ WARN] [1646814363.807976884, 113.145000000]: The simulation handler for base/get_product_configuration is not implemented, so the service calls will return the default response.
[ERROR] [1646814363.811806143, 113.149000000]: Unsupported action type 0 : could not execute simulated action.

## Version

ROS distribution : melodic

Branch and commit you are using : melodic

## Steps to reproduce

1. launch spawn
2. launch full_arm_movement_python.launch

## Code example (if necessary)

```
    # paste code here #
```

## Expected behavior
Nothing happens but this WARNING AND ERROR

[ WARN] [1646814359.447489155, 108.788000000]: The simulation handler for base/clear_faults is not implemented, so the service calls will return the default response.
[ WARN] [1646814361.953915393, 111.292000000]: The simulation handler for base/activate_publishing_of_action_topic is not implemented, so the service calls will return the default response.
[ WARN] [1646814363.555786747, 112.893000000]: The simulation handler for control_config/set_cartesian_reference_frame is not implemented, so the service calls will return the default response.
[ WARN] [1646814363.807976884, 113.145000000]: The simulation handler for base/get_product_configuration is not implemented, so the service calls will return the default response.
[ERROR] [1646814363.811806143, 113.149000000]: Unsupported action type 0 : could not execute simulated action.

## Any other information

Dont know why but other example codes works fine with both simulation and real arm. Only this one does not work in simulation but no warn or error with real arm.

felixmaisonneuve, 2022-03-09:
Hi @AaronLiu666,

This is because waypoints were added in the 2.3 release of the Kortex API about a year ago. 
Unfortunately, this new waypoint system is not implemented in our simulator (that gazebo uses). To move the arm in Gazebo, you must use `PlayJointTrajectory` and `PlayCartesianTrajectory`. These two are still available in the current API, but they are deprecated.
A brief description of the simulator is available in the kortex_example [readme page](https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel/kortex_examples#the-ways-to-use-a-kortex-arm-with-ros).

You can use the old version of the example (before it was updated with the waypoints system) : https://github.com/Kinovarobotics/ros_kortex/blob/v2.2.2/kortex_examples/src/full_arm/example_full_arm_movement.py

I have not tested it, but it should still work pretty much out of the much with maybe some minor tweaks. 

Regards,
Felix

felixmaisonneuve, 2022-04-12:
Hi @AaronLiu666 

I assume I answered the original issue. I will close it.
If you still encounter problems, feel free to comment/re-open it or create a new issue

Regards,
Felix

closed 2022-06-08 by felixmaisonneuve